gem5/gem5 #442: RISC-V Vector Extension float32_t bugs/unsupported widening instructions

OMaghiarIMG, 2023-10-12:
**Describe the bug**
Hello, tried running some RISC-V RVV code using syscall mode and ran into a couple of unsupported instructions/bugs.

**Affects version**
Develop branch https://github.com/gem5/gem5/commit/486916b5d4339ea68141b11bafdb167176cf2fb4

**gem5 Modifications**
No modifications.

**To Reproduce**
```
scons build/RISCV/gem5.fast -j8
$GEM5/build/RISCV/gem5.fast $GEM5/configs/example/gem5_library/riscv-se.py 
```
Script used for SE mode using RISCVMatchedBoard:
```
from gem5.components.boards.riscv_board import RiscvBoard
from gem5.prebuilt.riscvmatched.riscvmatched_board import RISCVMatchedBoard
from gem5.components.processors.cpu_types import CPUTypes
from gem5.isas import ISA
from gem5.utils.requires import requires
from gem5.resources.resource import obtain_resource, BinaryResource
from gem5.simulate.simulator import Simulator

# Run a check to ensure the right version of gem5 is being used.
requires(isa_required=ISA.RISCV)

# Setup the board.
board = RISCVMatchedBoard()
binary = BinaryResource("/path/to/gem5_example")
board.set_se_binary_workload(binary)

simulator = Simulator(board=board, full_system=False)
print("Beginning simulation!")
simulator.run()
print("Exiting @ tick {} because {}.".format(simulator.get_current_tick(), simulator.get_last_exit_event_cause(),))
```
Using QEMU(v8.0) user-mode side-by-side to show expected results.

## 1. Some widening instructions cause panic for LMUL=8: 
Using a dot product example for _vwredsum_, normally most widening instructions don't work with LMUL=8 as you can't widen to LMUL=16, but some of them such as _vwredsum_ and _vwmacc_ should work correctly.
```
#include "riscv_vector.h"
#include "stdio.h"
#include "stdlib.h"
void rvv_dot_prod(int16_t *pSrcA, int16_t *pSrcB, uint32_t n, int64_t *result)
{
    size_t vl;
    vint16m4_t vSrcA, vSrcB;
    vint64m1_t vSum = __riscv_vmv_s_x_i64m1(0, 1);
    while (n > 0) {
        vl = __riscv_vsetvl_e16m4(n);
        vSrcA = __riscv_vle16_v_i16m4(pSrcA, vl);
        vSrcB = __riscv_vle16_v_i16m4(pSrcB, vl);
        vSum = __riscv_vwredsum_vs_i32m8_i64m1(__riscv_vwmul_vv_i32m8(vSrcA, vSrcB, vl), vSum, vl);
        pSrcA += vl;
        pSrcB += vl;
        n -= vl;
    }
    *result = __riscv_vmv_x_s_i64m1_i64(vSum);
}

void scalar_dot_prod(int16_t *pSrcA, int16_t *pSrcB, uint32_t n, int64_t *result)
{
    int64_t sum = 0;
    while (n > 0) {
        sum += ((int64_t)*pSrcA++ * *pSrcB++);
        n--;
    }
    *result = sum;
[... 138 lines cut ...]
board.remote_gdb: Listening for connections on port 7000
src/sim/simulate.cc:194: info: Entering event queue @ 0.  Starting simulation...
src/sim/mem_state.cc:448: info: Increasing stack size by one page.
src/sim/mem_state.cc:448: info: Increasing stack size by one page.
RVV: nan 
Scalar: 58.000000
Exiting @ tick 18640874 because exiting with last active thread context.

$ qemu-riscv64 -cpu rv64,v=true,vext_spec=v1.0,vlen=256,elen=64 gem5_example 
RVV: 58.000000 
Scalar: 58.000000
```

**Host Operating System**
Ubuntu 20.04

**Host ISA**
X86

**Compiler used**
gcc 9.4.0 to build gem5
Clang 16 for RVV code
```
clang -O3 -march=rv64gcv --target=riscv64-unknown-linux-musl -static -o gem5_example main.c
```

powerjg, 2023-10-12:
Thanks for the bug report! You've done a great job explaining the issue!

ivanaamit, 2024-01-18:
Hi @OMaghiarIMG, could you re-run this on develop and check if some of the issues have been resolved with the new version? Thank you.

kourzanov, 2024-02-08:
We analyzed the problem and the root cause is gem5's RVV not conforming to RISC-V's NaN-boxing requirements [image]
Basically there are float32 instructions (vfmv ones) that produce values in 64-bit float registers which are not 1-padded, which results in other instructions interpreting the value as a NaN (even with correct 32-bit contents, but no NaN-boxing)

ivanaamit, 2024-04-05:
Hi @OMaghiarIMG, I want to try to reproduce this issue on the current develop branch to see if it is still a bug. Would it be possible for you to share precompiled binaries of these programs to save me the trouble? Thanks.

ivanaamit, 2024-04-09:
You can use the following script to reproduce your errors. The problem with your script is that the RISCVMatchedBoard does not have RVV enabled. Both of the errors you reported are bugs and should be fixed. If you have fixes for either, please feel free to contribute. Thanks.

```
from gem5.components.boards.simple_board import SimpleBoard
from gem5.components.cachehierarchies.classic.no_cache import NoCache
from gem5.components.memory import SingleChannelDDR3_1600
from gem5.components.processors.cpu_types import CPUTypes
from gem5.components.processors.simple_processor import SimpleProcessor
from gem5.isas import ISA
from gem5.resources.resource import obtain_resource, BinaryResource
from gem5.simulate.simulator import Simulator
from gem5.utils.requires import requires

requires(isa_required=ISA.RISCV)

cache_hierarchy = NoCache()

memory = SingleChannelDDR3_1600(size="124MB")

processor = SimpleProcessor(cpu_type=CPUTypes.TIMING, isa=ISA.RISCV, num_cores=1)

board = SimpleBoard(
    clk_freq="3GHz",
    processor=processor,
    memory=memory,
    cache_hierarchy=cache_hierarchy,
)

board.set_se_binary_workload(BinaryResource("/path/to/gem5_example"))

simulator = Simulator(board=board)
simulator.run()
```

tommarin, 2024-07-13:
The float test seems to pass now (tested with gem5 24.0). I've submitted a PR to fix the widening instructions problem. Once merged, I think this issue can be closed.

BobbyRBruce, 2024-08-07:
This issue has been fixed by #1331

closed 2024-08-07 by BobbyRBruce