gem5/gem5 #449: Arm timing 2-core CHI Boot test failing with `fatal: ... alloc_on_atomic without default or user set value`

BobbyRBruce, 2023-10-13:
**Describe the bug**

The [gem5 Daily tests are currently failing](https://github.com/gem5/gem5/actions/runs/6492500986/job/17634711866).

It is the `SuiteUID:tests/gem5/arm_boot_tests/test_linux_boot.py:timing-cpu_2-cores_chi_DualChannelDDR4_2400_arm_boot_test_m5-exit-ALL-x86_64-opt-CHI` Test suite is failing with `fatal: board.cache_hierarchy.core_clusters0.dcache.alloc_on_atomic without default or user set value`.

I suspect given the error is coming from the `cache_hierarchy` that this is a bug with CHI, but I have not investigated this issue beyond just reporting the error.

**Affects version**

At the time of writing it affects the `develop` branch. This report was based on revision [f7ad8fe4350be7d4299b22d9030385e3f261c7d8](https://github.com/gem5/gem5/commit/f7ad8fe4350be7d4299b22d9030385e3f261c7d8).

**To Reproduce**

The failing test can be run with:

```sh
cd tests
./main.py run --length=long --uid timing-cpu_2-cores_chi_DualChannelDDR4_2400_arm_boot_test_m5-exit-ALL-x86_64-opt-CHI -j`nproc`
```

The code this tests runs is the following:

```sh
# Build gem5 to the CHI Protocol
scons build/ALL_CHI/gem5.opt -j `nproc`

# Execute the following.
.build/ALL_CHI/gem5.opt tests/gem5/arm_boot_tests/configs/arm_boot_exit_run.py --cpu timing --num-cpus 2 --mem-system chi --dram-class DualChannelDDR4_2400 --resource-directory tests/gem5/resources
```

**Terminal Output**

```sh
fatal: board.cache_hierarchy.core_clusters0.dcache.alloc_on_atomic without default or user set value
```

**Host Operating System**

Ubuntu 22.04

**Host ISA**

X86

**Compiler used**

GCC

giactra, 2023-10-13:
Thanks Bobby I am going to have a look at it

giactra, 2023-10-16:
The initial fix for this issue:

> fatal: board.cache_hierarchy.core_clusters0.dcache.alloc_on_atomic without default or user set value

is rather easy. You just need to setup the alloc_on_atomic variable for every CHI node in the stdlib:

- src/python/gem5/components/cachehierarchies/chi/nodes/directory.py
- src/python/gem5/components/cachehierarchies/chi/nodes/dma_requestor.py
- src/python/gem5/components/cachehierarchies/chi/nodes/private_l1_moesi_cache.py

However even after initializing the parameter we get a different runtime failure (invalid CHI transaction). Currently debugging that problem

powerjg, 2023-10-19:
@giactra Any updates on this? I believe this is blocking daily tests passing.

giactra, 2023-10-24:
Hi @powerjg apologies, but I have been relatively busy these days; I tried to fiddle a bit in the test params to make it work (like disabling the far-atomics policy by setting policy_type = 0) but this doesn't seem to work. I am going to try today/tomorrow to debug and provide a proper fix; If I don't manage I guess we can consider a different solution...

giactra, 2023-10-28:
So it seems like I have found the problem...The issue was the OOM process killing gem5. With a lot of pain I finally realized the leak is related to https://github.com/gem5/gem5/issues/508. I have provided a fix and I am rerunning regressions as of now.

giactra, 2023-11-10:
This has been fixed

closed 2023-11-10 by giactra