gem5/gem5 #450: memtests failing with `panic: system.cpu6: read of 10ff86 (blk 10ff80) @ cycle 2000 returns 3c, expected 0`

BobbyRBruce, 2023-10-13:
**Describe the bug**

The [gem5 Daily tests are currently failing](https://github.com/gem5/gem5/actions/runs/6492500986/job/17634711866).

The `SuiteUID:tests/gem5/memory/test.py:memtest-NULL-x86_64-opt` test suite is failing with `src/cpu/testers/memtest/memtest.cc:161: panic: system.cpu6: read of 10ff86 (blk 10ff80) @ cycle 2000 returns 3c, expected 0`.

This same bug appears to be affecting the other memtest `long` tests in "tests/gem5/memory".

**Affects version**

At the time of writing it affects the develop branch. This report was based on revision [f7ad8fe4350be7d4299b22d9030385e3f261c7d8](https://github.com/gem5/gem5/commit/f7ad8fe4350be7d4299b22d9030385e3f261c7d8).

**To Reproduce**

To run the test:

```sh
cd tests
./main.py run --length=long -j`nproc` -uid SuiteUID:tests/gem5/memory/test.py:memtest-NULL-x86_64-opt
```

To run the script directly (i.e., what the script calls):

```sh
# Build gem5.
scons build/NULL/gem5.opt -j12

# Run the memory test
./build/NULL/gem5.opt tests/gem5/memory/memtest-run.py
```

**Terminal Output**

```sh
src/cpu/testers/memtest/memtest.cc:161: panic: system.cpu6: read of 10ff86 (blk 10ff80) @ cycle 2000 returns 3c, expected 0
```

**Host Operating System**

Ubuntu 22.04

**Host ISA**

X86

**Compiler used**

GCC

powerjg, 2023-10-13:
Pretty sure this was introduced by #177. 4fd9d66c537671f0950687cac327c4513c9c8eed was the first "bad" commit in the bisect.

@vsoriap  Was the new atomic support in the `MemTest` tested with the classic caches? Should we disable this or did this expose a bug?

Also pinging @tiagormk and @giactra  who approved that change.

If atomics are known not to work in the classic caches, then we have two options:
1. Disable atomics in the `tests/gem5/memory/memtest-run.py` file
2. Set the default percent atomic to `0` in `src/cpu/testers/memtest/MemTest.py`

I would prefer (2).

If atomics are supposed to work in the classic caches... well, we have a bigger bug to track down. We should consider to (1) or (2) above to get tests passing and work on the bug in parallel.

vsoriap, 2023-10-14:
Hi @powerjg,

I didn't test the changes with the classic caches. I assumed that classic caches supported atomic ops, as Full System emulation works correctly with recent Linux versions that include atomics. Anyways, I'm not an expert in classic caches.

I would also opt for setting percent atomic to 0 (Option 2) as the preferred solution.

BobbyRBruce, 2023-10-18:
Fixed by #477

closed 2023-10-18 by BobbyRBruce