gem5/gem5 #451: gem5 simulates the full system under the riscv architecture, the coregets stall after switching the CPU

qqli579, 2023-10-13:
**Describe the bug**
A clear and concise description of what the bug is.

I use SimpleSwitchableProcessor for simulation.：
processor = SimpleSwitchableProcessor(
    starting_core_type=CPUTypes.ATOMIC,
    switch_core_type=CPUTypes.TIMING,
    isa=ISA.RISCV,
    num_cores=2,
)

After the OS starts up, I  log in and execute m5 exit, which triggers the CPU to switch to TIMING cpu, but the following information is printed at this time：
675404961000: board.processor.switch.core.fetch: Running stage
675404961000: board.processor.switch.core.fetch: Attemupting to fetch from [tid:0]
675404961000: board.processor.switch.core.fetch: [tid:0] Attempting to translate and read instruction, starting at PC(0x17322=>0x17326).(0=>1)
675404961000: board.processor.switch.core.fetch: [tid:0] Fetching cache line 0x17300 for addr 0x17320ttempting to traanslate
src/cpu/03/fetch.cc:604: warn: Address 0x17300 is outside of physical memory, stopping fetch
675404961000: board.processor.switch.core: Schedulingnext tick!
From the log, we can see that after switching the CPU, we have been fetching instructions from address 0x17300, but this address is is outside of physical memory, resulting in no instructions being fetched, and the CPU stalled. 
The analysis found that the status of the CSR register was wrong after switching the CPU, causing the virtual address 0x17300 to not be converted by TLB.
**Affects version**
v23.0.1.0

**gem5 Modifications**
If you have modified gem5 in some way please state, to the best of your ability, how it has been modified.

**To Reproduce**
Steps to reproduce the behavior. Please assume starting from a clean repository:
1. Compile gem5 with command ...
2. Execute the simulation with...

If writing code, or a terminal command, use code blocks. Either an inline code block, \`scons build/ALL/gem5.opt\` (enclosed in two \`) or a multi-line codeblock:

\`\`\`

int x=2;

int y=3'

print(x+y);

\`\`\`

If possible, please include the Python configuration script used and state clearly any parameters passed.

**Terminal Output**
If applicable, add the terminal output here. If long, only include the relevant lines.
Please put the terminal output in code blocks. I.e.:

\`\`\`

#Terminal output here#

\`\`\`

**Expected behavior**
A clear and concise description of what you expected to happen.

**Host Operating System**
Ubuntu 22.04, Mac OS X, etc.

**Host ISA**
ARM, X86, RISC-V, etc.

**Compiler used**
State which compiler was used to compile gem5. Please include the compiler version.

**Additional information**
Add any other information which does not fit in the previous sections but may be of use in fixing this bug.

hnpl, 2023-10-16:
I also ran into the same problem when switching from ATOMIC to other CPU types. The proposed solution looks good to me. Can you make a PR with your solution?

hnpl, 2023-10-18:
I made a change in #479. Please review if you have time! Thanks for providing very helpful information about the issue.

qqli579, 2023-10-18:
it's ok

closed 2023-11-20 by Harshil2107