gem5/gem5 #456: There are no GPUFS tests

abmerop, 2023-10-14:
There are no GPUFS tests despite being usable for several years.  All of the old resources are available so we could do something like the code below.  The last release is broken on stable and apparently fixing on stable is impossible (see #295) so testing is necessary. 

Something like this should work and takes 1-2 minutes with KVM:

```  gpufs-square-test:
    runs-on: [self-hosted, linux, x64, build]
    container: # whatever the default is ?
    needs: build-gem5  # don't know ?
    timeout-minutes: 20
    steps:
      # Is the VEGA_X86 build already available ??
      - name: Get disk image
        working-directory: ${{ github.workspace }}  # Is this already available somewhere?
        run:  build/VEGA_X86/gem5.opt util/obtain-resource.py x86-gpu-fs-img -p x86-gpu-fs-img
      - name: Get kernel
        working-directory: ${{ github.workspace }}  # Some cache folder??
        run:  build/VEGA_X86/gem5.opt util/obtain-resource.py x86-linux-kernel-5.4.0-105-generic -p x86-linux-kernel-5.4.0-105-generic
      - name: Get square
        working-directory: ${{ github.workspace }}  # Some cache folder??
        run:  build/VEGA_X86/gem5.opt util/obtain-resource.py x86-gpu-square -p x86-gpu-square
      - name: Run GPUFS square test
        working-directory: ${{ github.workspace }}
        run: build/VEGA_X86/gem5.opt configs/example/gpufs/vega10_kvm.py --disk-image x86-gpu-fs-img --kernel x86-linux-kernel-5.4.0-105-generic --app x86-gpu-square
      # Ideally would check for "PASSED!" in output too```

Looks like DevOps is in my future.

mattsinc, 2023-10-14:
@v-ramadas : can you please check if this works with your stuff?

v-ramadas, 2023-10-15:
Running the square command gives me the following error : 
`OSError: Can't find file 'disks/x86-gpu-fs-img' on M5_PATH`.
If I specify the disk image path as `--disk-image ../gem5/x86-gpu-fs-img` instead of `--disk-image x86-gpu-fs-img`, it moves past this.

Similarly, I need to pass `--kernel ../gem5/x86-linux-kernel-5.4.0-105-generic` instead of `--kernel x86-linux-kernel-5.4.0-105-generic` along with the `--gpu-mmio-trace` argument. Square passes with these changes

abmerop, 2023-10-16:
I did observe this OSError actually.  I had to use `./x86-gpu-fs-img` to fix it.  I'll update the original post, but I think gem5 should either check the current working directory at the very least a print statement saying it is falling back to M5_PATH instead of silently doing that, since the error message is very confusing otherwise

BobbyRBruce, 2023-10-16:
I'd love to quickly push these into our testing infrastructure but the big sticking point here is whether I can get KVM working on our runners. I simply don't know if this is possible or not. I've never tried.

If i can find 20 minutes when at some point I'll poke around. There are a few options herel.

abmerop, 2023-10-16:
Thanks!  Hopefully we have some way to test KVM in general.   There is an alternate atomic CPU config if that's not possible:

```run: build/VEGA_X86/gem5.opt configs/example/gpufs/vega10_atomic.py --disk-image ./x86-gpu-fs-img --kernel ./x86-linux-kernel-5.4.0-105-generic --app ./x86-gpu-square --gpu-mmio-trace ./gem5-resources/src/gpu-fs/vega_mmio.log
      # Ideally would check for "PASSED!" in output too```

v-ramadas, 2023-10-16:
@abmerop, I can take a look at the directory issue when I get some time.

abmerop, 2023-10-16:
> I'd love to quickly push these into our testing infrastructure but the big sticking point here is whether I can get KVM working on our runners. I simply don't know if this is possible or not. I've [...]

BTW, #95 and #433 might be useful if KVM exists, but the paranoid level is too high

BobbyRBruce, 2023-10-17:
I played around today with KVM and the runners, I submitted a PR here: https://github.com/gem5/gem5/pull/470. It's all good news but perhaps not good enough.

When the Github runners are created they now check to see if KVM can be setup, and, if so, it is. Another check is then performed to see if KVM can be run from within a docker container. If this is the case that action runner is given a `kvm` label indicating it "works with kvm".

The issue is, the only way to get KVM working within a docker container is to pass `--device /dev/kvm` to the `docker run` command. It doesn't appear GitHub Action runners do that, or can do that, so it may be a dead-end. I've not found anything hopeful online, but perhaps there is a way.

powerjg, 2023-10-17:
Can you use `createOptions` to add `--device /dev/kvm`? See https://docs.github.com/en/actions/hosting-your-own-runners/managing-self-hosted-runners/customizing-the-containers-used-by-jobs#arguments-for-prepare_job

BobbyRBruce, 2024-01-19:
Getting GPUFS tests incorporated is contingent on getting KVM working in our GitHub Action runners as these are what runs our tests. I've outlined the issue of incorporating KVM into the GitHub Action Runners infrastructure in Issue #787.

abmerop, 2025-04-23:
I have been experimenting with the latest ROCm (6.4) and there have been a few changes relevant to this issue.  The last comment here is about KVM but I believe we are testing KVM now.  If I remember correctly, the next issue was that the virtual environment the actions run in has ~64GB of disk (memory?) and the GPUFS disk image was too big? With ROCm 6.4 there was some effort to reduce the installation size.  I am currently seeing 42GB (bumped to 50GB for some user space): https://github.com/abmerop/gem5-resources/commit/923203737e8108b3b66fbc7cf22faebe8308adcc#diff-574dbd7ea9327720fdf91c9e48a25299eb670e7f0147a6686cb02e0ecafbf9e1 which I believe would now fit in the runner + gem5 build size.

@BobbyRBruce is my recollection correct? Do you think there would be other issues I am missing? I would like to get *some* tests as currently I just have a personal nightly tester running on develop.

powerjg, 2025-04-25:
* We still can't use KVM on our testers. They run inside a docker container inside a VM. I don't think the hardware or hypervisor that we're using has nested virtualization enabled (at least, last time we tried it didn't work).
* We only have ~16 GiB of RAM for each VM. If we want to be able to run 10s of tests in parallel, we have to keep each VM pretty small.
* Almost 100% of our cloud costs are bandwidth out of Google cloud. It costs us about $500/month right now. If we were to add tests that downloaded a large disk image then our costs would go even higher. (Side note: we're working on getting these costs down by migrating to Azure, but our costs will still be directly proportional to how much we download. Caching is a *option*, but the way our runners/testers work it would be a very high engineering cost to implement.)

mattsinc, 2025-04-25:
I don't know everything here, but is this something I could talk to the IT folks here for the test machine to see if the appropriate permissions need to be given?

powerjg, 2025-04-25:
It is going to take a significant amount of engineering effort to get KVM tests working. Unless your IT can take the engineering effort off of us, I don't think there's anything they can do.

abmerop, 2025-04-25:
I was pretty sure KVM was already working.  If it is not, how are these tests running? https://github.com/gem5/gem5/blob/stable/tests/gem5/kvm_switch_tests/test_kvm_cpu_switch.py Are they not really using KVM?

powerjg, 2025-04-25:
I don't think they are running:
https://github.com/gem5/gem5/blob/stable/tests/gem5/kvm_switch_tests/test_kvm_cpu_switch.py#L44

Here is an example:
https://github.com/gem5/gem5/actions/runs/14655755263/job/41133563548

If you click the "Run "tests/kvm..." button you'll see it doesn't run as shown below.

[image]

BobbyRBruce, 2026-09-07:
An update here: [#3401](https://github.com/gem5/gem5/pull/3401) adds three GPU full-system tests and integrates them into the normal `build/ALL/gem5.opt` TestLib CI and Daily matrices. These tests use an Atomic CPU, so they do not depend on KVM or the former `gcn-gpu` container.

The coverage is split into two levels:

- **Pull-request CI:** A quick test restores a warmed MI355X (`gfx950`) checkpoint. The checkpoint already contains an initialized Linux/ROCm environment and HIP runtime, a loaded `gfx950` module, and one completed warm-up dispatch. After restoration, the test launches a second dispatch of a real HIP kernel, synchronizes, and verifies its output. Only after (and conditional on) that succeeds does the guest print `GPU checkpoint restore test passed`, which TestLib requires in the serial output.
- **Daily testing:** Two longer tests cold-boot the GPUFS image with an MI210/MI200-family (`gfx90a`) GPU and an MI355X (`gfx950`) GPU. Each loads the guest driver, runs `rocminfo`, and verifies that the expected GPU architecture is discovered. These tests cover the cold-boot and driver/device-enumeration path that the warmed pull-request checkpoint intentionally skips; they do not verify a compute result.

All three tests run through the public `configs/example/gem5_library/x86-mi200-gpu.py` and `x86-mi355x-gpu.py` examples using `ViperBoard`, the Viper cache hierarchy, and a reduced four-CU topology. The disk, kernel, checkpoint, and kernel are versioned gem5 Resources, with reproducible sources in [gem5-resources #84](https://github.com/gem5/gem5-resources/pull/84). Sparse disk-image materialization reduces the allocated runner storage, while the warmed checkpoint keeps the pull-request test within a practical runtime.

We ran gcov over `src/gpu-compute/`: the pull-request CI GPUFS test covers 51.6% of executable lines in `src/gpu-computer` (line graph below). We ran the MI335 Daily test as well and around it covered 14.1%  -- smaller, as expected, as it barely runs any GPU code beyond `rocminfo`. 

[image]

closed 2026-09-11 by BobbyRBruce

abmerop, 2026-09-16:
Thanks again. FYI, the coverage is probably higher than reported since there are some SE-mode only files here that will never be used (e.g., gpu_compute_driver.*, gpu_render_driver.*) and represent ~5% of the code in the directory.