gem5/gem5 #472: Failed to build gem5 with SystemC in Ubuntu 23.10

rogerchang23424, 2023-10-17:
**Describe the bug**
A clear and concise description of what the bug is.

**Affects version**
Top of develop branch (commit-hash: adb54709969a06c0ed390cf0013aa86dd27785cf)

**To Reproduce**
```bash
scons -j$(nproc) build/RISCV/gem5.opt
```

**Terminal Output**

```shell
In file included from src/systemc/ext/tlm_utils/simple_initiator_socket.h:24,
                 from src/systemc/tlm_bridge/gem5_to_tlm.hh:73,
                 from src/systemc/tlm_bridge/gem5_to_tlm.cc:59:
src/systemc/ext/tlm_utils/../tlm_core/2/sockets/initiator_socket.hh: In instantiation of ‘class tlm::tlm_base_initiator_socket<32, tlm::tlm_fw_transport_if<>, tlm::tlm_bw_transport_if<>, 1, sc_core::SC_ONE_OR_MORE_BOUND>’:
src/systemc/ext/tlm_utils/../tlm_core/2/sockets/initiator_socket.hh:185:7:   required from ‘class tlm::tlm_initiator_socket<32, tlm::tlm_base_protocol_types, 1, sc_core::SC_ONE_OR_MORE_BOUND>’
src/systemc/ext/tlm_utils/simple_initiator_socket.h:37:7:   required from ‘class tlm_utils::simple_initiator_socket_b<sc_gem5::Gem5ToTlmBridge<32>, 32, tlm::tlm_base_protocol_types, sc_core::SC_ONE_OR_MORE_BOUND>’
src/systemc/ext/tlm_utils/simple_initiator_socket.h:156:7:   required from ‘class tlm_utils::simple_initiator_socket<sc_gem5::Gem5ToTlmBridge<32>, 32, tlm::tlm_base_protocol_types>’
src/systemc/tlm_bridge/gem5_to_tlm.hh:154:46:   required from ‘class sc_gem5::Gem5ToTlmBridge<32>’
src/systemc/tlm_bridge/gem5_to_tlm.cc:614:57:   required from here
src/systemc/ext/tlm_utils/../core/sc_port.hh:125:18: error: ‘void sc_core::sc_port_b<IF>::bind(sc_core::sc_port_b<IF>&) [with IF = tlm::tlm_fw_transport_if<>]’ was hidden [-Werror=overloaded-virtual=]
  125 |     virtual void bind(sc_port_b<IF> &p) { sc_port_base::bind(p); }
      |                  ^~~~
In file included from src/systemc/ext/tlm_utils/simple_initiator_socket.h:27:
src/systemc/ext/tlm_utils/../tlm_core/2/sockets/initiator_socket.hh:133:18: note:   by ‘tlm::tlm_base_initiator_socket<32, tlm::tlm_fw_transport_if<>, tlm::tlm_bw_transport_if<>, 1, sc_core::SC_ONE_OR_MORE_BOUND>::bind’
  133 |     virtual void bind(bw_interface_type &ifs) { (get_base_export())(ifs); }
      |                  ^~~~
src/systemc/ext/tlm_utils/../core/sc_port.hh:124:18: error: ‘void sc_core::sc_port_b<IF>::bind(IF&) [with IF = tlm::tlm_fw_transport_if<>]’ was hidden [-Werror=overloaded-virtual=]
  124 |     virtual void bind(IF &i) { sc_port_base::bind(i); }
      |                  ^~~~
src/systemc/ext/tlm_utils/../tlm_core/2/sockets/initiator_socket.hh:133:18: note:   by ‘tlm::tlm_base_initiator_socket<32, tlm::tlm_fw_transport_if<>, tlm::tlm_bw_transport_if<>, 1, sc_core::SC_ONE_OR_MORE_BOUND>::bind’
  133 |     virtual void bind(bw_interface_type &ifs) { (get_base_export())(ifs); }
      |                  ^~~~
src/systemc/ext/tlm_utils/../tlm_core/2/sockets/initiator_socket.hh: In instantiation of ‘class tlm::tlm_base_initiator_socket<64, tlm::tlm_fw_transport_if<>, tlm::tlm_bw_transport_if<>, 1, sc_core::SC_ONE_OR_MORE_BOUND>’:
src/systemc/ext/tlm_utils/../tlm_core/2/sockets/initiator_socket.hh:185:7:   required from ‘class tlm::tlm_initiator_socket<64, tlm::tlm_base_protocol_types, 1, sc_core::SC_ONE_OR_MORE_BOUND>’
src/systemc/ext/tlm_utils/simple_initiator_socket.h:37:7:   required from ‘class tlm_utils::simple_initiator_socket_b<sc_gem5::Gem5ToTlmBridge<64>, 64, tlm::tlm_base_protocol_types, sc_core::SC_ONE_OR_MORE_BOUND>’
src/systemc/ext/tlm_utils/simple_initiator_socket.h:156:7:   required from ‘class tlm_utils::simple_initiator_socket<sc_gem5::Gem5ToTlmBridge<64>, 64, tlm::tlm_base_protocol_types>’
src/systemc/tlm_bridge/gem5_to_tlm.hh:154:46:   required from ‘class sc_gem5::Gem5ToTlmBridge<64>’
src/systemc/tlm_bridge/gem5_to_tlm.cc:621:57:   required from here
src/systemc/ext/tlm_utils/../core/sc_port.hh:125:18: error: ‘void sc_core::sc_port_b<IF>::bind(sc_core::sc_port_b<IF>&) [with IF = tlm::tlm_fw_transport_if<>]’ was hidden [-Werror=overloaded-virtual=]
  125 |     virtual void bind(sc_port_b<IF> &p) { sc_port_base::bind(p); }
      |                  ^~~~
src/systemc/ext/tlm_utils/../tlm_core/2/sockets/initiator_socket.hh:133:18: note:   by ‘tlm::tlm_base_initiator_socket<64, tlm::tlm_fw_transport_if<>, tlm::tlm_bw_transport_if<>, 1, sc_core::SC_ONE_OR_MORE_BOUND>::bind’
  133 |     virtual void bind(bw_interface_type &ifs) { (get_base_export())(ifs); }
      |                  ^~~~
src/systemc/ext/tlm_utils/../core/sc_port.hh:124:18: error: ‘void sc_core::sc_port_b<IF>::bind(IF&) [with IF = tlm::tlm_fw_transport_if<>]’ was hidden [-Werror=overloaded-virtual=]
  124 |     virtual void bind(IF &i) { sc_port_base::bind(i); }
      |                  ^~~~
src/systemc/ext/tlm_utils/../tlm_core/2/sockets/initiator_socket.hh:133:18: note:   by ‘tlm::tlm_base_initiator_socket<64, tlm::tlm_fw_transport_if<>, tlm::tlm_bw_transport_if<>, 1, sc_core::SC_ONE_OR_MORE_BOUND>::bind’
  133 |     virtual void bind(bw_interface_type &ifs) { (get_base_export())(ifs); }
      |                  ^~~~
src/systemc/ext/tlm_utils/../tlm_core/2/sockets/initiator_socket.hh: In instantiation of ‘class tlm::tlm_base_initiator_socket<128, tlm::tlm_fw_transport_if<>, tlm::tlm_bw_transport_if<>, 1, sc_core::SC_ONE_OR_MORE_BOUND>’:
src/systemc/ext/tlm_utils/../tlm_core/2/sockets/initiator_socket.hh:185:7:   required from ‘class tlm::tlm_initiator_socket<128, tlm::tlm_base_protocol_types, 1, sc_core::SC_ONE_OR_MORE_BOUND>’
src/systemc/ext/tlm_utils/simple_initiator_socket.h:37:7:   required from ‘class tlm_utils::simple_initiator_socket_b<sc_gem5::Gem5ToTlmBridge<128>, 128, tlm::tlm_base_protocol_types, sc_core::SC_ONE_OR_MORE_BOUND>’
src/systemc/ext/tlm_utils/simple_initiator_socket.h:156:7:   required from ‘class tlm_utils::simple_initiator_socket<sc_gem5::Gem5ToTlmBridge<128>, 128, tlm::tlm_base_protocol_types>’
src/systemc/tlm_bridge/gem5_to_tlm.hh:154:46:   required from ‘class sc_gem5::Gem5ToTlmBridge<128>’
src/systemc/tlm_bridge/gem5_to_tlm.cc:628:57:   required from here
src/systemc/ext/tlm_utils/../core/sc_port.hh:125:18: error: ‘void sc_core::sc_port_b<IF>::bind(sc_core::sc_port_b<IF>&) [with IF = tlm::tlm_fw_transport_if<>]’ was hidden [-Werror=overloaded-virtual=]
  125 |     virtual void bind(sc_port_b<IF> &p) { sc_port_base::bind(p); }
      |                  ^~~~
src/systemc/ext/tlm_utils/../tlm_core/2/sockets/initiator_socket.hh:133:18: note:   by ‘tlm::tlm_base_initiator_socket<128, tlm::tlm_fw_transport_if<>, tlm::tlm_bw_transport_if<>, 1, sc_core::SC_ONE_OR_MORE_BOUND>::bind’
  133 |     virtual void bind(bw_interface_type &ifs) { (get_base_export())(ifs); }
      |                  ^~~~
src/systemc/ext/tlm_utils/../core/sc_port.hh:124:18: error: ‘void sc_core::sc_port_b<IF>::bind(IF&) [with IF = tlm::tlm_fw_transport_if<>]’ was hidden [-Werror=overloaded-virtual=]
  124 |     virtual void bind(IF &i) { sc_port_base::bind(i); }
      |                  ^~~~
src/systemc/ext/tlm_utils/../tlm_core/2/sockets/initiator_socket.hh:133:18: note:   by ‘tlm::tlm_base_initiator_socket<128, tlm::tlm_fw_transport_if<>, tlm::tlm_bw_transport_if<>, 1, sc_core::SC_ONE_OR_MORE_BOUND>::bind’
[... 43 lines cut ...]
 [     CXX] src/systemc/tlm_core/2/generic_payload/gp.cc -> RISCV/systemc/tlm_core/2/generic_payload/gp.o
 [     CXX] src/systemc/tlm_core/2/generic_payload/phase.cc -> RISCV/systemc/tlm_core/2/generic_payload/phase.o
 [     CXX] src/systemc/tlm_core/2/quantum/global_quantum.cc -> RISCV/systemc/tlm_core/2/quantum/global_quantum.o
 [     CXX] src/systemc/tlm_core/2/quantum/global_quantum_python.cc -> RISCV/systemc/tlm_core/2/quantum/global_quantum_python.o
 [     CXX] src/systemc/utils/functions.cc -> RISCV/systemc/utils/functions.o
 [     CXX] src/systemc/utils/messages.cc -> RISCV/systemc/utils/messages.o
 [     CXX] src/systemc/utils/report.cc -> RISCV/systemc/utils/report.o
 [     CXX] src/systemc/utils/sc_report.cc -> RISCV/systemc/utils/sc_report.o
cc1plus: all warnings being treated as errors
scons: *** [build/RISCV/systemc/tlm_bridge/gem5_to_tlm.o] Error 1
scons: building terminated because of errors.

```

**Expected behavior**
Build gem5 successfully

**Host Operating System**
Ubuntu 23.10

**Host ISA**
X86-64

**Compiler used**
gcc 13.2.0

powerjg, 2023-10-17:
Does https://github.com/gem5/gem5/pull/454 not fix this?

powerjg, 2023-10-30:
Should this be closed? Was this fixed by #454 or is it still open?

rogerchang23424, 2023-10-31:
Apologize for leave issue for a long time. I have try to make the gcc13 happy by revert the #454, but I will make the clang unhappy. The `-Woverloaded-virtual` need to be ignore for both class `sc_port_b<IF>` and `tlm_base_initiator_socket`.

rogerchang23424, 2023-11-14:
Fixed after #520

closed 2023-11-14 by rogerchang23424