gem5/gem5 #484: Potential bug in TLB lookup

Harshil2107, 2023-10-18:
I am opening this issue in response to the following email on the gem5-users mailing list: https://www.mail-archive.com/gem5-users@gem5.org/msg21992.html

### The email states the following:

In the TLB::lokup method (https://github.com/gem5/gem5/blob/stable/src/arch/riscv/tlb.cc#L109), there is an input argument (i.e., hidden) which determines whether the TLB lookup should or should not update TLB statistics.
  
  In the TLB::doTranslate (https://github.com/gem5/gem5/blob/stable/src/arch/riscv/tlb.cc#L278) method, the first lookup updates TLB statistics (i.e., hidden=false).
  After a TLB miss, the page walker is invoked and the TLB is looked up again in the same way.
  
  The source code is the following:
  ```
  TlbEntry *e = lookup(vaddr, satp.asid, mode, false);
      if (!e) {
          Fault fault = walker->start(tc, translation, req, mode);
          if (translation != nullptr || fault != NoFault) {
              // This gets ignored in atomic mode.
              delayed = true;
              return fault;
          }
          e = lookup(vaddr, satp.asid, mode, false);
          assert(e != nullptr);
      }
  ```
      
 I was wondering if this behavior is correct.
  
Shouldn't we just count the first TLB miss and use the second TLB access only to check the correctness of the page walker? Also because surely the second TLB lookup will definitely result in a TLB hit, otherwise the simulation is interrupted by an assert.

-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Is this a potential bug?

hnpl, 2023-10-20:
That's probably a bug. The second lookup is definitely double count a TLB miss.

powerjg, 2023-11-17:
I assume there has been no progress on this? @Harshil2107 can you add this to the list of "to dos" for v23.1, please.

closed 2023-11-29 by powerjg