gem5/gem5 #487: O3CPU stuck in spinlock after switching CPU during multi-core simulation under riscv architecture

qqli579, 2023-10-19:
**Describe the bug**
I use SimpleSwitchableProcessor for simulation.：
processor = SimpleSwitchableProcessor(
                      starting_core_type=CPUTypes.ATOMIC,
                      switch_core_type=CPUTypes.O3,
                      isa=ISA.RISCV,
                      num_cores=2,)

After the OS starts up, I log in and execute m5 exit, which triggers the CPU to switch to O3CPU, but the following information is printed at this time：
sor.switch1.core: T0 : 0x800086ee @sbi_fifo_dequeue+18    : c_beqz a1, 118             : IntAlu :                                                               
737900311000: board.processor.switch1.core: T0 : 0x800086f0 @sbi_fifo_dequeue+20    : addi s2,a0, 8             : IntAlu :  D=0x0000000080047120                                           
737900311000: board.processor.switch1.core: T0 : 0x800086f4 @sbi_fifo_dequeue+24    : c_mv s1,a0                : IntAlu :  D=0x0000000080047118                                           
737900311000: board.processor.switch1.core: T0 : 0x800086f6 @sbi_fifo_dequeue+26    : c_mv a0,s2                : IntAlu :  D=0x0000000080047120                                           
737900311000: board.processor.switch1.core: T0 : 0x800086f8 @sbi_fifo_dequeue+28    : jal ra, -12374             : IntAlu :  D=0x00000000800086fc                                           
737900312000: board.processor.switch1.core: T0 : 0x800056a2 @spin_lock    : c_addi sp, -16    : IntAlu :  D=0x0000000080046bc8                                                     
737900312000: board.processor.switch1.core: T0 : 0x800056a4 @spin_lock+2    : c_sdsp s0, 8(sp): MemWrite :  D=0x0000000080046c08 A=0x80046bd0                                    
737900312000: board.processor.switch1.core: T0 : 0x800056a6 @spin_lock+4    : c_lui a5, 16   : IntAlu :  D=0x0000000000010000                                                   
737900312000: board.processor.switch1.core: T0 : 0x800056a8 @spin_lock+6    : c_addi4spn s0, sp, 16      : IntAlu :  D=0x0000000080046bd8                                                  
737900312000: board.processor.switch1.core: T0 : 0x800056aa @spin_lock+8    : addi a4, a5, -1 : IntAlu :  D=0x000000000000ffff                                                   
737900312000: board.processor.switch1.core: T0 : 0x800056ae @spin_lock+12    : amoadd_w_aqrl a3, a5, (a0)                                                                                   
737900312000: board.processor.switch1.core: T0 : 0x800056ae @spin_lock+12. 0 : 0x06f526af fence           : No_OpClass :                                                                    
737900312000: board.processor.switch1.core: T0 : 0x800056ae @spin_lock+12. 1 : amoadd_w[l] a3,a5, (a0)   : MemRead :  D=0xffffffff9e6d9e6d A=0x80047120                                    
737900312000: board.processor.switch1.core: T0 : 0x800056ae @spin_lock+12. 2 : 0x06f526af fence           : No_OpClass :                                                                    
737900313000: board.processor.switch1.core: T0 : 0x800056b2 @spin_lock+16    : srli a2, a3, 16: IntAlu :  D=0x0000ffffffff9e6d                                                  
737900313000: board.processor.switch1.core: T0 : 0x800056b6 @spin_lock+20    : c_and a2, a4   : IntAlu :  D=0x0000000000009e6d                                                  
737900313000: board.processor.switch1.core: T0 : 0x800056b8 @spin_lock+22    : and a1, a3, a4 : IntAlu :  D=0x0000000000009e6d                                                 
737900313000: board.processor.switch1.core: T0 : 0x800056bc @spin_lock+26    : beq a2, a1, 12 
At this time, O3CPU stukc in spin lock.
If set num_cores=1：
processor = SimpleSwitchableProcessor(
                      starting_core_type=CPUTypes.ATOMIC,
                      switch_core_type=CPUTypes.O3,
                      isa=ISA.RISCV,
                      num_cores=1,)
it will not encounter this problem. it can continue to run after switching to O3CPU。If you change isa to ISA.X86,  it also will not encounter this problem. So this problem may only be encountered during multi-core simulation under riscv isa

Affects version
v23.0.1.0

gem5 Modifications：
 No solution found yet.

To Reproduce
Steps to reproduce the behavior. Please assume starting from a clean repository:
1 .Change SimpleProcessor in configs/example/gem5_library/riscv-fs.py to SimpleSwitchableProcessor，and make some other modifications：riscv-disk-img needs to be made by yourself, including the m5 package
2. Compile gem5 with command scons ./build/RISCV/gem5.opt -j 32
3. Execute the simulation with ./build/RISCV/gem5.opt configs/example/gem5_library/riscv-fs.py
4. After logging in to the OS, execute command： m5 exit

Host Operating System
Ubuntu 22.04,

Host ISA
X86

powerjg, 2023-10-19:
Is this a duplication of #451?

What *commit* did you run into this problem? Was it before or after #479

qqli579, 2023-10-20:
it is after https://github.com/gem5/gem5/pull/479

hnpl, 2023-10-20:
I'll check this.

hnpl, 2023-10-20:
@qqli579 Just to make sure, are you using the develop branch or the stable branch?

qqli579, 2023-10-21:
I use gem5 stable branch, and i don't know how to check OpenSBI version

qqli579, 2023-10-21:
I use gem5 stable branch, and opensbi  version is v1.3-54-g901d3d7

hnpl, 2023-10-23:
Based on my experiment, switching to O3 CPU with current gem5/develop does not cause the CPU to be stuck.

The following is the Python system configuration I used, which is mostly the same as `riscv-fs.py` but with a switchable processor, and a different disk image.

```Python
# Copyright (c) 2021 The Regents of the University of California
# All rights reserved.
#
# Redistribution and use in source and binary forms, with or without
# modification, are permitted provided that the following conditions are
# met: redistributions of source code must retain the above copyright
# notice, this list of conditions and the following disclaimer;
# redistributions in binary form must reproduce the above copyright
# notice, this list of conditions and the following disclaimer in the
# documentation and/or other materials provided with the distribution;
# neither the name of the copyright holders nor the names of its
# contributors may be used to endorse or promote products derived from
# this software without specific prior written permission.
#
# THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS
# "AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT
# LIMITED TO, THE IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR
# A PARTICULAR PURPOSE ARE DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT
# OWNER OR CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL,
# SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT
# LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; LOSS OF USE,
# DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER CAUSED AND ON ANY
# THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY, OR TORT
# (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE
# OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.

from gem5.components.boards.riscv_board import RiscvBoard
from gem5.components.memory import SingleChannelDDR3_1600
from gem5.components.processors.simple_processor import SimpleProcessor
from gem5.components.processors.simple_switchable_processor import SimpleSwitchableProcessor
from gem5.components.cachehierarchies.classic.private_l1_private_l2_cache_hierarchy import (
    PrivateL1PrivateL2CacheHierarchy,
)
from gem5.components.processors.cpu_types import CPUTypes
from gem5.isas import ISA
from gem5.utils.requires import requires
from gem5.resources.resource import obtain_resource, DiskImageResource
from gem5.simulate.simulator import Simulator
from gem5.utils.override import overrides

requires(isa_required=ISA.RISCV)

cache_hierarchy = PrivateL1PrivateL2CacheHierarchy(
    l1d_size="32KiB", l1i_size="32KiB", l2_size="512KiB"
)

class ModifiedRiscvBoard(RiscvBoard):
    @overrides(RiscvBoard)
    def get_default_kernel_args(self):
        return [
            "earlycon=uart8250,0x10000000",
            "console=uart8250,0x10000000",
            # "console=ttyS0",
            "lpj=7999923",
            "root=/dev/vda1",
            "rw",
            "init=/root/gem5-init.sh"
        ]

# Setup the system memory.
memory = SingleChannelDDR3_1600()

# Setup a single core Processor.
processor = SimpleSwitchableProcessor(
    starting_core_type=CPUTypes.ATOMIC,
    switch_core_type=CPUTypes.O3,
    isa=ISA.RISCV,
    num_cores=2,
)

# Setup the board.
board = ModifiedRiscvBoard(
    clk_freq="1GHz",
    processor=processor,
    memory=memory,
    cache_hierarchy=cache_hierarchy,
)

# Set the Full System workload.
board.set_kernel_disk_workload(
    kernel=obtain_resource("riscv-bootloader-vmlinux-5.10"),
    disk_image=DiskImageResource("../DISK_IMAGES/rv64-2204.img"),
    readfile_contents=f"m5 exit; m5 exit;"
)

simulator = Simulator(board=board)
print("Beginning simulation!")
simulator.run()
processor.switch()
simulator.run()

```

blackwatch997, 2023-10-24:
> Based on my experiment, switching to O3 CPU with current gem5/develop does not cause the CPU to be stuck.
> 
> 根据我的实验，切换到当前 gem5/development 的 O3 CPU 并不会导致 CPU 卡住。
> 
> The following is the Python system configuration I used, which is mostly the same as `riscv-fs.py` but with a switchable processor, and a different disk image.
> 
> 下面是我使用的 Python 系统配置，它与 riscv-fs.py 基本相同，但是带有可切换的处理器和不同的磁盘映像。
> 
> ```python
> # Copyright (c) 2021 The Regents of the University of California
> # All rights reserved.
> #
> # Redistribution and use in source and binary forms, with or without
> # modification, are permitted provided that the following conditions are
> # met: redistributions of source code must retain the above copyright
> # notice, this list of conditions and the following disclaimer;
> # redistributions in binary form must reproduce the above copyright
> # notice, this list of conditions and the following disclaimer in the
> # documentation and/or other materials provided with the distribution;
> # neither the name of the copyright holders nor the names of its
> # contributors may be used to endorse or promote products derived from
> # this software without specific prior written permission.
> #
> # THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS
> # "AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT
> # LIMITED TO, THE IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR
> # A PARTICULAR PURPOSE ARE DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT
> # OWNER OR CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL,
> # SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT
> # LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; LOSS OF USE,
> # DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER CAUSED AND ON ANY
> # THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY, OR TORT
> # (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE
> # OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
> 
> from gem5.components.boards.riscv_board import RiscvBoard
> from gem5.components.memory import SingleChannelDDR3_1600
> from gem5.components.processors.simple_processor import SimpleProcessor
> from gem5.components.processors.simple_switchable_processor import SimpleSwitchableProcessor
> from gem5.components.cachehierarchies.classic.private_l1_private_l2_cache_hierarchy import (
>     PrivateL1PrivateL2CacheHierarchy,
> )
> from gem5.components.processors.cpu_types import CPUTypes
> from gem5.isas import ISA
> from gem5.utils.requires import requires
> from gem5.resources.resource import obtain_resource, DiskImageResource
> from gem5.simulate.simulator import Simulator
> from gem5.utils.override import overrides
> 
> requires(isa_required=ISA.RISCV)
> 
> cache_hierarchy = PrivateL1PrivateL2CacheHierarchy(
>     l1d_size="32KiB", l1i_size="32KiB", l2_size="512KiB"
> )
> 
> class ModifiedRiscvBoard(RiscvBoard):
>     @overrides(RiscvBoard)
>     def get_default_kernel_args(self):
>         return [
>             "earlycon=uart8250,0x10000000",
>             "console=uart8250,0x10000000",
>             # "console=ttyS0",
>             "lpj=7999923",
>             "root=/dev/vda1",
>             "rw",
>             "init=/root/gem5-init.sh"
>         ]
> 
> # Setup the system memory.
> memory = SingleChannelDDR3_1600()
> 
> # Setup a single core Processor.
> processor = SimpleSwitchableProcessor(
>     starting_core_type=CPUTypes.ATOMIC,
>     switch_core_type=CPUTypes.O3,
>     isa=ISA.RISCV,
>     num_cores=2,
> )
> 
> # Setup the board.
> board = ModifiedRiscvBoard(
>     clk_freq="1GHz",
>     processor=processor,
>     memory=memory,
>     cache_hierarchy=cache_hierarchy,
> )
> 
> # Set the Full System workload.
> board.set_kernel_disk_workload(
>     kernel=obtain_resource("riscv-bootloader-vmlinux-5.10"),
>     disk_image=DiskImageResource("../DISK_IMAGES/rv64-2204.img"),
>     readfile_contents=f"m5 exit; m5 exit;"
> )
> 
> simulator = Simulator(board=board)
> print("Beginning simulation!")
> simulator.run()
> processor.switch()
> simulator.run()
> ```
We switched to the develop branch and tried to use the bootload in resouse (riscv-bootloader-vmlinux-5.10"). It can indeed boot, but we are trying to use the bbl+kernel (sbi v0.1) or opensbi + kernel we built ourselves. (sbi v0.2) will encounter the above problems (it passes normally on qemu and spike, as well as the MINOR and atmoicm models in gem5). Can you try to build the bootload yourself to reproduce the above problems? We will try our best I kept the versions of bbl and kernel close to riscv-bootloader-vmlinux-5.10, but it still didn’t work.
Here is some version information we built for your reference：
CROSS_COMPILE riscv64-unknown-linux-gnu-gcc (g2ee5e430018) 12.2.0
KERNEL                linux-5.10.198
BBL                      commit:acbe166dac9d1db752ee95f61e65ca82bb875afb
OPENSBI              tags: V1.0 or V1.3

hnpl, 2023-10-30:
Hi, sorry for the late reply and thanks for the information.

Can you give more details about the Linux kernel boot parameters that you used?

I'm quite confused about what works and what doesn't work in your setup. Can you verify where the following information is true?
(1) You are using the develop branch and tried to boot gem5-resources' riscv-bootloader-vmlinux-5.10 and it worked properly.
(2) You are using the develop branch and tried to boot bbl(sbi 0.1)+kernel(5.10) and it got stuck after switching from ATOMIC to O3CPU.
(3) You are using the develop branch and tried to boot sbi 0.2 + kernel 5.10 and it got stuck after switching from ATOMIC to O3CPU.
(4) You are using the develop branch and tried to boot sbi 1.3 + kernel 5.10 and it got stuck after switching from ATOMIC to O3CPU.

On a side note, I think it's better to use newer versions of the Linux kernel and SBI in general.

I was able to use sbi1.3 + kernel 6.5 and switch the CPU after Linux kernel is done with booting like this, https://github.com/gem5/gem5/issues/487#issuecomment-1775926875.

blackwatch997, 2023-11-29:
I tested the python configuration provided by @hnpl  on the latest develop (d94d6017b051d1) branch, and it can pass normally, but workalod may be too simple. Maybe you can try changing readfile_contents=f"m5 exit; m5 exit;" to readfile_contents =f"m5 exit; bash;m5 exit;", in my test, the terminal could not interact normally. 
This issue is consistent with #613 
We have also located it to be related to wfi. Our solution is as follows:
```
diff --git a/src/arch/riscv/isa/decoder.isa b/src/arch/riscv/isa/decoder.isa
index 3d1d396165..a2f9bd83fd 100644
--- a/src/arch/riscv/isa/decoder.isa
+++ b/src/arch/riscv/isa/decoder.isa
@@ -4644,7 +4644,7 @@ decode QUADRANT default Unknown::unknown() {
                                 tc->quiesce();
                             }
                         }}, IsNonSpeculative, IsQuiesce,
-                            IsSerializeAfter, No_OpClass);
+                            IsSerializeAfter, No_OpClass,IsSquashAfter);
                     }
                     0x9: sfence_vma({{
                         STATUS status = xc->readMiscReg(MISCREG_STATUS);
```
Based on this modification, we have Some benchmarks (up to 32 cores) were run in the simulation, but I haven’t had time to ensure in more detail whether the modification is reasonable. Some details related to the problem will be commented later. 
In addition: I am very sorry for such a late reply. As we thought, we would like to further analyze the rationality of the modification and the problem before replying, but there have been some other things that have been delayed recently, so I am very sorry.

hnpl, 2023-11-29:
@blackwatch997 thank you for the information. I'll try to reproduce this problem.
@kaustav-goswami can you try this WFI fix?