gem5/gem5 #501: Potential misconfiguration for Vega-related systems

mysoreanoop, 2023-10-24:
**Describe the bug**
(Might be useful for debugging recent changes related to GCN3)
Potential unconnected configuration, unmapped address range, unimplemented debug code. The error I see when simulating some CHAI benchmarks (not all) is as follows:
```
src/mem/xbar.cc:368: fatal: Unable to find destination for [0x20001000:0x20001001] on system.piobus
Use --debug-flags=PortTrace to see the port trace of the packet.
```
But with the debug flag, I see:
```
src/mem/port.cc:209: panic: panic condition !ext occurred: There is no TracingExtension in the packet.
```
**Affects version**
develop branch, commit: 6ddf8c94ee1368f8a573fc7f782dcf6a8534293b

**gem5 Modifications**
Unmodified

**To Reproduce**
1. Clone [HIPified CHAI ](https://github.com/mysoreanoop/chai.git) benchmark-suite into gem5-resources/src/gpu/chai
3. Compile gem5 develop branch (commit ID supplied above) and with gem5 and gem5-resources in the same directory (otherwise update the application command-line arguments accordingly), run with docker:
`docker run -u $UID:$GID --volume $(pwd):$(pwd) -w $(pwd) gcr.io/gem5-test/gcn-gpu:v22-1 gem5/build/GCN3_X86/gem5.opt -d chai/sc gem5/configs/example/apu_se.py --gfx-version=gfx902 --cpu-type=DerivO3CPU --num-cpus=12 -c gem5-resources/src/gpu/chai/HIP-U-gem5/SC/bin/sc.gem5 1>chaiscsimout 2>chaiscsimerr`

**Host Operating System**
Ubuntu 20.04, but run inside docker (also same Ubuntu)

**Host ISA**
X86_64

mattsinc, 2023-10-25:
You're going to have to give @abmerop and me more information here.  For example, what is the backtrace that gets to this point?

Honestly, I'm not sure why you are using PortTrace with the GPU in the first place, or what exactly led to this combination.

mysoreanoop, 2023-10-25:
Apologies; here's more information.
Some background: I'm running a HIP version of CHAI benchmark that is failing in the develop branch compilation of gem5.
I added the debug-flag PortTrace based off of the recommendation in the log. Please let me know if you need more information.

The logs are when running a CHAI benchmark called SC.
simerr (filtered out repeating warnings):
```
src/mem/dram_interface.cc:690: warn: DRAM device capacity (8192 Mbytes) does not match the address range assigned (512 Mbytes)
src/base/statistics.hh:279: warn: One of the stats is a legacy stat. Legacy stat is a stat that does not belong to any statistics::Group. Legacy stat is deprecated.
src/base/stats/storage.hh:278: warn: Bucket size (5) does not divide range [1:75] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (2) does not divide range [1:10] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (2) does not divide range [1:64] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1e+06] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (5) does not divide range [1:75] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (2) does not divide range [1:10] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (2) does not divide range [1:64] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1e+06] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (5) does not divide range [1:75] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (2) does not divide range [1:10] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (2) does not divide range [1:64] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1e+06] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (5) does not divide range [1:75] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (2) does not divide range [1:10] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (2) does not divide range [1:64] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1.6e+06] into equal-sized buckets. Rounding up.
...
warn: The `get_runtime_isa` function is deprecated. Please migrate away from using this function.
warn: system.ruby.network adopting orphan SimObject param 'ext_links'
warn: system.ruby.network adopting orphan SimObject param 'int_links'
src/sim/simulate.cc:194: info: Entering event queue @ 0.  Starting simulation...
src/mem/ruby/system/Sequencer.cc:668: warn: Replacement policy updates recently became the responsibility of SLICC state machines. Make sure to setMRU() near callbacks in .sm files!
src/sim/syscall_emul.cc:74: warn: ignoring syscall set_robust_list(...)
...
src/sim/syscall_emul.cc:85: warn: ignoring syscall rt_sigaction(...)
      (further warnings will be suppressed)
src/sim/syscall_emul.cc:74: warn: ignoring syscall get_mempolicy(...)
build/GCN3_X86/arch/x86/generated/exec-ns.cc.inc:27: warn: instruction 'frndint' unimplemented
...
src/sim/syscall_emul.hh:1890: warn: mmap: writing to shared mmap region is currently unsupported. The write succeeds on the target, but it will not be propagated to the host or shared mappings
src/gpu-compute/gpu_compute_driver.cc:144: warn: Unrecognized kfd mmap type 0
src/gpu-compute/gpu_compute_driver.cc:887: warn: unimplemented ioctl: AMDKFD_IOC_MAP_MEMORY_TO_GPU
...
src/sim/syscall_emul.cc:74: warn: ignoring syscall sched_setaffinity(...)
src/sim/syscall_emul.cc:74: warn: ignoring syscall set_robust_list(...)
src/sim/syscall_emul.cc:85: warn: ignoring syscall sched_yield(...)
      (further warnings will be suppressed)
src/mem/xbar.cc:368: fatal: Unable to find destination for [0x20001000:0x20001001] on system.piobus
Use --debug-flags=PortTrace to see the port trace of the packet.
Memory Usage: 2707008 KBytes

```
simout (filtered out):
```
Global frequency set at 1000000000000 ticks per second
gem5 Simulator System.  https://www.gem5.org
gem5 is copyrighted software; use the --copyright option for details.

gem5 version DEVELOP-FOR-23.1
gem5 compiled Oct 23 2023 10:53:27
gem5 started Oct 23 2023 20:37:46
gem5 executing on dd3212def076, pid 1
command line: gem5/build/GCN3_X86/gem5.opt -d chai/sc_902 gem5/configs/example/apu_se.py --gfx-version=gfx902 --cpu-type=DerivO3CPU --num-cpus=12 -c gem5-resources/src/gpu/chai/HIP-U-gem5/SC/bin/sc.gem5

info: Standard input is not a terminal, disabling listeners.
Num SQC =  1 Num scalar caches =  1 Num CU =  4
```

abmerop, 2023-10-25:
This only seems to happen on O3CPU, so adding label for that.  We would have to figure out where the address is coming from next, probably, to figure out what is happening.

No GPU kernels have been launched yet so the address is coming from CPU.

mysoreanoop, 2023-10-26:
Debug with AddrRanges,CoherentXBar,NoncoherentXBar:
```
35205524500: system.piobus: recvTimingReq: src system.piobus.cpu_side_ports[4] ReadReq 0x20000000
35205525010: system.piobus: recvTimingResp: src system.piobus.mem_side_port[1] BadAddressError 0x20000000
```
mem_side_port[1] belongs to hsapp in config.json.
However, in [configs/examples/apu_se.py](https://github.com/gem5/gem5/blob/e9da8d67bdbb23fbd0578a379f08f42bce50121d/configs/example/apu_se.py#L641), the address start and range specified is nowhere used.
[sc_debug.zip](https://github.com/gem5/gem5/files/13177388/sc_debug.zip)

abmerop, 2023-10-26:
I see. It looks like the physical address of the HSA packet processor is simply assumed to be the end of main memory (0x2000000 by default).  I am not too familiar with SE mode but I think only the physical address is used, which is why the virtual address / size do not need to be used anywhere.

I was able to get around this issue by increasing the "pioSize" of the HSAPacketProcessor.  I don't know if Vega has an increased address range for the packet processor vs. GCN3 which does not appear to have this issue though, so I am not sure if that is the "best" way to fix this.

It's also suspicious as I do not see any read/writes outside of the current PIO range from the HSAPacketProcessor debug flags, so I am not sure what type of packet has this bad address

mysoreanoop, 2023-10-26:
Sorry, I unable to follow how increasing the pioSize is helping when you try. Did you also have to manually override the end of main memory address somehow? Because end of main memory is where pio address range for hsapp begins, right?

The type of packet is recvTimingResp (per the 2nd debug line) -- are you looking for something else I can help with?

mysoreanoop, 2023-10-27:
Thanks for the clarification. From where did you get to know the address range of HSAPP?
What's the best way to decode this? I can give you more information if you guide me how to obtain the packet whereabouts.

Here is what I have so far:
```
35206040500: system.ruby.cp_cntrl1.sequencer.response_ports1: Timing request for address 0x20000000 on port 1
35206040500: system.ruby.cp_cntrl1.sequencer.response_ports1: Request address 0x20000000 assumed to be a pio address
35206037000: system.cpu02: T0 : 0x7fffefc72aff @_end+140737211849023    : cdqe  rax                  
35206037000: system.cpu02: T0 : 0x7fffefc72aff @_end+140737211849023. 0 :   CDQE_R : sexti   rax, rax, 0x1f : IntAlu :  D=0xfffffffffffff000                                                                       
35206037000: system.cpu02: T0 : 0x7fffefc72b01 @_end+140737211849025    : hint                       
35206037000: system.cpu02: T0 : 0x7fffefc72b01 @_end+140737211849025. 0 :   HINT_NOP : fault   NoFault : No_OpClass : 
35206041000: system.piobus: recvTimingReq: src system.piobus.cpu_side_ports[4] ReadReq 0x20000000
35206037000: system.cpu02: T0 : 0x7fffefc72b08 @_end+140737211849032    : movzx ecx, DS:[rbx]       
35206037000: system.cpu02: T0 : 0x7fffefc72b08 @_end+140737211849032. 0 :   MOVZX_B_R_M : ld   t1b, DS:[rbx] : MemRead :  A=0x7fff5e1e5000
35206041010: system.piobus: recvTimingResp: src system.piobus.mem_side_port[1] BadAddressError 0x20000000
35206041500: system.ruby.cp_cntrl1.sequencer.response_ports1: Timing request for address 0x20001000 on port 1
35206041500: system.ruby.cp_cntrl1.sequencer.response_ports1: Request address 0x20001000 assumed to be a pio address
```
The first request is to 0x2000_0000 -- which is seeing the bad address (why?). Subsequently, there's the request to 0x2000_1000 -- which is fatal because it doesn't belong to a mapped address range.

mysoreanoop, 2023-10-27:
[sc_debug.zip](https://github.com/gem5/gem5/files/13185799/sc_debug.zip)

powerjg, 2023-11-27:
Does this need to be fixed before the gem5-23.1 release?

ivanaamit, 2024-01-08:
I'm just checking what is the status of this issue. Is someone currently working on it? Thank you.

closed 2024-02-16 by github-actions[bot]