gem5/gem5 #504: Updating Power Model

mattsinc, 2023-10-25:
This is not a bug directly, but since the Discussions option seems to be used for other purposes, I'm posting it here.

Ultimately, one of the goals my group is working on is updating the power model to design an interface that is easy (easier?) for users to plug in power models for CPUs, GPUs, caches/main memory, or other accelerators.  @pol023 and I have been looking into this, and have a proposal we'd like to run by the community.

Currently, the power model (e.g., https://gem5.googlesource.com/public/gem5/+/refs/heads/develop/configs/example/arm/fs_power.py#54) allows a user to essentially multiply/sum anything a given component accesses as long as there is a stat that goes with it, and as long as the user provides some sort of dynamic power (likewise, static power) to multiply/sum these parts together.  This works ok if the number of components being expressed in a string (e.g., lines 54-55 in the above) are small, but if the number of components are large (e.g., all the different FUs/ROBs/branch predictors/etc. in a CPU), this seems unwieldy as the string on lines 54-55 would need to go on for many lines to incorporate all of these pieces.

What we propose to do is create a configuration file that a component like the above fs_power.py link could use.  Each line in such a configuration file would be a triplet of <componentName, componentDynamicPowerPerAcc, correspondStat>.  Then, we'd change the string multiplying/summing these components together would into a loop sort of like this:

sum = 0.0
for each instance of component (e.g., multiple CPUs)
    for each line $i in config file:
        sum += $i.componentDynamicPowerPerAcc * $i.$correspondStat

This seems like it retains the flexibility of the current system, while also expanding it to make it more orderly and clear for any given component about how to express the information in a logic, systematic manner.  Likewise, for anyone who adds a future component (e.g., accelerator, new memory, etc.) they could express a power model by passing in a dynamic power value for any component within it -- as long as there is a gem5 stat that goes with it.

Note that this proposal does not tie us to Cacti, McPat, or any other specific tool.  It just specifies a way for users to specify how their power model should behave in terms of its parts.

Thoughts?

CC: @BobbyRBruce , @powerjg

powerjg, 2023-10-25:
I get the problem you're trying to solve, but I am quite strongly against another type of configuration file or another domain-specific language.

Why not extend the `PowerModel` instead of `MathExprPowerModel`?

mattsinc, 2023-10-25:
@powerjg : I'm not sure I understand your vision here and how extending PowerModel instead would not devolve into requiring a similar configuration file?