gem5/gem5 #505: Provide a way to replicate tet suite resources storage setup.

trollet91, 2023-10-25:
Enchencement feature:

In our lab, we don't have access to external network everywhere.
Before gem5 v23, we used to duplicate the resources into an internal database.
This way we could run the test suite without accessing outside network. 

Since v23, the resource setup became much much more complex and I didn't find a way to replicate this setup internaly.

Providing an automated way to replicate the resource storage would allow:
- Using gem5 test suite in environment with no external network with miniml effort.
- Orgs can use their own servers to store the resources. Reduce the demand on UCDavis servers.

powerjg, 2023-10-25:
cc: @Harshil2107

Harshil2107, 2023-10-25:
I will work on an util that could copy the database and the downloaded resources to a local directory. I think it is possible but I need to test it to be sure.

I'd like to confirm the following requirements:

1. Maintain a local copy of the database as a JSON file, which will serve as the primary data source.

2. Keep a local copy of resources with URLs, and ensure that the JSON data of these resources references our locally downloaded versions.

Harshil2107, 2023-10-25:
Do you require all resources in the database to be accessed without external connection or only the ones that are used in the tests?

I just want to confirm the use cases.

trollet91, 2023-10-26:
The best flow would be for me:
1- Extract all the resources from the current database server.
2- User manualy moves all the resources from public network to internal network.
3- Re-create the current data base in a server located in the internal network. I don't know how easy it is to automate from a script.
3- Change gem5_default_config.py to point to the new database
[image]

Harshil2107, 2023-10-26:
I created a PR with the script that would create an offline copy of the database. Tell me if I missed anything.

trollet91, 2023-10-26:
Thanks @Harshil2107 , this seems to cover the point 1. I will test it tomorrow.
For the point 3, do you think it's also possible to script it?

Harshil2107, 2023-10-26:
I think it covers it.

This the the high level description of the script
- Get the resource JSON objects and make a `resources.json`
- Download the resources which have a `url` field.
- As gem5 works with local resources, the only requernment is that the `url` should start with `file://` followed by the absolute path of the resource. The script then modifies the `resources.json` to these `file://absolute-path`.

All you would have to do is to update the config to use the JSON file. The config structure for using a JSON file in config is:
```
{
  "url" : JSON file path,
  "isMongo": False,
  }
  ```
  There are multiple ways to do it:
  - Set the GEM5_CONFIG env variable to point to a new config file. This will override the default config.
  
- Have a file named `gem5-config.json` in the current working directory. This will also override the default config.

- Use the GEM5_RESOURCE_JSON env variable to override the default config to use a JSON file or use the GEM5_RESOURCE_JSON_APPEND env variable to add a JSON file to the current config.

Harshil2107, 2024-01-08:
I am closing this issue as the PR related to this issue is merged.

closed 2024-01-08 by Harshil2107

kaddkaka, 2025-08-21:
Hi, I'm new to gem5. I would also like to run it in offline mode. Where do I start?