gem5/gem5 #508: m_atomicLog not getting cleared

mysoreanoop, 2023-10-26:
**Describe the bug**
An assertion in [GPUCoalescer.cc ](https://github.com/gem5/gem5/blob/b6ce2d0db891a2dee0bace10222dec06cb4c175b/src/mem/ruby/system/GPUCoalescer.cc#L635) fails with the below debug log (also attached):
```
152221825000: system.ruby.tcp_cntrl0.coalescer: Got hitCallback for 0x1025F00
152221825000: system.ruby.tcp_cntrl0.coalescer: Responding to 4 packets for addr 0x1025F00; type: ATOMIC_NO_RETURN
152221825000: system.ruby.tcp_cntrl0.coalescer: numAtomicLogEntries() = 4  (This is the failing assertion variable)
```
Maybe related to updates from this commit: https://github.com/gem5/gem5/commit/f6a453362fbc330bda69d187ac3ffed991bee1ef

This is not seen with the latest master/stable version of gem5

**Affects version**
Develop branch: 6ddf8c94ee1368f8a573fc7f782dcf6a8534293b

**gem5 Modifications**
None

**To Reproduce**
1. Compile gem5 with command `scons build/GCN3_X86/gem5.opt`
2. Download [hsti.zip](https://github.com/gem5/gem5/files/13171331/hsti.zip)
3. Run with: `docker run -u $UID:$GID --volume $(pwd):$(pwd) -w $(pwd) gcr.io/gem5-test/gcn-gpu:v22-1  gem5/build/GCN3_X86/gem5.opt -d chai/hsti --debug-file=debug --debug-flags=GPUCoalescer --debug-start=152000000000 gem5/configs/example/apu_se.py --cpu-type=DerivO3CPU --num-cpus=12 -c gem5-resources/src/gpu/chai/HIP-U-gem5/HSTI/bin/hsti.gem5 1>chai/hsti/simout 2>chai/hsti/simerr`

**Terminal Output**
[logs.zip](https://github.com/gem5/gem5/files/13172103/logs.zip)

**Host Operating System**
Ubuntu 20.04, Docker (same OS)

**Additional Information**
Seems like adding a pop for every ATOMIC_NO_RETURN pkt [here](https://github.com/gem5/gem5/blob/f6a453362fbc330bda69d187ac3ffed991bee1ef/src/mem/ruby/system/GPUCoalescer.cc#L573) similar to the case of ATOMIC_RETURN solves the issue for the benchmark I'm running, but perhaps the cleaner solution would be to not record m_atomicLog entry when processing ATOMIC_NO_RETURN.

abmerop, 2023-10-26:
This also happens with #397 applied from my testing, so another fix is needed.  I think the GPUCoalescer solution would be fine.  I don't think the coherence protocol considers return vs. no return and therefore always uses the log.

mattsinc, 2023-10-27:
We could add a flag/field (similar to the GLC/SLC fields we recently added) to indicate if it's a no return?

mysoreanoop, 2023-10-27:
It is already part of RubyRequestType -- so that information is available to Ruby, it's just not using it.

mattsinc, 2023-10-30:
@mysoreanoop : please go ahead and submit a patch then, and we can review this.

CC @DanKouch as #397 is also running into this issue.

powerjg, 2023-10-30:
I mentioned this in another thread, but is this related to the memory leak discussed in #514?

mysoreanoop, 2023-11-09:
Seems like at the cache closest to the core (TCP), the accepted request has a RubyRequestType, and when it passes on to the next cache (TCC), it switches to CoherenceRequestType which does not distinguish between RETURN and NO_RETURN. Should we add it into CoherenceRequestType? If we replace the default atomic type there with two (RETURN and NO_RETURN), things would be cleaner, but would involve changes in files of other protocols using this type.

BujSet, 2023-11-14:
Since #546 was merged into develop, I believe this issue can be closed, right?

closed 2023-11-14 by mysoreanoop