gem5/gem5 #509: [ARM] map() method on Process does not honor cacheable parameter

derchr, 2023-10-26:
This bug report is based on a discussion in the [gem5-users mailing list](https://www.mail-archive.com/gem5-users@gem5.org/msg22012.html).

**Describe the bug**
The map method is a way in gem5-SE mode to map a range of physical addresses directly into the virtual address space of the running SE application.
Besides the `vaddr`, `paddr` and `size`, it is possible to pass in the `cacheable` paramter to specify if accesses to this virtual memory region should be passed to the cache.

In x86 mode, this parameter seems to work fine. However, it can be observed that this is not the case for ARM, where it gets ignored and the cache is accessed nonetheless.

**Affects version**
Current stable and 20f5555f30e13ae0931464f36f32c65336711569 develop.

**gem5 Modifications**
Altered the arm-hello.py script to map the regions. See below for an example.

**To Reproduce**
Steps to reproduce the behavior. Please assume starting from a clean repository:

1. Compile gem5 with command: `scons build/ARM/gem5.debug`
2. Alter the arm-hello.py example script:

```
from gem5.isas import ISA
from gem5.utils.requires import requires
from gem5.resources.resource import BinaryResource
from gem5.components.memory import SingleChannelDDR3_1600
from gem5.components.processors.cpu_types import CPUTypes
from gem5.components.boards.simple_board import SimpleBoard
from gem5.components.cachehierarchies.classic.private_l1_private_l2_cache_hierarchy import PrivateL1PrivateL2CacheHierarchy
from gem5.components.processors.simple_processor import SimpleProcessor
from gem5.simulate.simulator import Simulator

requires(isa_required=ISA.ARM)

cache_hierarchy = PrivateL1PrivateL2CacheHierarchy(
    l1d_size="16kB", l1i_size="16kB", l2_size="256kB"
)

memory = SingleChannelDDR3_1600(size="32MB")
processor = SimpleProcessor(cpu_type=CPUTypes.TIMING, isa=ISA.ARM, num_cores=1)

board = SimpleBoard(
    clk_freq="3GHz",
    processor=processor,
    memory=memory,
    cache_hierarchy=cache_hierarchy,
)

board.set_se_binary_workload(
    BinaryResource("physical")
)

simulator = Simulator(board=board)
simulator._instantiate()

for core in processor.get_cores():
    core.core.workload[0].map(vaddr=0x1000000, paddr=0x123000, size=1024, cacheable=False)

simulator.run()

print(
    "Exiting @ tick {} because {}.".format(
        simulator.get_current_tick(), simulator.get_last_exit_event_cause()
    )
)
```

3. Create a basic C-Application that accesses a fixed virtual address:
```
#include <stdio.h>

int main()
{
    volatile int *address = (int *)0x1000000;
    printf("%d\n", *address);
    *address = 123;
    printf("%d\n", *address);
}
```

4. Compile the program to ARM: `aarch64-linux-gnu-gcc physical.c -o physical -static`
5. Run the simulation:  `build/ARM/gem5.debug --debug-flags=Cache configs/example/gem5_library/arm-uncacheable.py > cache.log`
6. The program works fine. However, in the logs it can be seen that the caches are clearely traversed:
```
22474170: board.cache_hierarchy.l1dcaches: access for ReadReq [123000:123003] miss
22474503: board.cache_hierarchy.l1dcaches: sendMSHRQueuePacket: MSHR ReadReq [123000:123003]
22474503: board.cache_hierarchy.l1dcaches: createMissPacket: created ReadSharedReq [123000:12303f] from ReadReq [123000:123003]
22474503: board.cache_hierarchy.l2caches: access for ReadSharedReq [123000:12303f] miss
22478166: board.cache_hierarchy.l2caches: sendMSHRQueuePacket: MSHR ReadSharedReq [123000:12303f]
22478166: board.cache_hierarchy.l2caches: createMissPacket: created ReadSharedReq [123000:12303f] from ReadSharedReq [123000:12303f]
22547763: board.cache_hierarchy.l2caches: recvTimingResp: Handling response ReadResp [123000:12303f]
22547763: board.cache_hierarchy.l2caches: Block for addr 0x123000 being updated in Cache
22547763: board.cache_hierarchy.l2caches: Block addr 0x123000 (ns) moving from  to state: 6 (E) writable: 1 readable: 1 dirty: 0 prefetched: 0 | tag: 0x48 secure: 0 valid: 1 | set: 0xc0 way: 0
22548429: board.cache_hierarchy.l1dcaches: recvTimingResp: Handling response ReadResp [123000:12303f]
22548429: board.cache_hierarchy.l1dcaches: Block for addr 0x123000 being updated in Cache
22548429: board.cache_hierarchy.l1dcaches: Block addr 0x123000 (ns) moving from  to state: 615f6176 (E) writable: 1 readable: 1 dirty: 0 prefetched: 0 | tag: 0x246 secure: 0 valid: 1 | set: 0 way: 0x3
```

**Expected behavior**
The caches should not be traversed.

**Host Operating System**
Ubuntu 23.04

**Host ISA**
X86

**Compiler used**
gcc (Ubuntu 12.3.0-1ubuntu1~23.04) 12.3.0
aarch64-linux-gnu-gcc (Ubuntu 12.3.0-1ubuntu1~23.04) 12.3.0

giactra, 2023-10-30:
This issue has been fixed on develop and it will be part of v23.1

closed 2023-10-30 by giactra

derchr, 2023-10-31:
Thank you for the quick fix!

giactra, 2023-10-31:
Thanks to you for spotting the issue!