gem5/gem5 #547: Potential bug in MOESI_AMD_Base-dir.sm

mysoreanoop, 2023-11-09:
**Describe the bug**
https://github.com/gem5/gem5/blob/0442c9a88c5e0031d766470c76d1cd8138b313cb/src/mem/ruby/protocol/MOESI_AMD_Base-dir.sm#L1041-L1057
`wd_writeBackData` is an action that is called only after all the dispatched probes have been acknowledged.
`tbe.Dirty` is only set during: DMA writes, GPU write-throughs, or a dirty data probe response from CPU/GPU caches.
The condition here writes the updated data in the TBE (DMA data or GPU WT data, or dirty data some L2 has relinquished and has assumed owned state) to memory. I believe the condition has to be `tbe.Dirty == true` (in line 1042). Please correct me if I'm wrong in my understanding. I tried the changes; at least, I don't see any failures.

**Affects version**
Develop/stable branch

mattsinc, 2023-11-09:
@mysoreanoop : please provide more details.  What are you running and what is failing with the code as is?  I'm not sure I fully understand the issue here.

mysoreanoop, 2023-11-09:
I have no associated failing code. But I am looking into the source code
for modifying it for a different purpose, and this piece of code does not
make sense to me. Do you have a reasoning as to why the condition exists
that way?

mattsinc, 2023-11-09:
In general, I am hesitant to change code if we cannot test it -- it is often easy to convince one's self that Ruby behaves a certain way until actually testing it and showing it does the wrong thing in a given use case.  So while I understand your logic, I am worried changing it will break something you aren't testing or that someone is relying on but you/I aren't thinking about.  I do agree it seems like we'd want tbe.Dirty to be true to write back something, but I haven't looked through every corner case of tbe.Dirty in this file.

@abmerop : thoughts?  It appears you changed this line in March 2022 as part of bfcab1258fa (https://gem5-review.googlesource.com/c/public/gem5/+/57409).

mattsinc, 2023-11-09:
Actually, looking through @abmerop 's commit I linked above, my guess is tbe.Dirty = false here because when the TBE is not dirty, we can use the copy of the data we stored in the TBE when we created it -- i.e., since the data hasn't been updated since, it's "safe" to use that data.  At least that's what it seems like ...

mysoreanoop, 2023-11-09:
We do update tbe.Dirty := false when receiving a WT from GPU or DMA Wr,
which are the two cases when we would need to write to memory right from
the beginning. There's the 3rd chance of receiving a dirty probe (which
means a responding CPU core has now switched over to Owned state from some
variant of Modified state) in which case we'd need to write to memory too
-- this also updates tbe.Dirty when probe responses are processed. There's
also the victim data that needs to be written to memory, but that seems to
be handled differently.

abmerop, 2023-11-10:
In that change I simply copied whatever logic was previously there and replaced the DirectoryMemory writes with an actual memory write.

mattsinc, 2023-11-13:
After reviewing the directory code a lot as part of #540, I believe this is probably a bug.  However, without anything to test it, I am leery of making a change -- else how can we confirm?

mysoreanoop, 2023-11-13:
Thanks for looking into it, and #540 is a very cool update!

I can write a detailed reasoning as to why it makes sense. And I'll try to come up with a microbenchmark to test it.

mattsinc, 2023-11-13:
I do not need a detailed reasoning -- I have already done that by looking at the code.  Unless we have a test, I don't think proceeding makes sense though.

mysoreanoop, 2023-11-15:
Also, while we are at it -- a question about something in the directory:
https://github.com/gem5/gem5/blob/30787b59d4eaccdfaab153a8f031010d5ffa349e/src/mem/ruby/protocol/MOESI_AMD_Base-dir.sm#L1104-L1109

What is double data here? I could not come up with any scenario where there could be two probe responses both being dirty. On the CPU side, at most one CPU can be owned or modified, so would respond with dirty data when probed; all other CPUs would simply acknowledge the probe without sending any dirty data. The GPU does not respond to probes with dirty data even if it's a write-back. Is my understanding correct?

closed 2024-01-04 by github-actions[bot]