gem5/gem5 #548: gem5::IGbE::TxDescCache::pktComplete(): Assertion `ip || ip6' failed with ARM SMMU enabled

sammy17, 2023-11-11:
**Describe the bug**
Following assertion fails when running an ethernet workload between two systems simulated through ARM bigLITTLE setup using https://github.com/gem5/gem5/blob/stable/util/dist/test/test-2nodes-AArch64.sh 
```
7521909181000: system.realview.smmu: [t] requestor resp addr=0x8120e07e size=0x2
7521909181000: system.realview.smmu: [t] responder resp addr=0xffe5807e size=0x2
7521909183000: system.realview.smmu: [t] requestor resp
addr=0x8120e080 size=0x40
7521909183000: system.realview.smmu: [t] responder resp
addr=0xffe58080 size=0x40
7521909184000: system.ethernet.TxDesc: DMA of packet complete
7521909184000: system.ethernet.TxDesc: TxDescriptor data d1:
0xffe5807e d2: 0x200ab100042
7521909184000: system.ethernet.TxDesc: TxDescriptor data d1:
0xffe5807e d2: 0x201ab100042
7521909184000: system.ethernet.TxDesc: Proccesing Non-Ip packet
7521909184000: system.ethernet.TxDesc: Calculating checksums for packet
gem5.opt: src/dev/net/i8254xGBe.cc:1820: void
gem5::IGbE::TxDescCache::pktComplete(): Assertion `ip || ip6' failed.
```
This failure is only observed when ARM SMMU is enabled and the ethernet device in both systems are connected to the SMMU. 
Also, fail is observed both in timing mode and atomic mode. 

**Affects version**
v23.0.1.0

**gem5 Modifications**
https://github.com/gem5/gem5/blob/48a40cf2f5182a82de360b7efa497d82e06b1631/configs/example/arm/devices.py#L461
attach_pci function was changed to add *args, **kwargs and pass **kwargs as below:
```
def attach_pci(self, dev, *args, **kwargs):
        self.realview.attachPciDevice(dev, self.iobus, **kwargs)
```
https://github.com/gem5/gem5/blob/48a40cf2f5182a82de360b7efa497d82e06b1631/configs/example/arm/dist_bigLITTLE.py#L138
addEthernet function was changed to instantiate SMMU and connect the Ethernet device to SMMU as below:
```
def addEthernet(system, options):
        # Create an SMMU but don't connect it to any devices.
        system.realview.attachSmmu([], system.iobus)
        system.realview.smmu.irq_interface_enable = True
        # create NIC
        dev = IGbE_e1000()
        system.attach_pci(dev, dma_ports=[])
        system.ethernet = dev
        # Connect the device's DMA ports to the SMMU
        system.realview.smmu.connect(dev)
```
A manually edited device tree was used to include the SMMU and map the PCI ethernet device to the SMMU/IOMMU. 
Following are the relevant device tree nodes of SMMU and PCI host. 
```
	iommu: smmuv3@2b400000 {
		compatible = "arm,smmu-v3";
		reg = <0x00 0x2b400000 0x00 0x20000>;
		#iommu-cells = <0x01>;
		phandle = <0x11>;
		dma-coherent;
	};

	pci {
		compatible = "pci-host-ecam-generic";
		device_type = "pci";
		#address-cells = <0x03>;
		#size-cells = <0x02>;
		#interrupt-cells = <0x01>;
		reg = <0x00 0x40000000 0x00 0x10000000>;
		ranges = <0x1000000 0x00 0x00 0x00 0x50000000 0x00 0x10000 0x2000000 0x00 0x00 0x40 0x00 0x00 0x40000000>;
		interrupt-map = <0x00 0x00 0x00 0x00 0x01 0x00 0x00 0x00 0x44 0x01 0x800 0x00 0x00 0x00 0x01 0x00 0x00 0x00 0x45 0x01 0x1000 0x00 0x00 0x00 0x01 0x00 0x00 0x00 0x46 0x01 0x1800 0x00 0x00 0x00 0x01 0x00 0x00 0x00 0x47 0x01>;
		interrupt-map-mask = <0x1800 0x00 0x00 0x00>;
		iommu-map = <0x10 &iommu 0x0 0x1>;
		dma-coherent;
	};
```
I added SMMU addr range to the iobridge addr range at 
https://github.com/gem5/gem5/blob/e9da8d67bdbb23fbd0578a379f08f42bce50121d/configs/example/arm/devices.py#L436C26-L436C26 as below:
```
 self.iobridge = Bridge(delay="50ns", ranges=[AddrRange(0x2b400000, size="128KiB")])
```
Also, changed the attachOnChipIO function to include this range at https://github.com/gem5/gem5/blob/e9da8d67bdbb23fbd0578a379f08f42bce50121d/src/dev/arm/RealView.py#L879C13-L879C13 to below:
```
            bridge.ranges = bridge.ranges + self._off_chip_ranges
```

**To Reproduce**
1. Make the above mentioned changes and rebuild gem5 
2. Edit the util/dist/test/test-2nodes-AArch64.sh to remove the bootscript, enable timing mode and use the custom dtb (I have attached the dtb and edited script in the issue https://github.com/gem5/gem5/issues/374)
3. Install iperf3 in the Linux disk image as this is used to run an ethernet workload
4. Start the simulation using test-2nodes-AArch64.sh
5. Setup the ethernet in both systems and run iperf3 as below:
```
System 1:
/sbin/ifconfig eth0 hw ether 00:90:00:00:00:02
/sbin/ifconfig eth0 192.168.0.2 netmask 255.255.255.0 up
iperf3 -s -4 

System 2:
/sbin/ifconfig eth0 hw ether 00:90:00:00:00:04
/sbin/ifconfig eth0 192.168.0.4 netmask 255.255.255.0 up
iperf3 -c 192.168.0.2 -4 -d -n 4M 
```

**Expected behavior**
iperf should complete without crashing the simulation with above assert failure. 

**Host Operating System**
Ubuntu 22.04

**Host ISA**
 X86

**Compiler used**
gcc 11.3.0

**Additional information**
Linux kernel 5.15 was run on both systems with default gem5 linux config file.

sammy17, 2023-11-13:
Some additional information after further debugging:
It seems like the crash is happening due to Ethernet packets with random header data. 
Ethernet packets that trigger the assertion has unknown Ethernet payload type. 
In two crashes that I analyzed, I saw different data in the Ethernet payload type (0x7e83 and 0x17ec). 
Ethernet payload type is checked at: https://github.com/gem5/gem5/blob/e9da8d67bdbb23fbd0578a379f08f42bce50121d/src/base/inet.hh#L371C14-L371C14

Therefore, I suspect this issue might be due to some data corruption that happens through the SMMU (since the issue is only seen with it enabled). 
I would really appreciate any help on resolving this or debugging this further.

giactra, 2023-11-13:
Have you tried adding a watchpoint on the pktPtr to check when it gets modified? Please make sure to run with gem5.debug.. I have the suspicion something wrong might be happening when popping the sender state (and safe_cast does runtime checking when in debug mode)

sammy17, 2023-11-15:
@giactra Thanks for the suggestion.
I was able to figure out where the pktPtr type change to an incorrect value. 
Type of the (IP) packet changed to an incorrect value after the below memcpy. 
https://github.com/gem5/gem5/blob/d0d3c74ce00f6d852a39060f3b7ce083887b1cfd/src/dev/net/i8254xGBe.cc#L1683C13-L1683C13

I did not use gdb as this issue can only be triggered when running an Ethernet workload between two simulations. I was not sure how to run gdb for this kind of a setup. I just printed the pktPtr type at different locations. 

I can see that tsoHeader is initialized with a DMA read that goes through the SMMU - https://github.com/gem5/gem5/blob/d0d3c74ce00f6d852a39060f3b7ce083887b1cfd/src/dev/net/i8254xGBe.cc#L1588C10-L1588C10

**Is there any chance that DMA read data for tsoHeader gets corrupted/overwritten at some point?**

sammy17, 2023-11-15:
Also, I could not reproduce the issue in debug mode. 
However, debug mode with timing enabled was really slow so there is a slight chance that I did not run the ethernet workload long enough to trigger the issue.

sammy17, 2023-11-20:
Thanks for the suggestions @giactra !

1. I removed the dma_coherent flag from the DTB SMMU entry and the assert fail was still reproducible. When I removed the dma_coherent from both the SMMU and PCI Host, an earlier assertion failed with below error: 

```
src/mem/cache/cache.cc:1211: panic: panic condition !invalidate && !pkt->hasSharers() occurred: system.iocache is passing a Modified line through WriteReq [81d6100c:81d6100f] UC, but keeping the block
```
I believe this is expected as PCI Host and SMMU is now connected to a coherent bus. 

To verify 2 and 3, I may need to instrument the Linux kernel so I will keep it for last (let me know if you have any alternative ways or if I misunderstood what you said). 

**I have few other updates:**
I noticed that when the assert in question fails, there is always 2 SMMU device interface TLB hits. 
To confirm this, I disabled the device interface TLB in SMMU and the issue was gone!
So it seems like the issue is coming from device interface TLB, I will debug it further. 

Unfortunately, I am facing another issue after applying above workaround. 
I encounter below assert failures but this error is random and has a less probability to trigger than the previous assert fail (ip || ip6).
```
8923844909000: system.ethernet.dma: TransmitList: 2, retryPending: 0
8923844910000: system.ethernet.dma: Trying to send WriteReq addr 0xfff2b480
8923844910000: system.realview.smmu.device_interfaces: [t] req from system.ethernet.dma addr=0xfff2b480 size=0x40
8923844910000: system.ethernet.dma: -- Failed, waiting for retry
8923844910000: system.ethernet.dma: TransmitList: 2, retryPending: 1
8923844913000: system.realview.smmu.device_interfaces.port: IdReg:  p=0x5581cd1cc1e0 oid=9
8923844913000: system.realview.smmu.device_interfaces.port: 4kHold: p=0x5581cd1cc1e0 a4k=0xfff2b000 Q: p=0x5581cced5d60 a4k=0xfff2b000
8923844913000: system.realview.smmu.device_interfaces.port: 4kHold: p=0x5581cd1cc1e0 a4k=0xfff2b000 WAIT on p=0x5581cced5d60 a4k=0xfff2b000
8923844915000: system.realview.smmu.device_interfaces.port: Config hit sid=0 ssid=0 ttb=0x8195a000 asid=0x1
8923844916000: system.realview.smmu.device_interfaces.port: Fetching S1 L1 PTE from pa=0x8195a018
8923844917000: system.realview.smmu: [t] requestor req  addr=0x8195a018 size=0x8
8923844917000: system.realview.smmu.device_interfaces: sched responder retry
8923844918000: system.ethernet.dma: Trying to send WriteReq addr 0xfff2b480
8923844918000: system.realview.smmu.device_interfaces: [t] req from system.ethernet.dma addr=0xfff2b480 size=0x40
8923844918000: system.ethernet.dma: -- Failed, waiting for retry
8923844918000: system.ethernet.dma: TransmitList: 2, retryPending: 1
gem5.opt: src/mem/packet.hh:752: void gem5::Packet::setSatisfied(): Assertion `!flags.isSet(SATISFIED)' failed.
Program aborted at tick 8923844924000
```
Let me know if you have any ideas on whats going wrong here.

sammy17, 2023-11-21:
I figured out the root cause for the original issue. 
It seems like TLB has a stale translation at some point and it returns an incorrect physical address for DMA read. 
This obviously results in the ethernet device reading some junk data and giving the original assertion failure.
It also makes sense that the issue is not seen when the TLBs are disabled. 
So the issue seems to be with the TLB invalidation.

giactra, 2023-11-21:
Thank you so much @sammy17 for debugging this problem... We do perform TLB invalidation at the device interface. I am wondering if you are seeing any invalidation command issued to the command queue https://github.com/gem5/gem5/blob/stable/src/dev/arm/smmu_v3.cc#L389... If not it could be because the kernel is assuming it can invalidate the SMMU via DVM messages (TLBI broadcasting) which is something we do not support in gem5 at the moment...

This shouldn't be the case anyway as IIRC we do flag absence of DVM support via some SMMU ID register

sammy17, 2023-11-21:
Below is a detailed description of the issue, I am not sure whether this is related to the invalidation or not.
I have enabled only the device interface TLB for this experiment, all other TLBs in the SMMU are disabled.  

1. Ethernet device read a packet located at vaddr=0xffe56000 which maps to the physical address 0x81ea8000. This DMA transfer completes without any issue. 
2. Later Ethernet device read another packet located at vaddr=0xffe56880. Translation of this page is now available in the device interface TLB so there is a hit and the address is translated as 0x81ea8880. This functionality seems correct. But the issue is IO page table is somehow changed and 0xffe56000 no longer maps to 0x81ea8000. 

So I would expect an TLB invalidation for vaddr=0xffe56000 between step 1 and 2 but I don't see any invalidation. 
Also, usually Linux kernel does not allocate the same virtual address until previous mapping is invalidated. 

So two things can be the cause:

1. SMMU is somehow not recognizing the TLB invalidation command send by the kernel (this is unlikely as I don't see any relevant unimplemented commands during step 1 and 2) or kernel is not sending the invalidation request for some reason. 
2. SMMU or ethernet device is reading an incorrect vaddr (what you initially mentioned). 

Do you know whether SMMU ID register indicate the existence of a TLB? 
I could not find this in the ARM SMMU spec.
I am trying to think any reason for Linux to not send an invalidation command.

ivanaamit, 2024-01-24:
Hi @giactra and @sammy17, I noticed there hasn't been any action on this issue for the past two months. Any chance you could give us a status update or share your thoughts on it? Thank you.

sammy17, 2024-02-15:
This issue was fixed, I believe the issue was that I have been using a device tree file with dma-coherent not set in the SMMU node, leading to coherency issues. Once that is fixed, SMMU worked as expected with the changes that I have mentioned in the first comment of this post.

closed 2024-02-15 by sammy17