gem5/gem5 #550: gem5's Linux printk helpers no longer work when using Linux kernel v5.18+ in the guest system

hnpl, 2023-11-12:
**Describe the bug**
Mismatch between the gem5's DmesgRingBuffer's `struct metadata` and the corresponding `struct prb_desc_ring` in Linux kernel v5.18+.

**Affects version**
Current develop (https://github.com/gem5/gem5/tree/1204267fd8dc2076b4f3b6288a0040116017193d).

**gem5 Modifications**
N/A

**To Reproduce**
Induce a kernel panic where a PCEvent is added for the `panic()` function in the guest.

**Terminal Output**

```shell
458840683000: global: PC based event serviced at 0xffffffff8088c8b4: Kernel panic in simulated kernel                            
src/kern/linux/events.cc:74: info: Dumping kernel dmesg buffer to board.workload.dmesg...                                        
src/mem/port_proxy.hh:185: fatal: readBlob(0x11, ...) failed  
```

**Additional information**
gem5's DmesgRingBuffer's `struct metadata` is used to interpret the `struct prb_desc_ring` in Linux kernel.

gem5's DmesgRingBuffer's `struct metadata` has 5 fields and is 40-byte long. [1]
In Linux kernel v5.10-v5.17, the corresponding `struct prb_desc_ring` in Linux kernel has 5 fields and is 40-byte long. [2]
However, since v5.18, the struct has a new field, which makes the struct 48-byte long. [3]

[1] https://github.com/gem5/gem5/blob/1204267fd8dc2076b4f3b6288a0040116017193d/src/kern/linux/helpers.cc#L288
[2] https://elixir.bootlin.com/linux/v5.17.15/source/kernel/printk/printk_ringbuffer.h#L71
[3] https://elixir.bootlin.com/linux/v5.18.19/source/kernel/printk/printk_ringbuffer.h#L71

hnpl, 2023-11-14:
@rjc-arch can you take a look at this?

rjc-arch, 2023-11-24:
Quick update: I have a proposed fix for this bug. I'm just finishing off testing and review, and hope to post a PR soon.

closed 2023-12-13 by Harshil2107