gem5/gem5 #558: Create gem5 v23.1 Staging branch

BobbyRBruce, 2023-11-13:
As discussed and agreed upon during our November Developer meeting (https://github.com/orgs/gem5/discussions/482), we have scheduled the creation of our staging branch from `develop` for December 1st. This means **gem5 developers have until December 1st to have their contributions merged into `develop` for inclusion in version v23.1**. Contributions made after this deadline will have to wait for the v24.0 release.

Once created the staging branch will be tested rigorously and our [release procedures](https://www.gem5.org/documentation/general_docs/development/release_procedures/) followed. The branch shall exist for at least 2 weeks in order to run tests and give the community time to inspect the branch. Once the staging branch is found to be ready for release it shall be merged into the `stable` branch thus officially releasing gem5 v23.1. Assuming a typical 2-week staging branch period this release to occur on December 15th.

Contributions to the staging branch via pull request are permitted but, in the interests of ensuring new bugs are not introduced, good justification should be given as to why the change cannot wait for the following release (v24.0). Usually contributions to the staging branch are bug fixes or contributions that have no chance of changing software functionality (e.g. documentation updates, format fixes, etc.). New features or significant enhancements to gem5 should continue to be made to the `develop` branch during this time. The staging branch will be periodically merged into the `develop` branch to ensure contributions made to the staging branch are reflecting in the `develop` branch,

I have highlighted the following **must** be included in v23.1 and will therefore be prioritized for review and inclusion into the staging branch:

- [ ] All tests (CI, Daily, Weekly, and Compiler) to be passing.
- [ ] The gem5 Resources "Suites" to be fully incorporated (discussion which includes "Suites" can be found here: https://github.com/orgs/gem5/discussions/201).
- [ ] Support for classic prefetchers with Ruby controllers (#502).
- [ ] Remove `get_runtime_isa()` (#241).
- [ ] Add KConfigs (#69)

If there is anything I have missed, or you create a new PR you believe needs to be prioritized for inclusion in v23.1, please reply to this thread to let me know.

abmerop, 2023-11-13:
Not sure if resources should be discussed here, but I'd like to upload the kernel for this new disk image as a resource: https://github.com/gem5/gem5-resources/pull/12 .  Alternately I can attempt to figure out how to copy from the disk image out using packer to obtain it that way.

BobbyRBruce, 2023-11-13:
> Not sure if resources should be discussed here, but I'd like to upload the kernel for this new disk image as a resource: [gem5/gem5-resources#12](https://github.com/gem5/gem5-resources/pull/12) . Alternately I can attempt to figure out how to copy from the disk image out using packer to obtain it that way.

Resources can be added at any time. They doesn't need to be in prior to the staging branch, or even prior to the release. However, I've noted down we should look into this (I'm quite bad at checking in on the gem5-resources PRs).

Harshil2107, 2023-11-17:
Adding #484 in to-dos for 23.1. I edited the original commit message.

Harshil2107, 2023-11-28:
I edited the original comment and deleted my comment.

abmerop, 2023-12-06:
Hi all,  I believe the staging branch hasn't been created yet, is that correct?  I will guess it is due to the tests failing.  Let me know if I can help out, in particular any X86 / GPU related failures

BobbyRBruce, 2023-12-06:
> Hi all, I believe the staging branch hasn't been created yet, is that correct? I will guess it is due to the tests failing. Let me know if I can help out, in particular any X86 / GPU related failures

Just created it! https://github.com/gem5/gem5/tree/release-staging-v23-1

There was a small delay (as always), but it's created now. I'm going to create a separate Issue to track the health of the staging branch and what may need cherry-picked.

closed 2023-12-06 by BobbyRBruce