gem5/gem5 #574: [bug] Segmentation Fault in fc Feature and MIOpen Error in convolution with GCN3_X86 in DNNMark which is in gem5-resources

EricChenSudo, 2023-11-19:
I have been working on implementing the 'fc' (fully connected) and 'convolution' functionalities of DNNMark using the GCN3_X86 architecture, as instructed in the README. However, I am encountering some issues:

1. Fully Connected (fc) Functionality Issue:

When executing the 'fc' feature, the program attempts to access memory outside the allocated virtual memory, resulting in a segmentation fault. This happens without any modifications to the original code or configuration.

2. Convolution Feature Issue:

On the other hand, while trying to run the 'convolution' feature, I encounter an MIOpen Error. Again, this occurs without any changes made to the code or default settings.
I haven't modified any content or settings in the simulation process. Could you please provide some guidance on how to resolve these issues? Any help or suggestions would be greatly appreciated.

Thank you for your time and assistance.

[image]

mattsinc, 2023-11-19:
This bug has been reported before here: https://www.mail-archive.com/gem5-dev@gem5.org/msg44557.html, however annoyingly it appears the email archive is not showing the additional emails on this chain that discuss more important details.  Not sure what is going on with the mail archive ...

Here is what I suggested in the end for Conv on that chain:

---

Ok I've found a fix to this (i.e., that works on a real AMD GPU) although I don't have a perfect understanding of why it is required.  The bottom line is that the num_outputs and c values in the configuration appear to need to be the same for convolution layers -- not exactly sure why.

But the implication of that is either:

- change https://gem5.googlesource.com/public/gem5-resources/+/refs/heads/develop/src/gpu/DNNMark/config_example/conv_config.dnnmark#7 to 32
- or change https://gem5.googlesource.com/public/gem5-resources/+/refs/heads/develop/src/gpu/DNNMark/config_example/conv_config.dnnmark#12 to 3 and leave line 7 as is

Either change results in the X and W values being the same size for the convolution.  I do not understand the MIOpen code well enough to appreciate the implications of this, but hopefully this is helpful and you can check if it runs in gem5 now.

Sidenote: if you want to see the details, you can set this:

export MIOPEN_ENABLE_LOGGING=1

This will produce a lot of output when you run the benchmark, which is how I saw that X (really Y, as it seems convolution is assuming X and Y are flipped since it's in transposed mode) and W were being set to 32 and 3.  It is also possible that the issue here is that the MIOpen code is assuming that the matrix is transposed and the original num_outputs and c values would work if not transposed, but again I don't quite understand the MIOpen code well enough to reason through this.

Finally, I strongly recommend you reduce N anyways to get gem5 to run this in a reasonable amount of time.  It took 7.4 seconds to run the provided config (with my change above to set num_ouputs to 3) on my real GPU, which is way longer than gem5 can simulate in a reasonable time.

---

However, even after making this change, you will find a new bug later on in the program:

...
build/GCN3_X86/arch/x86/faults.cc:166: panic: Tried to read unmapped address 0x20.
PC: (0x7ffff30ee5fc=>0x7ffff30ee5ff).(0=>1), Instr:   MOV_R_M : ld   rdi, DS:[rdi]
Memory Usage: 20185460 KBytes
Program aborted at tick 173459458000

This "second" bug in some later GPU kernel is what you'll see after making the above change.  It is a very common gem5 failure with the GPU model.  Unfortunately, it's sort of the equivalent of a segfault in a program running on a real machine -- this failure is just saying that the address the instruction is trying to access is not valid.  Which makes sense because 0x20 does not look like a valid address.  However, in terms of how to fix it, there is not an easy/obvious fix.  What you would need to do is get a trace (e.g., using gem5's debug flags) for the address in question, then follow the bad address backwards -- so at the end of the trace is where 0x20 appears, you would need to figure out where 0x20 is originating from.  I don't have time to take that task on right now, but if you are able to start digging through it, I can try to provide feedback.  And of course it would be wonderful if you push any patches required to fix this problem to the public code base!

However, even up to ROCm 5.4, I have found that Conv has this problem on real AMD GPUs.  So I suspect the ultimate root issue is that ROCm does not properly support Conv.

mattsinc, 2023-11-19:
For FC, please provide @abmerop and I more information. What exact sequence of commands did you run on the command line?  And what is the exact segfault failure gem5 produces?

EricChenSudo, 2023-11-20:
Thank you for your response. I am more concerned about the 'fc' (fully connected) part compared to 'conv'. Below are the complete commands I used to execute the fc simulation (essentially just following the instructions in the README):

cd src/gpu/DNNMark
docker run --rm -v ${PWD}:${PWD} -w ${PWD} -u $UID:$GID gcr.io/gem5-test/gcn-gpu:v22-1 ./setup.sh HIP
docker run --rm -v ${PWD}:${PWD} -w ${PWD}/build -u $UID:$GID gcr.io/gem5-test/gcn-gpu:v22-1 make

cd src/gpu/DNNMark
docker run --rm -v ${PWD}:${PWD} -v${PWD}/cachefiles:/root/.cache/miopen/2.9.0 -w ${PWD} gcr.io/gem5-test/gcn-gpu:v22-1 python3 generate_cachefiles.py cachefiles.csv

cd src/gpu/DNNMark
g++ -std=c++0x generate_rand_data.cpp -o generate_rand_data
./generate_rand_data

# Working directory is your gem5 directory
docker run --rm -v ${PWD}:${PWD} -w ${PWD} -u $UID:$GID gcr.io/gem5-test/gcn-gpu:v22-1 scons -sQ -j$(nproc) build/GCN3_X86/gem5.opt

# I replaced fwd softmax with fwd fc to simulate fc
docker run --rm -v ${PWD}:${PWD} -v ${PWD}/gem5-resources/src/gpu/DNNMark/cachefiles:/root/.cache/miopen/2.9.0 -w ${PWD} gcr.io/gem5-test/gcn-gpu:v22-1 gem5/build/GCN3_X86/gem5.opt gem5/configs/example/apu_se.py -n30 --benchmark-root=gem5-resources/src/gpu/DNNMark/build/benchmarks/test_fwd_fc -cdnnmark_test_fwd_fc --options="-config gem5-resources/src/gpu/DNNMark/config_example/fc_config.dnnmark -mmap gem5-resources/src/gpu/DNNMark/mmap.bin"

Initially, I encountered an issue with insufficient physical memory, so I modified options.py to change the default physical memory to 4GB. Additionally, to achieve a reasonable simulation time, I modified the configuration as follows:
[image]

However, the simulation still took a very long time, over 4 hours, and ended with a Segmentation Fault.

I would like to ask if the fc simulation can be successfully executed in your environment. What kind of hardware simulation was it performed on? And if it can be executed completely, what does the configuration look like?

Thank you for your time and assistance.

mattsinc, 2023-11-27:
ok, it looks like you did make changes then.  I will not have time to look into FC before the ASPLOS deadline, but will try to look into running FC on my end next weekend.

For what it's worth 4 hours is not too long for gem5 in my experience.  Anything less than 8 hours is pretty short in my experience (and likely a sign that it's a small benchmark).  The segfault is the bigger problem.  But like I said above, it's just a generic failure (like a segfault) that could originate from many things.  So one would have to get a trace and dig in.

EricChenSudo, 2023-11-30:
Thank you for your response.

I understand the time constraints and appreciate your willingness to look into the Fully Connected (FC) layer simulation issue next weekend.

To clarify, initially, I attempted to run the simulation without any modifications, but unfortunately, I encountered errors related to physical memory allocation, which prevented the simulation from completing successfully. This was the primary reason I considered making changes to the address mapping configuration.

I’m aware that gem5 simulations can be time-intensive, and I agree that the segmentation fault is of greater concern. The intention behind my modifications was to troubleshoot and potentially circumvent the memory allocation issues I was facing.

I am keen on finding a successful method to simulate the FC layer and would greatly value any guidance or suggestions you may have. If there are specific traces or logs that would be helpful for diagnosing the issue, please let me know, and I will provide them.

Following is the error output I facing right now (after fixing the size of physical memory and workload virtual memory):
[image]

Thank you again for your time and assistance.

EricChenSudo, 2023-12-23:
Hello authors,

I was wondering if you have tested whether the current fully connected (fc) functionality is now fully operational? If it is not functioning, could you please inform me if the issue has been resolved? Thank you for your assistance.

mattsinc, 2024-01-02:
Sorry, I have not had a chance to look into this yet.  Are you able to get a trace and help?  If you require me to do everything, it will be slower.  If you are able to help debug, I can help here.

abmerop, 2024-01-22:
This error does not appear to happen in VEGA_X86 build.  FC does not run to completion, but there is a separate issue.  Can you confirm you do not see the unmapped address error in VEGA_X86?  We are planning to deprecate and remove GCN3 so I am not going to spend any time fixing issues related to GCN3.

EricChenSudo, 2024-01-26:
Thank you for your response. I have a question regarding the execution of VEGA_X86: Are the steps for running VEGA_X86 the same as for GCN3? I noticed that the instructions in DNNmark are specifically for GCN3. Could you guide me on how to proceed with running VEGA_X86 on DNNmark with gem5? Any assistance or direction would be greatly appreciated

mattsinc, 2024-01-26:
Thanks @abmerop for digging into this a bit more!  @EricChenSudo : we have been updating the directions here: https://github.com/gem5/gem5-resources/pull/22

In terms of changes, compiling VEGA_X86 instead of GCN3_X86 is the main/first one.  You would then replace GCN3_X86 with VEGA_X86.  The second thing is you'll need to generate the cachefile for gfx902 (assuming you want an APU) or gfx900 (if you want a dGPU).  See here: https://github.com/gem5/gem5-resources/pull/22/files#diff-0aaab3f01d1f9655d35a1dc32b57815f294eebfd185c2f88ca3a80c961f60594

ivanaamit, 2024-02-12:
From my understanding, this is not a bug. I am closing the issue.

closed 2024-02-12 by ivanaamit