gem5/gem5 #578: `Abort due to deadlock` encountered when running gpu-random-test-wbL2 [GPU Daily Test Failure]

BobbyRBruce, 2023-11-20:
**Describe the bug**

The Daily `testlib-long-tests (gpu)` tests have been failing: https://github.com/gem5/gem5/actions/runs/6912511984/job/18808339821

The tests that fail are `ruby-gpu-random-test-wbL2-nightly-VEGA_X86-x86_64-opt` and `ruby-gpu-random-test-wbL2-perCheckin-VEGA_X86-x86_64-opt`. 

They both fail with exactly the same panic:

```sh
src/mem/ruby/system/GPUCoalescer.cc:314: panic: Aborting due to deadlock!
```

**Revisions affected and candidate breaking commits**

This was seen on the develop branch. The earliest known commit to have this bug is db6a8697864564aab1a11081edc05152c0478f83, found in this failing Daily Test: https://github.com/gem5/gem5/actions/runs/6901014917/job/18783093263.

The last commit known to not to contain this bug is ceabe86b311e3bdd7603b4e4631d2ed18365a5bd.

Based on a basic `git rev-log` of this information (`git rev-list --oneline ceabe86..db6a869`) the candidate breaking commits are:

* db6a869786 mem-cache: Prefetchers Improvements (#564)
* 4ca2efac16 mem-ruby: AtomicNoReturn should check comp_anr instead of comp_wu (#545)
* 4965367724 mem-ruby, gpu-compute: fix SQC/TCP requests to same line (#540)
* bfe899e48e stdlib, resources: Update JSON data in workload (#532)
* 047a494c2b mem-cache: Optimize strided prefetcher address generation
* 2abd65c270 mem: added distance parameter to stride prefetcher
* c3326c78e6 mem-ruby, gpu-compute: fix SQC/TCP requests to same line
* 065ddf759f mem-ruby, gpu-compute: fix bug with GPU bypassing loads
* 38045d7a25 mem-cache: Added clean eviction check for prefetchers.
* 6416304e07 mem-cache: Update default prefetch options.
* 8598764a03 mem-cache: Squash prefetch queue entries by block address.
* 50c9cbf613 stdlib, resources: Fixed deprecation warning

**To Reproduce**

```sh
scons build/VEGA_X86/gem5.opt -j`nproc`
./build/VEGA_X86/gem5.opt configs/example/ruby_gpu_random_test.py --WB_L2 --test-length 50000 --num-dmas 0
```

**Terminal Output**

Here is a copy of the **stderr** (note: stdout not shown):

```shell
warn: system.ruby.network adopting orphan SimObject param 'ext_links'
warn: system.ruby.network adopting orphan SimObject param 'int_links'
warn: rounding error > tolerance
    1.250000 rounded to 1
warn: rounding error > tolerance
    1.250000 rounded to 1
warn: rounding error > tolerance
    1.250000 rounded to 1
warn: rounding error > tolerance
    1.250000 rounded to 1
warn: rounding error > tolerance
    1.250000 rounded to 1
warn: rounding error > tolerance
    1.250000 rounded to 1
warn: rounding error > tolerance
    1.250000 rounded to 1
warn: rounding error > tolerance
    1.250000 rounded to 1
src/mem/dram_interface.cc:690: warn: DRAM device capacity (8192 Mbytes) does not match the address range assigned (1024 Mbytes)
src/cpu/testers/gpu_ruby_test/protocol_tester.cc:140: warn: Data race check is not enabled
src/cpu/testers/gpu_ruby_test/protocol_tester.cc:150: warn: If `random_seed == 0` (or `random_seed` is unset) ProtocolTester does not seed the RNG. This will NOT result in the RNG generating different results each run. In this case the RNG is seeded by a default value. This differs from behavior in previous versions of gem5. Setting `random_seed` to a non-zero value is strongly recommended.
src/base/statistics.hh:279: warn: One of the stats is a legacy stat. Legacy stat is a stat that does not belong to any statistics::Group. Legacy stat is deprecated.
src/sim/simulate.cc:194: info: Entering event queue @ 0.  Starting simulation...
src/mem/ruby/system/GPUCoalescer.cc:312: warn: GPUCoalescer 0 Possible deadlock detected!
Printing out 1 outstanding requests in the coalesced table
	Addr: [0x4fc80, line 0x4fc80]
	Instruction sequence number: 12
		Type: ATOMIC_RETURN
		Number of associated packets: 1
		Issue time: 1367
		Difference from current tick: 998634
Listing pending packets from 0 instructions
src/mem/ruby/system/GPUCoalescer.cc:314: panic: Aborting due to deadlock!
Memory Usage: 1376440 KBytes
Program aborted at tick 1000001
--- BEGIN LIBC BACKTRACE ---
/__w/gem5/gem5/build/VEGA_X86/gem5.opt(_ZN4gem515print_backtraceEv+0x30)[0x56362f91b440]
/__w/gem5/gem5/build/VEGA_X86/gem5.opt(_ZN4gem512abortHandlerEi+0x4c)[0x56362f94762c]
/lib/x86_64-linux-gnu/libc.so.6(+0x42520)[0x7f1fd6231520]
/lib/x86_64-linux-gnu/libc.so.6(pthread_kill+0x12c)[0x7f1fd6285a7c]
/lib/x86_64-linux-gnu/libc.so.6(raise+0x16)[0x7f1fd6231476]
/lib/x86_64-linux-gnu/libc.so.6(abort+0xd3)[0x7f1fd62177f3]
/__w/gem5/gem5/build/VEGA_X86/gem5.opt(+0xd8b4c5)[0x56362f89c4c5]
/__w/gem5/gem5/build/VEGA_X86/gem5.opt(_ZN4gem54ruby12GPUCoalescer6wakeupEv+0x27f)[0x5636304e59cf]
/__w/gem5/gem5/build/VEGA_X86/gem5.opt(_ZN4gem510EventQueue10serviceOneEv+0xc2)[0x56362f933412]
/__w/gem5/gem5/build/VEGA_X86/gem5.opt(_ZN4gem59doSimLoopEPNS_10EventQueueE+0x68)[0x56362f96d728]
/__w/gem5/gem5/build/VEGA_X86/gem5.opt(_ZN4gem58simulateEm+0x27b)[0x56362f96dd1b]
/__w/gem5/gem5/build/VEGA_X86/gem5.opt(+0x257ff90)[0x563631090f90]
/__w/gem5/gem5/build/VEGA_X86/gem5.opt(+0xd69634)[0x56362f87a634]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x12b6d3)[0x7f1fd71776d3]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyObject_Call+0x5c)[0x7f1fd713322c]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x4b26)[0x7f1fd70c2766]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c681f)[0x7f1fd721281f]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x9d78)[0x7f1fd70c79b8]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c681f)[0x7f1fd721281f]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(PyEval_EvalCode+0xbe)[0x7f1fd720d94e]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c1edd)[0x7f1fd720dedd]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x12eb13)[0x7f1fd717ab13]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x69ee)[0x7f1fd70c462e]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c681f)[0x7f1fd721281f]
/__w/gem5/gem5/build/VEGA_X86/gem5.opt(+0xe38677)[0x56362f949677]
/__w/gem5/gem5/build/VEGA_X86/gem5.opt(main+0x213)[0x56362f7ebf93]
/lib/x86_64-linux-gnu/libc.so.6(+0x29d90)[0x7f1fd6218d90]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80)[0x7f1fd6218e40]
/__w/gem5/gem5/build/VEGA_X86/gem5.opt(_start+0x25)[0x56362f86abb5]
--- END LIBC BACKTRACE ---
For more info on how to address this issue, please visit https://www.gem5.org/documentation/general_docs/common-errors/ 
```

**Host Operating System**

This was run in a docker image (ghcr.io/gem5/ubuntu-22.04_all-dependencies:latest) on top of an Ubuntu 22.04 OS.

**Host ISA**

X86

BobbyRBruce, 2023-11-20:
@mattsinc @abmerop : Bringing this to your attention.

mattsinc, 2023-11-22:
Thanks for taking the lead on this @abmerop

BobbyRBruce, 2023-11-27:
The test appears to be passing now thanks to #585. Closing.