gem5/gem5 #580: `Assertion 'flags.isSet(STATIC_DATA|DYNAMIC_DATA)' failed` encountered when running "configs/example/gem5_library/riscv-ubuntu-run.py" [GPU Daily Test Failure]

BobbyRBruce, 2023-11-20:
**Describe the bug**

Running gem5 using the [configs/example/gem5_library/riscv-ubuntu-run.py](https://github.com/gem5/gem5/blob/3896673ddc943cdf2157651d090920da466bc6a7/configs/example/gem5_library/riscv-ubuntu-run.py) configuration file produces the following error:

```sh
gem5.opt: src/mem/packet.hh:1236: const T* gem5::Packet::getConstPtr() const [with T = unsigned char]: Assertion `flags.isSet(STATIC_DATA|DYNAMIC_DATA)' failed.
```

**Revisions affected and candidate breaking commits**

This was seen on the develop branch. The earliest known commit to have this bug is db6a8697864564aab1a11081edc05152c0478f83, found in this failing Daily Test: https://github.com/gem5/gem5/actions/runs/6901014917/job/18783090892.

The last commit known to not to contain this bug is ceabe86b311e3bdd7603b4e4631d2ed18365a5bd.

Based on a basic `git rev-log` of this information (`git rev-list --oneline ceabe86..db6a869`) the candidate breaking commits are:

* db6a869786 mem-cache: Prefetchers Improvements (#564)
* 4ca2efac16 mem-ruby: AtomicNoReturn should check comp_anr instead of comp_wu (#545)
* 4965367724 mem-ruby, gpu-compute: fix SQC/TCP requests to same line (#540)
* bfe899e48e stdlib, resources: Update JSON data in workload (#532)
* 047a494c2b mem-cache: Optimize strided prefetcher address generation
* 2abd65c270 mem: added distance parameter to stride prefetcher
* c3326c78e6 mem-ruby, gpu-compute: fix SQC/TCP requests to same line
* 065ddf759f mem-ruby, gpu-compute: fix bug with GPU bypassing loads
* 38045d7a25 mem-cache: Added clean eviction check for prefetchers.
* 6416304e07 mem-cache: Update default prefetch options.
* 8598764a03 mem-cache: Squash prefetch queue entries by block address.
* 50c9cbf613 stdlib, resources: Fixed deprecation warning

**To Reproduce**

```sh
scons build/ALL/gem5.opt -j`nproc`
./build/ALL/gem5.opt configs/example/gem5_library/riscv-ubuntu-run.py
```

**Terminal Output**

```sh
gem5.opt: src/mem/packet.hh:1236: const T* gem5::Packet::getConstPtr() const [with T = unsigned char]: Assertion `flags.isSet(STATIC_DATA|DYNAMIC_DATA)' failed.
Program aborted at tick 5261728243428
--- BEGIN LIBC BACKTRACE ---
/__w/gem5/gem5/build/ALL/gem5.opt(_ZN4gem515print_backtraceEv+0x30)[0x56239b912010]
/__w/gem5/gem5/build/ALL/gem5.opt(_ZN4gem512abortHandlerEi+0x4c)[0x56239b936afc]
/lib/x86_64-linux-gnu/libc.so.6(+0x42520)[0x7f207d30b520]
/lib/x86_64-linux-gnu/libc.so.6(pthread_kill+0x12c)[0x7f207d35fa7c]
/lib/x86_64-linux-gnu/libc.so.6(raise+0x16)[0x7f207d30b476]
/lib/x86_64-linux-gnu/libc.so.6(abort+0xd3)[0x7f207d2f17f3]
/lib/x86_64-linux-gnu/libc.so.6(+0x2871b)[0x7f207d2f171b]
/lib/x86_64-linux-gnu/libc.so.6(+0x39e96)[0x7f207d302e96]
/__w/gem5/gem5/build/ALL/gem5.opt(_ZN4gem58prefetch4Base12PrefetchInfoC1EPNS_6PacketEmb+0x17b)[0x56239bd2263b]
/__w/gem5/gem5/build/ALL/gem5.opt(_ZN4gem58prefetch4Base16PrefetchListener6notifyERKPNS_6PacketE+0x19a)[0x56239bd22dfa]
/__w/gem5/gem5/build/ALL/gem5.opt(_ZN4gem59BaseCache13recvTimingReqEPNS_6PacketE+0x1f1)[0x56239bc91161]
/__w/gem5/gem5/build/ALL/gem5.opt(_ZThn72_N4gem59BaseCache11CpuSidePort13recvTimingReqEPNS_6PacketE+0x8a)[0x56239bc8c0fa]
/__w/gem5/gem5/build/ALL/gem5.opt(_ZN4gem512CoherentXBar13recvTimingReqEPNS_6PacketEs+0x4a4)[0x56239bb564a4]
/__w/gem5/gem5/build/ALL/gem5.opt(_ZN4gem59BaseCache19sendMSHRQueuePacketEPNS_4MSHRE+0x202)[0x56239bca22e2]
/__w/gem5/gem5/build/ALL/gem5.opt(_ZN4gem59BaseCache19CacheReqPacketQueue18sendDeferredPacketEv+0x101)[0x56239bc90cc1]
/__w/gem5/gem5/build/ALL/gem5.opt(_ZN4gem510EventQueue10serviceOneEv+0xc2)[0x56239b926dc2]
/__w/gem5/gem5/build/ALL/gem5.opt(_ZN4gem59doSimLoopEPNS_10EventQueueE+0x68)[0x56239b94e208]
/__w/gem5/gem5/build/ALL/gem5.opt(_ZN4gem58simulateEm+0x27b)[0x56239b94e7fb]
/__w/gem5/gem5/build/ALL/gem5.opt(+0x2616830)[0x562399704830]
/__w/gem5/gem5/build/ALL/gem5.opt(+0x1db2764)[0x562398ea0764]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x12b6d3)[0x7f207e2516d3]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyObject_Call+0x5c)[0x7f207e20d22c]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x4b26)[0x7f207e19c766]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c681f)[0x7f207e2ec81f]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x9d78)[0x7f207e1a19b8]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c681f)[0x7f207e2ec81f]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x82ae)[0x7f207e19feee]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c681f)[0x7f207e2ec81f]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(PyEval_EvalCode+0xbe)[0x7f207e2e794e]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c1edd)[0x7f207e2e7edd]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x12eb13)[0x7f207e254b13]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x69ee)[0x7f207e19e62e]
--- END LIBC BACKTRACE ---
For more info on how to address this issue, please visit https://www.gem5.org/documentation/general_docs/common-errors/ 
```

**Host Operating System**

This was run in a docker image ([ghcr.io/gem5/ubuntu-22.04_all-dependencies:latest](ghcr.io/gem5/ubuntu-22.04_all-dependencies:latest)) on top of an Ubuntu 22.04 OS.

**Host ISA**

X86

BobbyRBruce, 2023-11-20:
#564 seems like the offending PR. It's the only PR which modifies the classic cache memory system. The "riscv-ubuntu-run.py" uses a classic cache so i rule out the other memory changes which concern Ruby.

I'll see if reverting this fixes fixes the error to make sure.

BobbyRBruce, 2023-11-21:
I've confirmed that reverting #564 with #581 fixes this error. I'd prefer not to revert these changes in order to get our test passing, but this does bug need this resolved soon for the v23.1 release.

@rjc-arch : Do you have any idea what's going on here?

rjc-arch, 2023-11-21:
Sorry for the trouble @BobbyRBruce. I'll take a look at this.

BobbyRBruce, 2023-11-27:
#581 Works for now and has been merged. It would be nice if we could un-revert #581 and come up with a better solution at some point.

PR #581 (2023-11-27): mem-cache: Revert "Prefetchers Improvements"
Reverts gem5/gem5#564 to fix #580.

Exactly what in #530 is causing the bug is unknown.
commit 0b2c56ef (2023-11-27): mem-cache: Revert "Prefetchers Improvements" (#581)

Reverts gem5/gem5#564 to fix #580.

Discussion in #581 showed there may be a fix to this but reverting for now until 
a better solution is found.
closed 2023-12-28 by BobbyRBruce