gem5/gem5 #586: Question on Customizing SQC and TCP Cache Settings in apu_se.py Architecture

EricChenSudo, 2023-11-21:
Hello authors,

I am currently working with the gem5 simulator, specifically with the `apu_se.py` architecture. My project requires customized configurations for the Stream Queue Cache (SQC) and the Texture Cache Processor (TCP). I have encountered some challenges in this process and would appreciate guidance on the following aspects:

1. **Modifying Cache Parameters:** I am looking to adjust various parameters of the SQC and TCP caches, such as size, associativity, and latency. Could you provide insights or examples on how to modify these parameters within the `apu_se.py` architecture?

2. Switching the Cache to DDR3 DRAM Controller: Additionally, I am interested in transitioning the cache configuration to use a DRAM controller, specifically a DDR3 model. Could you provide guidance on integrating a DDR3 DRAM controller into the existing apu_se.py setup? This integration should focus on ensuring compatibility with the SQC and TCP caches. I am looking for detailed steps or references that could help in this modification, aiming for a setup that aligns with my simulation's performance and accuracy requirements.

Thank you for your assistance and looking forward to your suggestions.

mattsinc, 2023-11-22:
GPU_VIPER.py has a number of arguments to adjust cache size/latency/associativity/etc., starting here: https://github.com/gem5/gem5/blob/develop/configs/ruby/GPU_VIPER.py#L413.  You can either use these on the command line or change the defaults in the Python file per your preference.  Note that this will change once the GPU has a board like the CPUs though.

Regarding your second point, I'm assuming what you really mean is you want the main memory to be DDR3 instead of HBM?  Because making the caches use DDR3 instead of SRAM I have no idea how to do.  In terms of that, you can use the --mem-type parameter on the command line or in the Python file: https://github.com/gem5/gem5/blob/develop/configs/common/MemConfig.py#L127.  In terms of what memories are possible to use here, I believe all of them here (including DDR3) are possible: https://github.com/gem5/gem5/blob/develop/src/mem/DRAMInterface.py.  Note that this will also change once the GPU gets a board.

Beyond this, it is up to you to modify to your specifications and accuracy requirements.

Hope this helps,
Matt

abmerop, 2023-11-26:
Hi @EricChenSudo ,

I agree with everything that Prof. @ mattsinc mentioned.  Those files seem like they will allow you to modify the parameters you are looking to change.

Just as a nitpicky clarification to the original post, SQC / TCP actually decode to SeQencer Cache and Texture Cache per Pipeline (CU) - Want to make sure it's clear TCP is a cache and not a processor - Computer architects use a lot of overloaded acronyms and these are some of them