gem5/gem5 #591: Can not build tlm with ARM

Harshil2107, 2023-11-22:
**Description**
I am making this issue based on an email from gem5 dev mailing list. The original mail is https://www.mail-archive.com/gem5-dev@gem5.org/msg44952.html. The email states that they can not build gem5 with ARM and systemC support using tlm. They can build gem5 and can also build gem5 as a library as stated in the README in util/tlm/README but the `scons` command fails. I have also been able to reproduce this issue.

I think this is due to scons missing a dependency that is required to build.

**Affects version**
23.0

**To Reproduce**
Steps to reproduce the behavior. Please assume starting from a clean repository:

1. Compile gem5 with command: `scons build/ARM}/gem5.opt -j {cpus}`
2. Follow the steps highlighted in util/tlm/README
```
> scons build/ARM/gem5.opt
> scons --with-cxx-config --without-python --without-tcmalloc USE_SYSTEMC=0 \
>       build/ARM/libgem5_opt.so
> cd util/tlm

Note: For MAC / OSX this command should be used:
> scons --with-cxx-config --without-python --without-tcmalloc USE_SYSTEMC=0 \
>       build/ARM/libgem5_opt.dylib

To build all sources of the SystemC binding and the examples simply run scons:

> scons
```

**Expected behavior**
The `scons` build should fail with the following command `AttributeError: 'Values' object has no attribute 'duplicate_sources':
`.

BobbyRBruce, 2023-12-05:
I'm not getting this error when I reproduce. I'm running into this though when I try to build "libgem5_opt.so" though:

```sh
build/ARM/cxx_config/Gem5ToTlmBridge64.cc:149:24: error: cannot convert ‘sc_gem5::Gem5ToTlmBridge<64>*’ to ‘gem5::SimObject*’ in return
  149 |     return this->create();
      |            ~~~~~~~~~~~~^~
      |                        |
      |                        sc_gem5::Gem5ToTlmBridge<64>*
scons: *** [build/ARM/cxx_config/Gem5ToTlmBridge32.os] Error 1
scons: *** [build/ARM/cxx_config/Gem5ToTlmBridge64.os] Error 1
```

rogerchang23424, 2023-12-05:
Hi @BobbyRBruce,

If you are building it on the top of develop branch, steps of build gem5 so have changed to

```bash
scons build/ARM/gem5.opt
scons setconfig build/ARM USE_SYSTEMC=n
scons --with-cxx-config --without-python --without-tcmalloc build/ARM/libgem5_opt.so
cd util/tlm
scons
```

Thanks

ivanaamit, 2024-06-06:
I am closing this issue because PR #1105 has been merged. Please feel free to reopen it if I am mistaken and the issue is not resolved. Thanks.

closed 2024-06-06 by ivanaamit