gem5/gem5 #594: RISC-V RVV Bad execution of riscv rvv vss instruction

vmskv, 2023-11-23:
**Describe the bug**
RVV vsseg2e32 instruction doesn't work properly.

**Affects version**
Found in commit 3896673ddc943cdf2157651d090920da466bc6a7 (develop)
Date:   Sat Nov 18

**To Reproduce**
The reproducer is in https://github.com/vmskv/gem5/tree/vmskv/bad_interleave_reproducer
The steps to reproduce are in [README.txt](https://github.com/vmskv/gem5/blob/vmskv/interlieve_reproducer/bad_riscv_vss_reproducer/README.txt)

**Expected behavior**
Instruction should interleave the elements of input vectors

expected:
```
           inps/out         
rez : i = 0 : 1 5
rez : i = 1 : 2 1
rez : i = 2 : 3 6
rez : i = 3 : 4 2
rez : i = 4 : 5 7
rez : i = 5 : 6 3
rez : i = 6 : 7 8
rez : i = 7 : 8 4
```

but returns 0s
```
rez : i = 0 : 1 0
rez : i = 1 : 2 0
rez : i = 2 : 3 0
rez : i = 3 : 4 0
rez : i = 4 : 5 0
rez : i = 5 : 6 0
rez : i = 6 : 7 0
rez : i = 7 : 8 0
```
**Host Operating System**
Ubuntu 20.04

**Guest ISA**
RISC-V.

hnpl, 2023-11-26:
Thanks for pointing this out. From what I see In the gem5 VSStride macro-op template (https://github.com/gem5/gem5/blob/3896673ddc943cdf2157651d090920da466bc6a7/src/arch/riscv/isa/templates/vector_mem.isa#L1014-L1048) and the corresponding generated ISA code (`build/RISCV/arch/riscv/generated/exec-ns.cc.inc`), the VSS (vector strided-store) macro instructions do not take into account NFIELDS; so I don't think the segmented VsStride instructions with NFIELDS != 0 are supported.

ivanfv, 2023-11-28:
Hi,

I guess this is expected, since as far as I am aware unit-stride segmented loads/stores are not implemented yet. Their encoding only differ from vle/vse in NFIELDS, which is not currently checked in the decoder. Because of that, they are probably being interpreted as vle/vse ones (which makes sense with the results you are getting, only first field is right).

I am currently working on the implementation of these instructions (https://github.com/gem5/gem5/issues/382). While I expect to have them soon, it is still work in progress.

-Ivan

ivanaamit, 2024-04-15:
Hi @ivanfv, could you please let us know if this issue should be closed or if there is a real bug? Thank you.

ivanfv, 2024-04-16:
Hi @ivanaamit, yes, I think this issue should be closed since vector segmented store instructions were added with PR https://github.com/gem5/gem5/pull/913

closed 2024-04-16 by ivanaamit