gem5/gem5 #612: stdlib's ArmBoard does not support KVM

hnpl, 2023-11-28:
**Describe the bug**
Using the KVM core on the stdlib's ArmBoard does not work.

**Affects version**
Current develop (https://github.com/gem5/gem5/tree/3bf0b1d22a375f9a1da34150293e8f8ddf433391).

**gem5 Modifications**
N/A

**To Reproduce**
Using the `configs/example/gem5_library/arm-ubuntu-run.py` script with the following modifications,

```
--- a/configs/example/gem5_library/arm-ubuntu-run.py
+++ b/configs/example/gem5_library/arm-ubuntu-run.py
@@ -78,12 +78,12 @@ 
-processor = SimpleProcessor(cpu_type=CPUTypes.TIMING, num_cores=2, isa=ISA.ARM)
+processor = SimpleProcessor(cpu_type=CPUTypes.KVM, num_cores=2, isa=ISA.ARM)
 
 # The ArmBoard requires a `release` to be specified. This adds all the
 # extensions or features to the system. We are setting this to Armv8
 # (ArmDefaultRelease) in this example config script.
-release = ArmDefaultRelease()
+release = ArmDefaultRelease().for_kvm()
```

Then running the following command,

```sh
build/ARM/gem5.opt configs/example/gem5_library/arm-ubuntu-run.py
```

**Terminal Output**

```
src/cpu/kvm/device.cc:96: panic: Failed to set attribute (group: 3, attr: 0, errno: 22)       
Memory Usage: 2731744 KBytes                                                                  
Program aborted at tick 0                                                                     
--- BEGIN LIBC BACKTRACE ---                                                                  
build/ARM/gem5.opt(_ZN4gem515print_backtraceEv+0x3c)[0xaaaadbfdbafc]                          
build/ARM/gem5.opt(_ZN4gem512abortHandlerEi+0x5c)[0xaaaadbfff68c]                             
linux-vdso.so.1(__kernel_rt_sigreturn+0x0)[0xffff837647dc]                                    
/lib/aarch64-linux-gnu/libc.so.6(+0x7f200)[0xffff82cdf200]                                    
/lib/aarch64-linux-gnu/libc.so.6(raise+0x1c)[0xffff82c9a67c]                                  
/lib/aarch64-linux-gnu/libc.so.6(abort+0xe4)[0xffff82c87130]                                  
build/ARM/gem5.opt(+0x192a8e8)[0xaaaad9e2a8e8]                                                
build/ARM/gem5.opt(+0x25b33f8)[0xaaaadaab33f8]                                                
build/ARM/gem5.opt(_ZN4gem514KvmKernelGicV3C1ERNS_5KvmVMERKNS_20MuxingKvmGicV3ParamsE+0x84)[0x
aaaadbf62318]                                                                                 
build/ARM/gem5.opt(_ZN4gem512MuxingKvmGicINS_10GicV3TypesEEC2ERKNS_20MuxingKvmGicV3ParamsE+0xa
c)[0xaaaadbf646bc]                                                                            
build/ARM/gem5.opt(_ZNK4gem520MuxingKvmGicV3Params6createEv+0x24)[0xaaaadbf5d514]             
build/ARM/gem5.opt(+0x3a60590)[0xaaaadbf60590]                                                
build/ARM/gem5.opt(+0x190d354)[0xaaaad9e0d354]                                                
/lib/aarch64-linux-gnu/libpython3.10.so.1.0(+0x120334)[0xffff83260334]                        
/lib/aarch64-linux-gnu/libpython3.10.so.1.0(_PyObject_MakeTpCall+0x8c)[0xffff83219470]        
/lib/aarch64-linux-gnu/libpython3.10.so.1.0(+0xdc528)[0xffff8321c528]                         
/lib/aarch64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x91e0)[0xffff831b9a00]  
/lib/aarch64-linux-gnu/libpython3.10.so.1.0(+0x1b4104)[0xffff832f4104]                        
/lib/aarch64-linux-gnu/libpython3.10.so.1.0(+0xdc4d8)[0xffff8321c4d8]                         
/lib/aarch64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x91e0)[0xffff831b9a00]  
/lib/aarch64-linux-gnu/libpython3.10.so.1.0(+0x1b4104)[0xffff832f4104]                        
/lib/aarch64-linux-gnu/libpython3.10.so.1.0(+0xdc4d8)[0xffff8321c4d8]                         
/lib/aarch64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x91e0)[0xffff831b9a00]  
/lib/aarch64-linux-gnu/libpython3.10.so.1.0(+0x1b4104)[0xffff832f4104]                        
/lib/aarch64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x91e0)[0xffff831b9a00]  
/lib/aarch64-linux-gnu/libpython3.10.so.1.0(+0x1b4104)[0xffff832f4104]                        
/lib/aarch64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x6480)[0xffff831b6ca0]  
/lib/aarch64-linux-gnu/libpython3.10.so.1.0(+0x1b4104)[0xffff832f4104]                        
/lib/aarch64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x6480)[0xffff831b6ca0]  
/lib/aarch64-linux-gnu/libpython3.10.so.1.0(+0x1b4104)[0xffff832f4104]                        
/lib/aarch64-linux-gnu/libpython3.10.so.1.0(PyEval_EvalCode+0xa4)[0xffff832eef28]             
/lib/aarch64-linux-gnu/libpython3.10.so.1.0(+0x1af4d4)[0xffff832ef4d4]
--- END LIBC BACKTRACE ---
For more info on how to address this issue, please visit https://www.gem5.org/documentation/general_docs/common-errors/
```

**Host Operating System**
Ubuntu 22.04, kernel v5.15

**Host ISA**
Arm (Ampere Altra 2x 80 Cores Arm Server)

**Compiler used**
N/A

**Additional information**

I was able to trace down this problem and I don't know how to resolve this issue without hacky patches.

When using the host's GIC, the problems are,

- The default number of it_lines of the ArmBoard is not divisible by 32, while the linux kernel requires it to be divisible by 32 [1]. This can be fixed by making this number divisible by 32 in gem5.
- Using `release=ArmDefaultRelease.for_kvm()` instead of `release=ArmDefaultRelease()`.
- Avoiding updating the ICC_CTLR_EL1 if the destination GIC registers are the host GIC registers. It appears that the content of ICC_CTRL_EL1 of the ArmBoard is `0x48c00`, which is incompatible with ICC_CTLR_EL1 of the Ampere core, which is `0x8400`. Here's the kernel code that rejects the udpated ICC_CTRL_EL1 [2]. So if we don't update the ICC_CTRL_EL1 register of the host, which I believe what QEMU does as well, the ArmBoard can boot up the linux kernel.

[1] https://elixir.bootlin.com/linux/v5.15.139/source/arch/arm64/kvm/vgic/vgic-kvm-device.c#L188
[2] https://elixir.bootlin.com/linux/v5.15.139/source/arch/arm64/kvm/vgic-sys-reg-v3.c#L13

When using simulate_gic option, i.e. `board.realview.gic.simulate_gic = True`, the host's terminal does not show anything.

powerjg, 2023-11-28:
@giactra @andysan If you could provide some direction here, it would be appreciated.

@Harshil2107 this should be added to the list for v23.1

hnpl, 2023-11-29:
It looks like one of the reasons is that the ArmBoard uses the VExpress_Foundation, which does not have VGic.

giactra, 2023-12-06:
> It looks like one of the reasons why simulate_gic does not work is that the ArmBoard uses the VExpress_Foundation, which does not have VGic.

The problem is not really VGic, which is the gem5 module used to virtualize the guest hardware. The issue is that there is no userspace gic support for GICv3. Which means the only way to run KVM with the Foundation platform is to run it on a GICv3 machine and use the host interrupt controller, whereas by using simulate_gic, we ALWAYS use the gem5 GIC model which means we have a KVM simulation decoupled from the host IC.

This is the reason why we tend to use simulate_gic in CI as it is more "portable".

Shall we provide a different board for KVM runs (which is using VExpress_GEM5_V1 in the backend)?

giactra, 2023-12-14:
> Thanks for the information, Giacomo. I see that we have two options here,
> 
>     1. Using the VExpress_GEM5_V1 for the stdlib ArmKvmBoard.
> 
>     2. Avoiding copying `ICC_CTLR_EL1` from the guest to the host, and using the `ICC_CTLR_EL1` from the host instead. I believe QEMU does this, but I'm not entirely sure about the correctness when we switch from KVM to a gem5 CPU.
> 

I will try to assess the feasibility of 2 but I believe option one could be the easy/quick solution for now

> 
> I'd say I prefer option 1. However, what puzzles me is that why did we decide to create VExpress_Foundation for the stdlib ArmBoard. I'd also like your opinion on the option 2 as well.

The reason for preferring the Foundation platform is that it uses GICv3 which is what modern CPUs are using. GICv2 has the fundamental limitation of not supporting more than 8CPUs.. So if a user wants to simulate a multi/many core setup, it would be hard to do it with GICv2 (_V1).
The only problem with the Foundation platform as we see is with KVM...

closed 2024-02-05 by ivanaamit