gem5/gem5 #616: Update RISCV kernel and refactor stdlib to use the new kernel and bootloader workload.

Harshil2107, 2023-11-28:
A new RISCV kernel and bootloader was introduced with the patcher #390 and gem5-resources 13 (https://github.com/gem5/gem5-resources/pull/13). We can refactor stdlib to use these new resources.

These are the to-dos to implement this change:

- [ ] Mark the old kernel as deprecated, In our case we can just edit the `gem5-versions` field in the JSON of the resource to do this.
- [ ] build and add the resources mentioned in gem5-resources 13.
- [ ] Refactor the stdlib to use the RiscvISA::BootloaderKernelWorkload rather than RiscvISA::FsWorkload
- [ ] Update the tests in tests/ to use the newer kernel/bootloader.

powerjg, 2023-12-12:
@Harshil2107 what is the status of this?

Harshil2107, 2023-12-14:
The tests have been updated. The riscv matched and riscv board now support the new kernel bootloader workload. A new version of `riscv-ubuntu-20.04-boot` with the new kernel and the bootloader is uploaded on the database.  The tests will now get this new version for gem version 23.1 and develop.

I am closing the issue as completed.

closed 2023-12-14 by Harshil2107