gem5/gem5 #621: Cache model does not correctly handle flush in MSHR deferred target list

georgehodgkins, 2023-11-29:
**Describe the bug**
I am working on a basic benchmark to test some new hardware and accompanying execution model. The benchmark prefills the cache, then alternates between 8-byte reads and writes to pseudorandom locations within a fixed-size array. After each write, we issue a non-invalidating cache flush (RISCV `cbo.clean`) on the write address.

This seems to cause a bug in the standard cache model. Specifically, at one point in execution, we have the following list of accesses outstanding against an MSHR:
```
12070515720: system.cpu1.dcache.mshr_queue.entry: After target allocation: [0x4004c540:0x4004c57f](ns)  AllocOnFill  state:  InSvc                                                                                                                                                                                                                                                         
      Targets:                                                                                                                                                                                                                                                                                                                                                                             
        FromCPU: ReadReq [4004c540:4004c547]                                                                                                                                                                                                                                                                                                                                               
      Deferred Targets:                                                                                                                                                                      
        FromCPU: WriteReq [4004c568:4004c56f]                                                                                                                                                
        FromCPU: CleanSharedReq [4004c568:4004c56f] PoC
```

When the `ReadReq` returns and fills the cache line, the MSHR is deallocated without the `CleanSharedReq` being sent downstream:
```
12071410220: system.cpu1.dcache: recvTimingResp: Handling response ReadResp [4004c540:4004c57f]                                                                                              
12071410220: system.cpu1.dcache: Block for addr 0x4004c540 being updated in Cache                                                                                                            
12071410220: system.cpu1.dcache: Replacement victim: state: 4 (S) writable: 0 readable: 1 dirty: 0 prefetched: 0 | tag: 0x1000c secure: 0 valid: 1 | set: 0x15 way: 0                        
12071410220: system.cpu1.dcache: Create CleanEvict CleanEvict [40030540:4003057f]                                                                                                            
12071410220: system.cpu1.dcache: Block addr 0x4004c540 (ns) moving from  to state: 6 (E) writable: 1 readable: 1 dirty: 0 prefetched: 0 | tag: 0x10013 secure: 0 valid: 1 | set: 0x15 way: 0 
12071410220: system.cpu1.dcache: satisfyRequest for WriteReq [4004c568:4004c56f] (write)
12071410220: system.cpu1.dcache.mshr_queue: Deallocating all targets: [0x4004c540:0x4004c57f](ns)    state:  InSvc    
12071410220: system.cpu1.dcache.mshr_queue: MSHR deallocated. Number in use: 0/4
```

Investigating with a debugger shows that when the `ReadReq` returns, the block is filled, both of the deferred targets are moved to the main target list, and then `serviceMSHRTargets` calls `satisfyRequest` on all three targets. For the `CleanSharedReq`, all this does is incorrectly clear the dirty bit on the cache block (`src/mem/cache/base.cc:1170`). After this, the MSHR is deallocated, and the `CleanSharedReq` is never sent.

Putting an ordering fence between the write and the flush resolves the issue, but since this is not required by the ISA and artificially limits our performance on the benchmark, it is not an ideal solution.

**Affects version**
Produced with v22.1.0.0, likely to affect all newer versions as well, since the relevant code in (`src/mem/cache/{cache, base, mshr}.cc`) has not been changed since that release.

**gem5 Modifications**
We are developing a device which interposes on memory accesses (it sits between the memory bus and the backing store). Other than this device, we are using unmodified models from upstream gem5 for cache, core, membus, etc.  Given the information I wrote above, I think it is unlikely that our device is the cause of the issue, since it seems to be entirely contained within the cache model. However, I'd be happy to be wrong about that, since it's easier for me to fix my own code than wait for an upstream fix.

**To Reproduce**
Unfortunately, I am not sure how to reproduce this bug without our (proprietary) device model, or specifically how to make it a hard error. Our model is "expecting" the cache flush by design, and eventually throws an error when it fails to arrive. On a normal system, failure to flush a cache line when requested would not cause any immediate error, although incorrectly clearing the dirty bit on the block could lead to future data corruption. This might or might not surface as functional errors in a simulated application, and if so it would likely be far removed from when the original incorrect behavior happened.

However, I think that if you can get an MSHR to have a target list similar to the one I showed (`ReadReq`->`WriteReq`->`CleanSharedReq` on the same block), then you should be able to observe the incorrect behavior using a debugger and/or tracing. 

I would also be happy to test patches against our model to check if they resolve the issue.

**Terminal Output**
See description.

**Expected behavior**
The `ReadReq` should fill the block, the `WriteReq` should be satisfied against the newly filled block, and then  the `CleanSharedReq` should be sent downstream with the post-write contents of the block.

**Host Operating System**
Ubuntu 22.04.3

**Host ISA**
X86

**Compiler used**
g++ 11.4.0

**Additional information**
We are targeting the RISCV isa, using the O3CPU model and memory timing mode. Our configuration has two cores, each with private L1 caches connected directly to the memory bus (no shared caches). We change the size of the L1 caches (32 KB each for I and D), but otherwise use the default configuration provided by `CacheConfig`.

erin-le, 2025-03-07:
@georgehodgkins Hello, sorry for the delay in addressing this issue. Unfortunately, since the gem5 development team can't easily reproduce the issue, we don't have a good way of fixing it. If you have a fix, however, please consider opening a pull request. If you are interested in working on a fix, it looks like there have been some changes in the code relevant to this issue since v22.1.0.0, so it may be worth checking to see if this issue still occurs.

Thank you for the bug report!

georgehodgkins, 2025-03-09:
Thanks for the response. I actually found and fixed the bug in my fork of gem5 a while ago - the issue is that writes can be reordered in front of earlier cache cleans that are pending in an MSHR. If a cache clean is followed by a write to the same cache line, it is possible for the later write's contents to be written back as part of the earlier cache clean, which is a violation of at least the [RISCV ordering rules](https://five-embeddev.com/riscv-user-isa-manual/Priv-v1.12/rvwmo.html) for [the CMO extension](https://github.com/riscv/riscv-CMOs/blob/master/specifications/cmobase-v1.0.1.pdf).

To fix this, I changed `MSHR::promoteWritable` to stop promoting targets if it hits any cache maintenance operation (rather than just an invalidate), and `BaseCache::access` to force all writes matching an MSHR with CMO targets to coalesce with that MSHR, rather than being satisfied immediately. There's also a few other code paths in the cache code that needed to change to accomodate the presence of evict writebacks in the MSHR target list. 

[cmo_target_list.patch](https://github.com/user-attachments/files/19151614/cmo_target_list.patch) Here's a patch with the changes - it doesn't apply cleanly to the current version of gem5 but the affected code doesn't look like it's changed since 22.1.0.0.

Having the fix in my fork is sufficient for me, so I don't care whether this gets fixed upstream - it's definitely a bug, but not one you're going to encounter (or be able to replicate/test) unless you're paying close attention to the ordering of writes at the point of coherence, which I suspect most users of gem5 are not.