gem5/gem5 #632: sc_main.py can not run correctly

zhiqianglife, 2023-12-01:
the latest version of gem5.

i use the follwing command line, but meet coredumped

[image]

how to run it correctly?

Harshil2107, 2023-12-05:
Hi,
Please use code blocks instead of screenshots for commands and terminal outputs. Can you also give more information about how you built gem5 and some steps that can be followed to reproduced this issue?

Thank you.

zhiqianglife, 2023-12-05:
Thanks for your reply, i just cloned this project，then i used the following command to compile it,
```shell
scons build/ARM/gem5.debug -j72
``` 
then used the following command to run it:
```shell
./build/ARM/gem5.debug configs/example/sc_main.py
``` 
and the coredunp happend:
```shell
gem5 Simulator System.  https://www.gem5.org
gem5 is copyrighted software; use the --copyright option for details.

gem5 version 23.0.1.0
gem5 compiled Dec  1 2023 14:48:10
gem5 started Dec  5 2023 10:08:06
gem5 executing on ubuntu, pid 7977
command line: ./build/ARM/gem5.debug configs/example/sc_main.py

src/systemc/core/sc_main_fiber.cc:77: fatal: sc_main called but not defined.
Memory Usage: 227688 KBytes
src/base/fiber.cc:114: panic: panic condition stack && _currentFiber == this occurred: Fiber stack is in use.
Memory Usage: 227688 KBytes
Program aborted at tick 0
--- BEGIN LIBC BACKTRACE ---
./build/ARM/gem5.debug(+0x34ec94e)[0x55e6c3aad94e]
./build/ARM/gem5.debug(+0x351869b)[0x55e6c3ad969b]
/lib/x86_64-linux-gnu/libpthread.so.0(+0x12980)[0x7fbc238ac980]
/lib/x86_64-linux-gnu/libc.so.6(gsignal+0xc7)[0x7fbc216eee87]
/lib/x86_64-linux-gnu/libc.so.6(abort+0x141)[0x7fbc216f07f1]
./build/ARM/gem5.debug(+0x74b92c)[0x55e6c0d0c92c]
./build/ARM/gem5.debug(+0xcf716b)[0x55e6c12b816b]
./build/ARM/gem5.debug(+0xe44436)[0x55e6c1405436]
/lib/x86_64-linux-gnu/libc.so.6(+0x43031)[0x7fbc216f3031]
/lib/x86_64-linux-gnu/libc.so.6(+0x4312a)[0x7fbc216f312a]
./build/ARM/gem5.debug(+0xd070b4)[0x55e6c12c80b4]
./build/ARM/gem5.debug(+0x74b927)[0x55e6c0d0c927]
./build/ARM/gem5.debug(+0xe4419d)[0x55e6c140519d]
./build/ARM/gem5.debug(+0xcf75fc)[0x55e6c12b85fc]
./build/ARM/gem5.debug(+0xcf6c63)[0x55e6c12b7c63]
/lib/x86_64-linux-gnu/libc.so.6(+0x58680)[0x7fbc21708680]
--- END LIBC BACKTRACE ---
For more info on how to address this issue, please visit https://www.gem5.org/documentation/general_docs/common-errors/

Aborted (core dumped)
```

zhiqianglife, 2023-12-05:
> Hi, Please use code blocks instead of screenshots for commands and terminal outputs. Can you also give more information about how you built gem5 and some steps that can be followed to reproduced this issue?
> 
> Thank you.

Thanks for your reply, i just cloned this project，then i used the following command to compile it,
```shell
scons build/ARM/gem5.debug -j72
``` 
then used the following command to run it:
```shell
./build/ARM/gem5.debug configs/example/sc_main.py
``` 
and the coredunp happend:
```shell
gem5 Simulator System.  https://www.gem5.org
gem5 is copyrighted software; use the --copyright option for details.

gem5 version 23.0.1.0
gem5 compiled Dec  1 2023 14:48:10
gem5 started Dec  5 2023 10:08:06
gem5 executing on ubuntu, pid 7977
command line: ./build/ARM/gem5.debug configs/example/sc_main.py

src/systemc/core/sc_main_fiber.cc:77: fatal: sc_main called but not defined.
Memory Usage: 227688 KBytes
src/base/fiber.cc:114: panic: panic condition stack && _currentFiber == this occurred: Fiber stack is in use.
Memory Usage: 227688 KBytes
Program aborted at tick 0
--- BEGIN LIBC BACKTRACE ---
./build/ARM/gem5.debug(+0x34ec94e)[0x55e6c3aad94e]
./build/ARM/gem5.debug(+0x351869b)[0x55e6c3ad969b]
/lib/x86_64-linux-gnu/libpthread.so.0(+0x12980)[0x7fbc238ac980]
/lib/x86_64-linux-gnu/libc.so.6(gsignal+0xc7)[0x7fbc216eee87]
/lib/x86_64-linux-gnu/libc.so.6(abort+0x141)[0x7fbc216f07f1]
./build/ARM/gem5.debug(+0x74b92c)[0x55e6c0d0c92c]
./build/ARM/gem5.debug(+0xcf716b)[0x55e6c12b816b]
./build/ARM/gem5.debug(+0xe44436)[0x55e6c1405436]
/lib/x86_64-linux-gnu/libc.so.6(+0x43031)[0x7fbc216f3031]
/lib/x86_64-linux-gnu/libc.so.6(+0x4312a)[0x7fbc216f312a]
./build/ARM/gem5.debug(+0xd070b4)[0x55e6c12c80b4]
./build/ARM/gem5.debug(+0x74b927)[0x55e6c0d0c927]
./build/ARM/gem5.debug(+0xe4419d)[0x55e6c140519d]
./build/ARM/gem5.debug(+0xcf75fc)[0x55e6c12b85fc]
./build/ARM/gem5.debug(+0xcf6c63)[0x55e6c12b7c63]
/lib/x86_64-linux-gnu/libc.so.6(+0x58680)[0x7fbc21708680]
--- END LIBC BACKTRACE ---
For more info on how to address this issue, please visit https://www.gem5.org/documentation/general_docs/common-errors/

Aborted (core dumped)
```

giactra, 2023-12-06:
> src/systemc/core/sc_main_fiber.cc:77: fatal: sc_main called but not defined.

The issue is that you are missing the sc_main, which is the main SystemC function which instantiates and connect the simulation blocks. If you check the configs/example/sc_main.py config, you can see it does very little as it expects the system to be described within the sc_main function.

Please refer to the systemc within gem5 documentation: https://github.com/gem5/gem5/tree/stable/util/systemc/systemc_within_gem5 to have a better understanding on how to compose a gem5+SystemC simulation.

Closing the issue now, please reopen if necessary

closed 2023-12-06 by giactra