gem5/gem5 #644: Enable bundling of `obtain_workload` requests in gem5 Resources stdlib component

BobbyRBruce, 2023-12-02:
At present the `obtain_resource` function is used to get Resources from the gem5 standard library but runs on a "single-request-per-resource" model to our clients (in almost all cases the MongoDB database where we store our gem5 Resources data).

This is very inefficient in cases where we need to get many resources. With the introduction of `Suite` resource objects we have a single `Suite` resource containing many `Workload` resources which, in turn` contain several resources. A "single-request-per-resource" model is very costly in this case.

#625 was added to fix this case but is _very hacky_ and not an ideal long term solution.

I suggest extended the functionality of the `obtain_resource` to allow the specification of 1..N resources and, if multiple are specified, these requests are bundled.

Harshil2107, 2024-01-02:
My solution to this would be to separate `obtain_resource` function to the resource class level.
- We need to update Atlas and JSON clients to returns more than 1 resource
- Then we can make the `obtain_resource` function just get the original resource to create the resource specialized class
- In `AbstractResource` class we can have a general `obtain_resource` function (name not decided yet) that has the remaining functionality of the current `obtain_resource` class like downloading the actual resource files, etc.
   - This function can be overloaded to allow getting multiple resources and to do other specialized things that per category.

BobbyRBruce, 2024-01-02:
> My solution to this would be to separate `obtain_resource` function to the resource class level.

Can you explain to me why this is better than the current solution?

> * We need to update Atlas and JSON clients to returns more than 1 resource

Can you show me how the API will change at the level of Python code? The functions we have, how are their signatures going to be altered?

> * Then we can make the `obtain_resource` function just get the original resource to create the resource specialized class

Can you rewrite this? I'm unclear what the "original resource" is and what is needed to make it returned by the `obtain_resource` function.

> * In `AbstractResource` class we can have a general `obtain_resource` function (name not decided yet) that has the remaining functionality of the current `obtain_resource` class like downloading the actual resource files, etc.

I have some confusion here but I think it could be cleared up by answering the above questions.

>   
>   * This function can be overloaded to allow getting multiple resources and to do other specialized things that per category.

Can you give an example of how you'd override this function to get multiple resources? I'd appreciate it if you could reference it back to your answers to the other questions so I can understand. How does this interact with the changed APIs and how is this functionality enabled by moving the `obtain_resource` function to a the Resource classes? It might be helpful to write some pseudocode.

Harshil2107, 2024-01-02:
> Can you explain to me why this is better than the current solution?

I think this gives us more control and removes the extra if statements that we have in the current `obtain_resource` function based on category. It also allows us to create custom fields like how we did for workloads for the `parameters` argument. I feel this solution is overall cleaner and gives us more flexibility.

> Can you show me how the API will change at the level of Python code? The functions we have, how are their signatures going to be altered?

The major change to the API will be that we used an array of dictionaries like `[{id: "xyz", resource_version: 123},...]` instead of resources id and version being 2 different arguments. An example of this would be `get_resource` function in atlasclient (https://github.com/gem5/gem5/blob/bae34876780dfb2bc22b9151bfda1d39ee80cfb1/src/python/gem5/resources/client_api/atlasclient.py#L156) will change to
```
def get_resources(
        self,
        resources_info: Optional[Dict[str,str]],
        gem5_version: Optional[str] = None,
    )

> Can you show me how the API will change at the level of Python code? The functions we have, how are their signatures going to be altered?

I mean that we ask the API to get the resource that we want like we do currently and then get the remaining resources together as we need the original resources to get the category of the resource and create the appropriate resource class object

So the Abstract resource class will just download the resource if it has the url. We can override this function for workloads and suites to get the dependent resources.

BobbyRBruce, 2024-08-07:
Implemented with #779

closed 2024-08-07 by BobbyRBruce