gem5/gem5 #660: Release gem5 v23.1

BobbyRBruce, 2023-12-06:
The gem5 staging branch has been created (#558). The staging branch can be found at https://github.com/gem5/gem5/tree/release-staging-v23-1.

The [release procedures](https://www.gem5.org/documentation/general_docs/development/release_procedures/) outlines the changes to be applied to the staging branch prior to a release. In addition to this, the branch will be intensely tested to ensure the gem5 v23.1 release is of acceptable quality.

It is possible for changes to be cherry-picked onto the staging branch during this time but this _should_ be reserved for changes which have a low probability of introducing disruptive functional change and bug fixes.

The following changes should be cherry-picked onto the staging branch:

- [ ] #510
- [ ] #612
- [ ] #447
- [ ] #616

rogerchang23424, 2023-12-13:
Hi @BobbyRBruce,

I think the PR #674 should be cherry-pick in the gem5 v23.1. The RVV store instruction falsely mark as source because of the lack condition for `assignRE`. They should not mark as load.

BobbyRBruce, 2023-12-13:
> Hi @BobbyRBruce,
> 
> I think the PR #674 should be cherry-pick in the gem5 v23.1. The RVV store instruction falsely mark as source because of the lack condition for `assignRE`. They should not mark as load.

Thanks! I've noted it down in the top comment. I'll be cherry-picked to staging soon.

giactra, 2023-12-14:
Hi, I have this PR which is targeting the release-staging branch directly: https://github.com/gem5/gem5/pull/684

closed 2024-01-02 by Harshil2107