gem5/gem5 #668: SE with SMT hangs on RISCV and fails on ARM

cnpapado, 2023-12-10:
**Describe the bug**
SE SMT simulation hangs on RISCV and fails with `Assertion 'params.numPhysVecPredRegs >= numThreads * regClasses.at(VecPredRegClass)->numRegs()' failed.` (found on `src/cpu/o3/cpu.cc:202`) on ARM.

I've been reading that SMT should work in SE mode, that's why I am opening the issue. However if that's not the case, does anyone know a version were it works at least for ARM? 

**Affects version**
v23.0.1 
In v22.1 simulation completes successfully on RISCV but the same assertion fails on ARM (that part is true also for earlier versions) 

**To Reproduce**
1. Compile with `scons build/RISCV/gem5.opt -jX` and `scons build/ARM/gem5.opt -j48`
2. Execute with 
```
build/RISCV/gem5.opt configs/deprecated/example/se.py '--cmd=tests/test-progs/hello/bin/riscv/linux/hello;tests/test-progs/hello/bin/riscv/linux/hello' --caches --l1d_size=32kB --l1d_assoc=8 --l1i_size=32kB --l1i_assoc=8 --l2cache --cpu-type=DerivO3CPU --mem-type=DDR4_2400_16x4 --mem-size=64GB --smt
```
and 
```
build/ARM/gem5.opt configs/deprecated/example/se.py '--cmd=tests/test-progs/hello/bin/arm/linux/hello;tests/test-progs/hello/bin/arm/linux/hello' --caches --l1d_size=32kB --l1d_assoc=8 --l1i_size=32kB --l1i_assoc=8 --l2cache --cpu-type=DerivO3CPU --mem-type=DDR4_2400_16x4 --mem-size=64GB --smt
```
I know that `se.py` has been deprecated but I suspect it is not a bug in the config script, hence I'm still using it to reproduce. However I'm happy to try with some other configuration file as well.

**Expected behavior**
Successfully complete the simulation.

**Host Operating System**
Ubuntu 22.04

**Host ISA**
X86

**Compiler used**
gcc v10.2.1

Let me know if you need more information. Since I need this especially for ARM, I am happy trying to fix it, but I'll need some guidance since I'm new to gem5. Thanks a lot.

giactra, 2023-12-14:
Hi @cnpapado, thanks for reporting this. The issue for Arm is that the base O3CPU doesn't allocate enough physical predicate registers by default to implement the two architectural reg files required for the 2 HW threads.

The solution is rather simple, you should just set cpu.numPhysVecPredRegs to a higher value.
As:
`const int NumVecPredRegs = 18`

You should choose a number bigger > 18 * 2 = 36

cnpapado, 2023-12-19:
Thanks for the reply! It was indeed simple after all. Changing the number of predicate registers resolves the issue for v22.1 whether v23.0.1 hangs exactly as riscv. Should I leave the ticket open?

giactra, 2023-12-19:
> Thanks for the reply! It was indeed simple after all. Changing the number of predicate registers resolves the issue for v22.1 whether v23.0.1 hangs exactly as riscv. Should I leave the ticket open?

Yes as the issue still holds (SMT not working).. I am gonna have a look at this more seriously after release

ivanaamit, 2024-01-16:
Hi @giactra, have you had a chance to look at this? Thank you.

giactra, 2024-01-17:
> Hi @giactra, have you had a chance to look at this? Thank you.

Hi @ivanaamit I have something I will post this week

giactra, 2024-02-21:
This has been fixed by #828

closed 2024-02-21 by giactra