gem5/gem5 #686: Scons shows KVM not present even if it is enabled

Harshil2107, 2023-12-14:
While building gem5 with KVM. The build message shows KVM is not present even if I am in the KVM group as mentioned in https://www.gem5.org/documentation/general_docs/using_kvm/.

The following warning is shown in the terminal when building gem5
```
Checking for C header file linux/kvm.h... yes
Checking for timer_create(CLOCK_MONOTONIC, NULL, NULL) in C library None... no
Checking for timer_create(CLOCK_MONOTONIC, NULL, NULL) in C library rt... yes
Checking for member exclude_host in struct perf_event_attr...yes
Warning: Can not enable KVM, host seems to lack KVM support
Checking for backtrace_symbols_fd((void *)1, 0, 0) in C library None... yes
```

**Gem5 version**
23.1 staging branch

**Steps to reproduce**
- Make sure you have KVM enabled
- Clone the gem5 repo and checkout the staging branch `git checkout release-staging-v23-1`
- Build gem5 using the command `scons ./build/ALL/gem5.opt -j 20`

**Expected behavior**
gem5 should build with KVM.

9Tempest, 2024-01-15:
I encountered the same issue... I exactly followed [this document](https://www.gem5.org/documentation/general_docs/using_kvm/), and I can make sure hardware virtualization is presented by `egrep -c '(vmx|svm)' /proc/cpuinfo`, but I still got the same issue above

powerjg, 2024-01-15:
This message *says* that KVM isn't available, but when gem5 is compiled it *does include* KVM. You can check if KVM is compiled in by running `gem5 --list-sim-objects` and searching for `X86KVMCPU` in the output.

Anzi914, 2024-01-17:
So is this problem solved now？I met the same problem lately.If so, pls tell us how to do,thanks!

powerjg, 2024-01-17:
The "problem" is only in the text output, not the functionality of the simulator. The "problem" is that the simulator **SAYS** that KVM isn't supported. However, that is **NOT** correct. KVM is supported and compiled in.

EmilyG2413, 2024-04-04:
Hello, we are two students studying Computer Science at the University of Texas at Austin. Currently, we are enrolled in CS360V Virtualization, taught by Dr. Chidambaram (VJ). As part of our coursework, we are required to contribute to issues within gem5. Currently, we are tackling some other issues in the repo, however, we thought that this would be a good one to get started and learn more about virtualization/KVM so we were wondering if we may attempt to fix this text output problem.  

Best, 
Emily and Lukas

powerjg, 2024-04-04:
Sure! I think this is just a logical issue in the SConscripts files. There's nothing deeply wrong here. Basically, I think the wrong variable is being checked when the warning is printed.

Actually, now that I type this out, I think that @ivanaamit or @Harshil2107 may have made some progress on this. Let's see if they respond.

Harshil2107, 2024-04-08:
Hi,
This issue was introduced after the kconfig change (#69). As mentioned above the error is related to SConscript files. This file that sends the error messages is the following: https://github.com/gem5/gem5/blob/ee6f1377d7c54422137dfa47cd4d73407814867d/src/cpu/kvm/SConsopts. 

The kconfig file related to this change is: https://github.com/gem5/gem5/blob/ee6f1377d7c54422137dfa47cd4d73407814867d/src/cpu/kvm/Kconfig

I think the main issue is that the correct env variable is not being checked or the env variable is not set properly.

Lukas-Zenick, 2024-04-12:
I can't seem to replicate the issue. Since this post was made, v23-1 has been merged into stable, so I'm currently just running on the HEAD (ee6f1377d7c54422137dfa47cd4d73407814867d). I added an else case such that the code is now:
`if not main['CONF']['KVM_ISA']:
    warning("Can not enable KVM, host seems to lack KVM support")
else:
    warning("KVM is all good")`

And when I run this I get the "KVM is all good" message.

closed 2024-04-15 by ivanaamit