gem5/gem5 #697: Daily Ruby/Garnet memory tests fail with as they use the NULL ISA. Attempts to use X86 ISA have not work

BobbyRBruce, 2023-12-19:
The gem5 Daily tests are failing: https://github.com/gem5/gem5/actions/runs/7244842221/job/19735812907

The following tests are failing:

```sh
Test: garnet_synth_traffic-NULL-x86_64-opt Failed
Test: ruby_mem_test-garnet-NULL-x86_64-opt Failed
Test: ruby_mem_test-simple-NULL-x86_64-opt Failed
Test: ruby_mem_test-simple-extra-NULL-x86_64-opt Failed
Test: ruby_mem_test-simple-extra-multicore-NULL-x86_64-opt Failed
Test: ruby_random_test-NULL-x86_64-opt Failed
Test: ruby_direct_test-NULL-x86_64-opt Failed

They are failing with the following error: 

```sh
NameError: name 'DerivO3CPU' is not defined

At:
  /__w/gem5/gem5/configs/ruby/Ruby.py(334): send_evicts
  /__w/gem5/gem5/configs/ruby/MI_example.py(92): create_system
  <string>(1): <module>
  /__w/gem5/gem5/configs/ruby/Ruby.py(258): create_system
  /__w/gem5/gem5/configs/example/garnet_synth_traffic.py(156): <module>
  src/python/m5/main.py(669): main

```

First appears in 194: https://github.com/gem5/gem5/actions/runs/7097419149
Rev: f00d7f70a4c88ff6b3ee703cb09ff33cc617f74f

(( Obviously it did not appear in 193 https://github.com/gem5/gem5/actions/runs/7083272431
Rev: 895944fa27c2e84a27705f814e4782357df7514b )))

The candidates I get via `git log --oneline 895944fa27c2e84a27705f814e4782357df7514b..f00d7f70a4c88ff6b3ee703cb09ff33cc617f74`:

f00d7f70a4 #651
569e21f798 #241
7b98641953 #635

I think #651 is almost certainly the change at fault here.

I attempted to fix this bug with #695.

The first fix is to not check for `DerivO3CPU` typing if it's not compiled into the binary (5d09ff452580c5a42b9a2b3a825e2827f2551003). Though once is done the following error is returned:

```sh
X86TimingSimpleCPU is not a valid sub-class of BaseCPU.
Error: could not create sytem for ruby protocol MI_example
KeyError: 'X86TimingSimpleCPU'

At:
  /home/bbruce/Desktop/gem5/configs/common/ObjectList.py(72): get
  /home/bbruce/Desktop/gem5/configs/common/ObjectList.py(167): get_isa
  /home/bbruce/Desktop/gem5/configs/ruby/Ruby.py(337): send_evicts
  /home/bbruce/Desktop/gem5/configs/ruby/MI_example.py(92): create_system
  <string>(1): <module>
  /home/bbruce/Desktop/gem5/configs/ruby/Ruby.py(258): create_system
  configs/example/ruby_mem_test.py(148): <module>
  src/python/m5/main.py(669): main
```

I suspect the reason for this bug is somewhere in "ruby_mem_test.py" the cpu is default set to `X86TimingSimpleCPU` and, as the tests use `NULL/gem5.opt`, there's a failure. My "quick-fix" for this was to run the tests using X86 which I did in PR #695 with commit b22ca02a65927f8f3fe0bac0cce4e33100595028.

This fixed some of the tests by not all. The following is a copy and paste of a comment on PR #695 (https://github.com/gem5/gem5/pull/695#issuecomment-1861913795):

So, even with this change the following are still failing:

```sh
Test: ruby_mem_test-simple-X86-x86_64-opt-MESI_Two_Level Failed
Test: ruby_mem_test-simple-extra-X86-x86_64-opt-MESI_Two_Level Failed
Test: ruby_mem_test-simple-extra-multicore-X86-x86_64-opt-MESI_Two_Level Failed
```

To reproduce:

```sh
# Get this PR. Alt: `gh pr checkout 695` (https://cli.github.com)
git clone -b fix-daily-mem-tests git@github.com:BobbyRBruce/gem5.git gem5-fix-mem
cd gem5-fix-mem

# Compile
scons build/X86_MESI_Two_Level/gem5.opt -j`nproc`

# ruby_mem_test-simple-X86-x86_64-opt-MESI_Two_Level
./build/X86_MESI_Two_Level/gem5.opt configs/example/ruby_mem_test.py --abs-max-tick 20000000 --functional 10 --network=simple

#ruby_mem_test-simple-extra-X86-x86_64-opt-MESI_Two_Level
./build/X86_MESI_Two_Level/gem5.opt configs/example/ruby_mem_test.py --abs-max-tick 20000000 --functional 10 --network=simple --simple-physical-channels

#ruby_mem_test-simple-extra-multicore-X86-x86_64-opt-MESI_Two_Level
./build/X86_MESI_Two_Level/gem5.opt configs/example/ruby_mem_test.py --abs-max-tick 20000000 --functional 10 --network=simple --simple-physical-channels --num-cpus=4
```

They all fail in the same manner, with read access failures:

```sh
src/cpu/testers/memtest/memtest.cc:155: panic: Read access failed at 0x10c700
Memory Usage: 738168 KBytes
Program aborted at tick 19096836
--- BEGIN LIBC BACKTRACE ---
./build/X86_MESI_Two_Level/gem5.opt(_ZN4gem515print_backtraceEv+0x30)[0x55c8b8663230]
./build/X86_MESI_Two_Level/gem5.opt(_ZN4gem512abortHandlerEi+0x4c)[0x55c8b8688c7c]
/lib/x86_64-linux-gnu/libc.so.6(+0x42520)[0x7fdf66aa3520]
/lib/x86_64-linux-gnu/libc.so.6(pthread_kill+0x12c)[0x7fdf66af79fc]
/lib/x86_64-linux-gnu/libc.so.6(raise+0x16)[0x7fdf66aa3476]
/lib/x86_64-linux-gnu/libc.so.6(abort+0xd3)[0x7fdf66a897f3]
./build/X86_MESI_Two_Level/gem5.opt(+0xa3d155)[0x55c8b7dbe155]
./build/X86_MESI_Two_Level/gem5.opt(_ZN4gem57MemTest15completeRequestEPNS_6PacketEb+0x3bb)[0x55c8b8fc7dfb]
./build/X86_MESI_Two_Level/gem5.opt(_ZN4gem57MemTest4tickEv+0xa3d)[0x55c8b8fc93fd]
./build/X86_MESI_Two_Level/gem5.opt(_ZN4gem510EventQueue10serviceOneEv+0xc2)[0x55c8b8677bf2]
./build/X86_MESI_Two_Level/gem5.opt(_ZN4gem59doSimLoopEPNS_10EventQueueE+0x68)[0x55c8b86a3e38]
./build/X86_MESI_Two_Level/gem5.opt(_ZN4gem58simulateEm+0x283)[0x55c8b86a4433]
./build/X86_MESI_Two_Level/gem5.opt(+0x1f51510)[0x55c8b92d2510]
./build/X86_MESI_Two_Level/gem5.opt(+0xa2d534)[0x55c8b7dae534]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x128023)[0x7fdf679e6023]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyObject_Call+0x5c)[0x7fdf6799ffec]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x4b16)[0x7fdf67934776]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af)[0x7fdf67a803af]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x9d68)[0x7fdf679399c8]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af)[0x7fdf67a803af]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(PyEval_EvalCode+0xbe)[0x7fdf67a7b3de]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1bd96d)[0x7fdf67a7b96d]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1287b3)[0x7fdf679e67b3]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x69de)[0x7fdf6793663e]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af)[0x7fdf67a803af]
./build/X86_MESI_Two_Level/gem5.opt(+0x13090c7)[0x55c8b868a0c7]
./build/X86_MESI_Two_Level/gem5.opt(main+0x213)[0x55c8b7d583b3]
/lib/x86_64-linux-gnu/libc.so.6(+0x29d90)[0x7fdf66a8ad90]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80)[0x7fdf66a8ae40]
./build/X86_MESI_Two_Level/gem5.opt(_start+0x25)[0x55c8b7d9eab5]
--- END LIBC BACKTRACE ---
For more info on how to address this issue, please visit https://www.gem5.org/documentation/general_docs/common-errors/ 

Aborted (core dumped)
```

My two thoughts here are:

1. These need to be run with "NULL/gem5.opt" in which case the scripts need updated to support this case. The following error is returned in this case:

```sh

X86TimingSimpleCPU is not a valid sub-class of BaseCPU.
Error: could not create sytem for ruby protocol MI_example
KeyError: 'X86TimingSimpleCPU'

At:
  /home/bbruce/Desktop/gem5/configs/common/ObjectList.py(72): get
  /home/bbruce/Desktop/gem5/configs/common/ObjectList.py(167): get_isa
  /home/bbruce/Desktop/gem5/configs/ruby/Ruby.py(337): send_evicts
  /home/bbruce/Desktop/gem5/configs/ruby/MI_example.py(92): create_system
  <string>(1): <module>
  /home/bbruce/Desktop/gem5/configs/ruby/Ruby.py(258): create_system
  configs/example/ruby_mem_test.py(148): <module>
  src/python/m5/main.py(669): main
```

2. This should be fun for X86, and therefore there's something wrong with the test?

BobbyRBruce, 2023-12-20:
PR #695 has been updated and now fixes this issue.

Harshil2107, 2023-12-21:
Note: Fixed by #704

closed 2023-12-21 by Harshil2107