gem5/gem5 #700: the stat.txt generated by gem5 does not include statistics for vector memory access operations such as ‘VectorUnitStrideLoad’?

icuuuuuuer, 2023-12-20:
**Describe the bug**
When running SIMD instructions on a baseminor CPU with the default configuration in gem5, which utilizes the ARM-v8 architecture with Scalable Vector Extension (SVE), I observed that after compilation, there were vector memory access instructions. However, upon completion, the stat.txt file did not show any statistics related to vector unit memory accesses, such as “VectorUnitStrideLoad,” “VectorUnitStrideStore,” “VectorUnitStrideMaskLoad,” “VectorUnitStrideMaskStore,” “VectorStridedLoad,” “VectorStridedStore,” “VectorIndexedLoad,” or “VectorIndexedStore.” All these statistics were reported as 0. Nevertheless, it is indeed true that there are vector memory access instructions from the SVE instruction set.

**Affects version**
newest version

**gem5 Modifications**
just use simple baseminorcpu

**Additional information**
I want to conduct some experiments with vector processors using gem5. Below are the C source files, disassembly files, as well as the gem5 configuration file and the stat.txt output. Is it possible that all vector unit operations are being categorized as ‘memread’ instead of specific categories like ‘vectorunitstrided’?
[config.json](https://github.com/gem5/gem5/files/13722386/config.json)
[stats.txt](https://github.com/gem5/gem5/files/13722388/stats.txt)
[vect.txt](https://github.com/gem5/gem5/files/13722389/vect.txt)
[image]

Harshil2107, 2023-12-20:
Hi @icuuuuuuer,

Could you give more information on how to reproduce this issue? Please use code blocks instead of screenshots as code blocks are easier to read and code can be copy pasted in case any modifications are needed.

icuuuuuuer, 2023-12-24:
> Hi @icuuuuuuer,
> 
> Could you give more information on how to reproduce this issue? Please use code blocks instead of screenshots as code blocks are easier to read and code can be copy pasted in case any modifications are needed.

Hi,
okay.You can reproduce this issue with the following minimal configuration, which includes caches.py and two_level.py
caches.py:
```python
from m5.objects import Cache

class L1Cache(Cache):
    assoc = 2
    tag_latency = 2
    data_latency = 2
    response_latency = 2
    mshrs = 4
    tgts_per_mshr = 20

class L1ICache(L1Cache):
    size = '16kB'

class L1DCache(L1Cache):
    size = '64kB'

class L2Cache(Cache):
    size = '256kB'
    assoc = 8
    tag_latency = 20
    data_latency = 20
    response_latency = 20
    mshrs = 20
    tgts_per_mshr = 12

def connectCPU(self, cpu):
    # need to define this in a base class!
    raise NotImplementedError

def connectBus(self, bus):
    self.mem_side = bus.cpu_side_ports

class L1ICache(L1Cache):
    size = '16kB'

    def connectCPU(self, cpu):
        self.cpu_side = cpu.icache_port

class L1DCache(L1Cache):
    size = '64kB'

    def connectCPU(self, cpu):
        self.cpu_side = cpu.dcache_port

def connectCPUSideBus(self, bus):
    self.cpu_side = bus.mem_side_ports

def connectMemSideBus(self, bus):
    self.mem_side = bus.cpu_side_ports
```
two_levels.py:
```python
import m5

# import all of the SimObjects
from m5.objects import *
from gem5.runtime import get_runtime_isa

# Add the common scripts to our path
m5.util.addToPath("../../")

# import the caches which we made
from caches import *

# import the SimpleOpts module
from common import SimpleOpts

# Default to running 'hello', use the compiled ISA to find the binary
# grab the specific path to the binary
thispath = os.path.dirname(os.path.realpath(__file__))
default_binary = os.path.join(
    thispath,
 #   "../../../",
    "/home/sve_test/vect.out",
)

# Binary to execute
SimpleOpts.add_option("binary", nargs="?", default=default_binary)

# Finalize the arguments and grab the args so we can pass it on to our objects
args = SimpleOpts.parse_args()

# create the system we are going to simulate
system = System()

# Set the clock frequency of the system (and all of its children)
system.clk_domain = SrcClockDomain()
system.clk_domain.clock = "1GHz"
system.clk_domain.voltage_domain = VoltageDomain()

# Set up the system
system.mem_mode = "timing"  # Use timing accesses
system.mem_ranges = [AddrRange("512MB")]  # Create an address range

# Create a simple CPU
system.cpu = ArmMinorCPU()

# Create an L1 instruction and data cache
system.cpu.icache = L1ICache(args)
system.cpu.dcache = L1DCache(args)

# Connect the instruction and data caches to the CPU
system.cpu.icache.connectCPU(system.cpu)
system.cpu.dcache.connectCPU(system.cpu)

# Create a memory bus, a coherent crossbar, in this case
system.l2bus = L2XBar()

# Hook the CPU ports up to the l2bus
system.cpu.icache.connectBus(system.l2bus)
system.cpu.dcache.connectBus(system.l2bus)

# Create an L2 cache and connect it to the l2bus
system.l2cache = L2Cache(args)
system.l2cache.connectCPUSideBus(system.l2bus)

# Create a memory bus
system.membus = SystemXBar()

# Connect the L2 cache to the membus
system.l2cache.connectMemSideBus(system.membus)

# create the interrupt controller for the CPU
system.cpu.createInterruptController()
#system.cpu.interrupts[0].pio = system.membus.mem_side_ports
#system.cpu.interrupts[0].int_requestor = system.membus.cpu_side_ports
#system.cpu.interrupts[0].int_responder = system.membus.mem_side_ports

# Connect the system up to the membus
system.system_port = system.membus.cpu_side_ports

# Create a DDR3 memory controller
system.mem_ctrl = MemCtrl()
system.mem_ctrl.dram = DDR3_1600_8x8()
system.mem_ctrl.dram.range = system.mem_ranges[0]
system.mem_ctrl.port = system.membus.mem_side_ports

system.workload = SEWorkload.init_compatible(args.binary)

# Create a process for a simple "Hello World" application
process = Process()
# Set the command
# cmd is a list which begins with the executable (like argv)
process.cmd = [args.binary]
# Set the cpu to use the process as its workload and create thread contexts
system.cpu.workload = process
system.cpu.createThreads()

# set up the root SimObject and start the simulation
root = Root(full_system=False, system=system)
# instantiate all of the objects we've created above
m5.instantiate()

print("Beginning simulation!")
exit_event = m5.simulate()
print("Exiting @ tick %i because %s" % (m5.curTick(), exit_event.getCause()))
```
The file vect.out is obtained by compiling the following C language files using the gcc O3 option.

vect.c:
```C
// vect.c
#include<stdio.h>
int main()
{
  int N = 1000;
  int a[N], b[N], c[N];
  for (int i = 0; i < N; i++) {
    b[i]=c[i]=i;
    for (int j=0;j<500;j++){
      a[i] = b[i] * c[i];
    }
    a[i] = b[i] + c[i];
  }

  for (int i = 0; i < N; i++) {
   a[i] = a[i] + a[i];
  }

   printf("%d\n",a[100]);
   return 0;
}
```
After compiling this C language file, it should include vector memory access operations. However, when using this CPU configuration, I did not observe VectorUnitStrideLoad or similar vector memory access operations in the stat.txt file. Only memread operations were recorded. I have attempted more complex out-of-order CPU configurations with vector memory access units and benchmarks containing vector memory access operations, but still did not see statistics for VectorUnitStrideLoad or similar operations—only memread. I believe that even with the simplest baseminor CPU configuration and a handwritten C language file, this issue should be evident.

MAyang38, 2024-01-07:
> > Hi @icuuuuuuer,
> > Could you give more information on how to reproduce this issue? Please use code blocks instead of screenshots as code blocks are easier to read and code can be copy pasted in case any modifications are needed.
> 
> Hi, okay.You can reproduce this issue with the following minimal configuration, which includes caches.py and two_level.py caches.py:
> 
> ```python
> from m5.objects import Cache
> 
> class L1Cache(Cache):
>     assoc = 2
>     tag_latency = 2
>     data_latency = 2
>     response_latency = 2
>     mshrs = 4
>     tgts_per_mshr = 20
> 
> 
> class L1ICache(L1Cache):
>     size = '16kB'
> 
> class L1DCache(L1Cache):
>     size = '64kB'
> 
> class L2Cache(Cache):
>     size = '256kB'
>     assoc = 8
>     tag_latency = 20
>     data_latency = 20
>     response_latency = 20
>     mshrs = 20
>     tgts_per_mshr = 12
> 
> def connectCPU(self, cpu):
>     # need to define this in a base class!
>     raise NotImplementedError
> 
> def connectBus(self, bus):
>     self.mem_side = bus.cpu_side_ports
> 
> class L1ICache(L1Cache):
>     size = '16kB'
> 
>     def connectCPU(self, cpu):
>         self.cpu_side = cpu.icache_port
> 
> class L1DCache(L1Cache):
>     size = '64kB'
> 
>     def connectCPU(self, cpu):
>         self.cpu_side = cpu.dcache_port
> 
> def connectCPUSideBus(self, bus):
>     self.cpu_side = bus.mem_side_ports
> 
> def connectMemSideBus(self, bus):
>     self.mem_side = bus.cpu_side_ports
> ```
> 
> two_levels.py:
> 
> ```python
> import m5
> 
> # import all of the SimObjects
> from m5.objects import *
> from gem5.runtime import get_runtime_isa
> 
> # Add the common scripts to our path
> m5.util.addToPath("../../")
> 
[... 105 lines cut ...]
> {
>   int N = 1000;
>   int a[N], b[N], c[N];
>   for (int i = 0; i < N; i++) {
>     b[i]=c[i]=i;
>     for (int j=0;j<500;j++){
>       a[i] = b[i] * c[i];
>     }
>     a[i] = b[i] + c[i];
>   }
> 
>   for (int i = 0; i < N; i++) {
>    a[i] = a[i] + a[i];
>   }
> 
>    printf("%d\n",a[100]);
>    return 0;
> }
> ```
> 
> After compiling this C language file, it should include vector memory access operations. However, when using this CPU configuration, I did not observe VectorUnitStrideLoad or similar vector memory access operations in the stat.txt file. Only memread operations were recorded. I have attempted more complex out-of-order CPU configurations with vector memory access units and benchmarks containing vector memory access operations, but still did not see statistics for VectorUnitStrideLoad or similar operations—only memread. I believe that even with the simplest baseminor CPU configuration and a handwritten C language file, this issue should be evident.
> 
> Thank you very much!

You have what looks like a problem with the caches.py file. You can compare the official caches.py file and the location of the functions in your class seems different

powerjg, 2024-01-12:
Does #589 solve this problem?

aarmejach, 2024-02-09:
I just realized the issue specifies SVE loads/stores. #589 fixes this for RISV-V V but not for Arm SVE. The decoder for SVE marks all load/stores operations as MemRead/MemWrite. So in order to make finer grain classification the opClasses in the decoder of Arm should be changed.

ivanaamit, 2024-04-15:
@aarmejach, @icuuuuuuer, could you please let us know if this is still a bug after #589 has been merged? If there is still a bug, and you have a fix for it, please feel free to contribute with it. Thank you.

ivanaamit, 2024-10-18:
Since we haven't heard back from the author in a while, I'm closing this issue as it may no longer be relevant. Feel free to reopen it anytime if needed. Thanks!

closed 2024-10-18 by ivanaamit