gem5/gem5 #703: Deprecate GCN3 and move exclusively to Vega / gfx9

BobbyRBruce, 2023-12-20:
### Discussed in https://github.com/orgs/gem5/discussions/306

<div type='discussions-op-text'>

<sup>Originally posted by **abmerop** September 12, 2023</sup>
I'd like to propose deprecating the GCN3 ISA and moving exclusively to Vega.  Vega is essentially a superset ISA of GCN3, meaning it contains everything GCN3 has and more.  As a result of this, there is an enormous amount of redundant code. gem5 also now supports an APU model with Vega as well as a dGPU so users can simulate either one, just like GCN3. 

The plan would be as follows:
- Update all GPU gem5-resources Makefiles to build the gfx902 target (Vega APU).  gfx900 (dGPU) is already being built.
- Ensure that all GPU gem5-resources tests work as well or better with Vega compared to GCN3.
- Update documentation for removal references to GCN3.
- Change the default --gfx-version to gfx902 or gfx900.
   - I think if we choose gfx900, we should also implicitly set the --dgpu flag as this is no gfx900 APU.
- Delete GCN3 related files.

The benefits would be:
- Removes 90k lines of code from the gem5 source code.
- Less applications to test and one less build target to test.
- A single GPU ISA means GPU can be built into build/ALL/gem5.* further reducing testing (I think - Ruby protocol might be an issue).
- gfx9 is the current ISA used in Radeon™ Instinct™ products up to and including the most recent hardware - So we would be simulating *current* products.

Will add more to plan/benefits if I think of anything else, but wanted to get the discussion started.</div>

abmerop, 2023-12-21:
I was talking to @mattsinc about this and i think the hang up was DNNMark. However, we discussed that it is working on gfx902 so I think this plan is a go.

mattsinc, 2024-01-02:
This is fine with me, if @abmerop confirmed that.

abmerop, 2024-01-03:
Alright, I will put together a PR soon

abmerop, 2024-01-17:
Added PR #781 .  As far as the list of items goes (I am not sure how to make one of those checklists):  gem5-resources have been updated to build gfx902.  We have confirmed everything is working on gfx9 that was working on gfx8.  In that PR, I change the default gfx version to gfx902 and remove the files.  If we all agree on that last point the last step is to update documentations.

powerjg, 2024-01-17:
```
- [ ] This is how you make a checkbox
- [ ]
```

abmerop, 2024-03-22:
Believe this is done

closed 2024-03-22 by abmerop