gem5/gem5 #709: Transition to Bazel build system

rw1nkler, 2023-12-22:
Some time ago, a [discussion](https://www.mail-archive.com/gem5-dev@gem5.org/msg39686.html) was started on the gem5-dev mailing channel regarding simplifying the project's build system. This discussion mentioned a desire to replace [SCons](https://scons.org/) with [Bazel](https://bazel.build/) at some point in the future:

> One thing I feel is worth noting is that there is a desire to move away
from SCons at some point (likely towards bazel: https://bazel.build/). This
is definitely not something we could accomplish in the coming months, as I
believe, at the very least, it would require breaking gem5 further down
into separate modules; but it's something you should be aware is on some
peoples' minds.

Is this approach still desirable and has any work been started to modify the build system? I noticed that some attempts to provide Bazel build rules were reviewed on [Gerrit #58949](https://gem5-review.googlesource.com/c/public/gem5/+/58949). Will this effort be continued?

powerjg, 2024-01-11:
As far as I know, no one is actively working on this.

rw1nkler, 2024-01-11:
@powerjg Could you please provide more information on whether Bazel support is still desired? Do you know if any contributions in that area will be accepted, or if development will focus solely on the SCons build system in the future?

powerjg, 2024-02-01:
Oof, sorry that it's been so long. I just noticed this dropped through the cracks.

Let me try to brain dump here.

In my view, we want a build system that has the following "requirements". These are somewhat in order of importance (in my opinion.
- **Is "standard."** I.e., when someone googles for a problem they can find some help that we don't have to write. I think that scons, especially how we use it, absolutely fails at this requirement. The documentation for scons is hard to come by, and it usually doesn't apply to gem5 because we've customized it so much. It would be great to use the "most common" build system for C++/python. Unfortunately, I'm not sure there is a "most common" build system or if the common build systems (e.g., autoconf and cmake) actually support the other features we need.
- **Can deal with code generation.** We have a lot of code generation in gem5. E.g., the `isa_parser` generates the ISA files, SLICC generates Ruby protocol files, and scons generates a bunch of files including C++ wrappers for SimObjects, enums, and possibly other thing.
- **Easy to understand build process.** Again, scons is really failing us here. If you want to figure out how a particular `.o` file is generated or how the binary is linked, it's nearly impossible. You have to look at the SConstruct file, multiple SConscript files, many different files in `site_scons` and who knows what else. I've been working on this project for 13 years and I still don't understand how the build system works! Something that's simpler and doesn't implement as many "special cases" outside the default for the build tool would be wonderful.
- **Reproducible build process.** Right now, scons does not do a good job of running on lots of different systems and getting the same results. We often run into issues where it works on one platform and not another (this has gotten better since we implemented compiler tests, but it's still a maintenance overhead). Bazel really excels at this.

Another thing that I've been thinking for many years now is that I would like to see gem5 compiled as many different shared objects instead of just one giant (sometime ~1GiB) binary. Given the way Bazel works (at least the last time I looked at it many years ago), it seemed like transitioning to a new build system and modularizing gem5 would go hand-in-hand.
**However,** gem5 is *highly* coupled. Things in `base/` depend on `sim/` which depends on `mem/` which depends on `base/` and `sim/`. There are even circular dependences within a single file! It's a huge mess and will be a large undertaking to shake out. That said, it would also make testing, debugging, and developing gem5 so much better if we were able to do this!

In conclusion, my thoughts are this:
1. I can't stand (our esoteric version of) scons, and I want a replacement. I think we should carefully consider if bazel is the right choice or if something else is better. My biggest concern with bazel is that in 20 years will it be supported? gem5 chose scons 20 years ago in part because it was the new shiny thing (I wasn't there, so I'm speculating), and scons support 20 years later is generally not very good.
2. Transitioning the build system will be a large undertaking and it may make sense to do other things to clean up the codebase at the same time (e.g., making gem5 into modules with less coupling).

andysan, 2024-02-05:
I completely agree with Jason's reasoning here. We definitely need to go with something that is well-known and has a large community. My impression is that most large C++ projects uses cmake these days, but meson seems popular as well.

I once prototyped cmake support in gem5 and it is definitely doable to implement all of the code generation we need, especially after Gabe and others factored out much of it to separate Python tools. IIRC, my prototype that built the NULL target took 1-2 days to implement (and that included extracting the code generation tools from scons) and there was no need to restructure the code base. I think someone who is familiar with cmake could have a working prototype within a couple of weeks. The challenge will probably be all of the tricks we use to detect features (kvm etc.) and handling quirky compilers/systems.

rw1nkler, 2024-02-22:
Thank you @powerjg and @andysan for taking the time to provide a detailed answers to my questions