gem5/gem5 #719: Flags Request::READ_MODIFY_WRITE and Request::ACQUIRE are the same

tiagormk, 2023-12-27:
The Request::READ_MODIFY_WRITE and Request::ACQUIRE flags are both 0x00020000:
https://github.com/gem5/gem5/blob/025ccadc6823eff93ee9e0a20712000978bbc53e/src/mem/request.hh#L161C15-L161C15

Is there a reason behind this or it's a bug ?

This doesn't break functionality, but the Ruby sequencer will check loads with Request::READ_MODIFY_WRITE set and will treat them as a store, which is not the intended behavior for a load-acquire.

powerjg, 2024-01-04:
I'm pretty sure this is a bug. Here is where it was introduced: https://gem5-review.googlesource.com/c/public/gem5/+/48710

It looks like the ACQUIRE flag was overlooked. I don't see any problem picking a different unused bit for `READ_MODIFY_WRITE`

ivanaamit, 2024-03-07:
@tiagormk, just following up on this. Does Jason's suggested change of selecting a different unused bit for `READ_MODIFY_WRITE` make sense?

tiagormk, 2024-03-08:
Hi @ivanaamit , apologies for the lack of responsiveness. The solution makes sense. I will push a patch with the fix.

tiagormk, 2024-03-08:
https://github.com/gem5/gem5/pull/922

closed 2024-03-08 by tiagormk