gem5/gem5 #751: Modify gem5 to pass an object upon an exit event instead of a string

BobbyRBruce, 2024-01-09:
I've resurrected this issue from our old Jira service: https://gem5.atlassian.net/browse/GEM5-1128

This task is mostly related in how the gem5 simulator returns an exit event event. At present the only meaningful information communicated is the exit event cause string. In this Simulator package we parse this and classify exit events into broad categories. gem5 should be modified to return an object containing information on how an exit event was triggered, at what tick and instruction, and what kind of exit event was triggered (user interrupt, tick/instruction count reached, CPU switch request, etc).
 
This change will require considerable refactoring of the Simulation package.

> **Commend on this issue by @andysan** : https://gem5.atlassian.net/browse/GEM5-1128?focusedCommentId=13376
>
>My original long-term plan when rewriting the wrappers in PyBind was to to use a mechanism resembling exceptions where each exit is an object derived from GlobalSimLoopExitEvent. This would have a few nice benefits:
>
> * It’d make it possible to add custom data to exits (e.g., a work item ID).
> * It makes it possible to use the exit object as an exception in Python. This could be useful * when handling an unexpected simulator exit in m5.drain (these are currently dropped silently).
> * It makes exit code matching much less error prone since you end up doing isinstance(event, MyExitEvent) checks instead of string matching.
>* Properly exposing global events (not just exit events) to Python would make it possible to build custom global events implemented in Python. We currently only support this for local events. This could, for example, be used to schedule stat dumps without exiting from the simulation loop.
>
>The main reason I didn't export the constructor of GlobalSimLoopExitEvent is that the original SWIG code had issues with a memory leak. I think the main issue was that GlobalSimLoopExitEvent objects are normally owned by the C++ side and not returned as smart pointers.
>
>The first step would be to fix the memory leak. Should be reasonably easy without changing the Python API. I think we'd need the following changes:
>
>* C++: Change/add `GlobalSimLoopExitEvent that allows event creation without scheduling
>* C++: Remove the exitSimLoop helper function and replace it with a makeExitEvent helper that returns a shared pointer.
>* PyBind: Export more GlobalSimLoopExitEvent members for scheduling etc.
>* PyBind: Update the GlobalSimLoopExitEvent bindings to use reference counting
>
>Generalising this further would require Python bindings for GlobalEvent. I think the easiest way to make this work reliably would be to disallow instantiation of GlobalEvent from Python and create a Python-specific base class (we use this trick for normal events, see PyEvent). The unfortunate consequence of this trick is that exit events from C++ would have GlobalSimLoopExitEvent as their base, but that wouldn’t be the case for exit events implemented in Python. It might make sense to create a separate PyGlobalSimLoopExitEvent class that patches up that discrepancy

Modifying exit events to return objects instead of strings was also discussed here: https://gem5-review.googlesource.com/c/public/gem5/+/66231

ivanaamit, 2024-04-18:
- [ ] #474

erin-le, 2025-03-27:
Most of the objectives discussed here have been met by the PRs for the (cancelled) v24.1.1.0 release.

closed 2025-03-27 by erin-le