gem5/gem5 #753: Sv39 and Sv48 in gem5

hnpl, 2024-01-09:
### Discussed in https://github.com/orgs/gem5/discussions/752

<div type='discussions-op-text'>

<sup>Originally posted by **hnpl** January  9, 2024</sup>
gem5 is using Sv39 for address translation [1][2][4] and for PTE structure [3] (albeit, PTE layouts of Sv39, SV48 and Sv57 are identical except for the PPN layouts). This means, only 39 bits of a virtual address are meaningful [1][4] (i.e., the vaddr will be sign-extended using bit 38). However, in the device tree of RISC-V, gem5 is reporting support for Sv48, so the Linux kernel will assume Sv48 for address translation.

From the ISA manual, hardware supporting Sv48 must be compatible with Sv39 [5], and not the other way around.
This issue probably doesn't affect functional correctness if the memory size is smaller than 512GiB. However, this affect the correctness of TLB stats if we are assuming 4-level page (Sv48).

[1] https://github.com/gem5/gem5/blob/v23.1.0.0/src/arch/riscv/pagetable_walker.cc#L454-L459
[2] https://github.com/gem5/gem5/blob/v23.1.0.0/src/arch/riscv/pagetable.hh#L59
[3] https://github.com/gem5/gem5/blob/v23.1.0.0/src/arch/riscv/pagetable.hh#L94
[4] https://github.com/gem5/gem5/blob/v23.1.0.0/src/arch/riscv/tlb.cc#L284
[5] https://github.com/riscv/riscv-isa-manual/blob/riscv-isa-release-056b6ff-2023-10-02/src/supervisor.adoc?plain=1#L1513</div>

powerjg, 2024-01-26:
We should modify the default device trees for RISC-V to specify using Sv39 instead of Sv48.

We should also implement Sv48 at some point in the gem5 PTW.

qqli579, 2024-06-06:
I would like to ask if this is already in the development plan of gem5. I recently encountered the need to use sv48 in gem5 in a project

hnpl, 2024-06-06:
There's currently no active refactoring of the riscv address translation that I'm aware of. Making gem5 work with sv48 is probably straightforward, but ideally, gem5 should allow users to choose between sv39/sv48/sv57. It would be great if this issue is addressed!