gem5/gem5 #785: Prefetcher coverage and accuracy calculation is incorrect for classic cache

AbhishekUoR, 2024-01-18:
**Describe the bug**
1. The pfUseful statistic is only accounted on ppHit/ppMiss at recvTimingReq(PacketPtr pkt) function call. When a cache is an exclusive from its upstream cache, it invalidates the block at BaseCache::maintainClusivity(bool from_cache, CacheBlk *blk), which is called inside BaseCache::access(PacketPtr pkt, CacheBlk *&blk, Cycles &lat, PacketList &writebacks) that deletes the block before it can be accounted as useful prefetch block.

2. The demandMshrMisses does not account for the demand requests that have arrived at the cache and are missed. On miss, the MSHR is looked up, but if an MSHR packet is created due to prefetcher (late prefetch), the demandMshrMisses counter does not account for it.

3. The clearPrefetched() is not called on blocks for which a request is generated to go to the downstream cache due to not having the correct coherence state. This leads to additional pfUseful counts on WriteReqs (L1D cache).

**Affects version**
It was found in Version 23.0.0.1

**gem5 Modifications**
1. pfUseful: I have included new counters inside the incHitCount function call, but I can include ppHit in it as this will notify the prefetcher on Hit and perform training for a few algorithms. Similarly, ppMiss can be included on incMissCount for prefetched load blocks that need writable blocks and account for pfUsefulButMiss.

2. demandMshrMisses: I have added a check at void BaseCache::handleTimingReqMiss(PacketPtr pkt, MSHR *mshr, CacheBlk *blk, Tick forward_time, Tick request_time) to check the mshr target source and account for missed out demandMshrMisses due to late prefetch.

3. clearPrefetched(): I am clearing the flag at void Cache::serviceMSHRTargets(MSHR *mshr, const PacketPtr pkt, CacheBlk *blk) under this condition "if (blk && blk->isValid() && (!mshr->isForward || !pkt->hasData())) {.."

**To Reproduce**

1. Compile gem5 with the command:` scons build/ARM/gem5.opt`
2. Execute the simulation with: `./build/ARM/gem5.opt configs/deprecated/example/se.py -c (array_access_binary_arm) --cpu-type=DerivO3CPU --caches --l1d-hwp-type TaggedPrefetcher --l2-hwp-type TaggedPrefetcher`

If possible, please include the Python configuration script used and state clearly any parameters passed.
Changed clusivity in src/mem/cache/Cache.py to mostly_excl for L2 cache

**Host Operating System**
Rede Hat

**Host ISA**
 X86

ivanaamit, 2024-02-05:
@AbhishekUoR, do you have a fix for this issue? If yes, please feel free to contribute. Thank you.

ivanaamit, 2024-03-07:
Hi @AbhishekUoR, following up on this. Do you have a fix for this issue? Thanks.

AbhishekUoR, 2024-03-19:
> Hi @AbhishekUoR, following up on this. Do you have a fix for this issue? Thanks.

Hello, I have some changes, but they're more like temporary fixes. This month, I'm occupied with my work.

erin-le, 2025-01-30:
@AbhishekUoR Hello, would you still be interested in working on this issue? If so, could you try reproducing it on the latest version of gem5, since the bugs were reported for gem5 v23.0.0.1? Thank you!