gem5/gem5 #788: Q: highly likely incorrect aarch64 ldaxrh instruction  disassembly

alsam, 2024-01-19:
Hello, the code in the latest stable trunk  `src/arch/arm/insts/mem64.cc`

```c++
std::string
MemoryEx64::generateDisassembly(
        Addr pc, const loader::SymbolTable *symtab) const
{
    std::stringstream ss;
    printMnemonic(ss, "", false);
    printIntReg(ss, dest);
    ccprintf(ss, ", ");
    printIntReg(ss, result);
    ccprintf(ss, ", [");
    printIntReg(ss, base);
    ccprintf(ss, "]");
    return ss.str();
}
```
generates a disassembly for the aarch64 instruction `ldaxrh` e.g.
```asm
ldaxrh w10, wzr, [w0]
```
with 3 operands instead of 2 follow the ARM reference.
Android toolchain fails to assembly the generated disassembly as expects two operands for the instruction instead of given 3.

I would rather expect to see here:

```asm
ldaxrh w10, [w0]
```

giactra, 2024-02-15:
Disassembling in gem5 has some known issues... I am sure the reason why we are printing in that format is because MemoryEx64 is shared between load and store exclusive. The clean solution would be to provide a different generateDisassembly implementation for load exclusive.

This should definitely be fixed so I am keeping the issue open, but I regard this as low priority: I will deal with it when I'll have some free time. FYI To mitigate the annoyance of dealing with an incorrect disassembly we integrated gem5 with the capstone disassembler: https://github.com/gem5/gem5/pull/494

Hope this helps.

alsam, 2024-02-16:
Thank you @giactra !