gem5/gem5 #798: Drop support for Ubuntu 20.04

ivanaamit, 2024-01-23:
We plan to drop support for Ubuntu 20.04 in the upcoming v24.0.

pranith, 2024-01-25:
Shouldn't there be a period of deprecation before we drop support?

powerjg, 2024-01-25:
Yeah, the problem is that we support *too many compilers and libraries*. We officially support at least [14 major versions](https://github.com/gem5/gem5/blob/stable/.github/workflows/compiler-tests.yaml#L19). This is a maintenance overhead.

If we distributed gem5 as a binary, and we only had to compile gem5 for Ubuntu 20.04 once, it would be no problem at all to continue support. The difficulty is how often the compiler tests break and the maintenance overhead of making it possible for everyone to compile gem5 on all systems. #555 is a good example of the painfulness of this process. We've also recently had to add annoying special cases in the code for specific versions of compilers (e.g., #696 which was caused by #636 which was caused by #555)

Basically, there's only so much time in the day to work on development of gem5. The question is: should we spend the time ensuring that 20.04 continues to work for another year, or should we spend that time on something else. Given the 53 open issues at this moment, I believe there are higher priorities than upstream gem5 continuing to support 20.04.

Now, dropping support for 20.04 doesn't mean that gem5 will stop working overnight on machine with 20.04. It just means that we may stop doing tests on 20.04 and end up requiring a compiler or library version that is not default on 20.04. Given the prevalence of docker now, it seems pretty straightforward to use a new ubuntu distribution on an older distribution via docker containers.

It's a tradeoff with no perfect solution.

andysan, 2024-01-25:
I can totally see how it would simplify lots of things if we could just ditch older platforms more quickly. From a purely selfish perspective, I wouldn't mind dropping support for 20.04. I don't think any of my colleagues relies on that version of Ubuntu. I'm more worried about breaking other peoples' flows.

Do we have any data on the type of platform people actually use? After all, fixing bugs isn't very helpful if people can't actually run the simulator. I'd suggest running a user poll to try to gauge the impact before we pull the plug. It'd be interesting to know which platforms people actually use. For example, is anyone using Suse? Or plain Debian? Or do we have  a large user base on macOS? That'd help us make a more informed decision.

closed 2024-04-12 by ivanaamit