gem5/gem5 #807: Binary's compiled on Apple Silicon (ARM) are being considered processed as Power ISA binaries in SE mode

BobbyRBruce, 2024-01-25:
**Describe the bug**

From what I can observe, when a binary is compiled on an Apple Silicon device, and this binary is then passed into an ARM SE-mode simulation, the simulation crashes. At least one cause of this is the binary appears to have be cast as a `PowerSeWorkload`.

**Affects version**

I found this bug in [v23.1, on gem5 stable branch](https://github.com/gem5/gem5/releases/tag/v23.1.0.0).

**To Reproduce**

1. Compile gem5:

```shell
scons build/ALL/gem5.opt -j <num compilation threads>
```

2. Compile a binary on Apple Silicon. Here I provide a precompiled Apple Silicon program ("program-apple-silicon") and its source ("program.c"): 
[apple-silicon-program.zip](https://github.com/gem5/gem5/files/14058165/apple-silicon-program.zip).

4. Run the binary in SE-mode, targeting the ARM ISA.

I use the following configuration, using the program compiled in step (2.):

```py
from gem5.isas import ISA
from gem5.resources.resource import BinaryResource
from gem5.components.memory import SingleChannelDDR3_1600
from gem5.components.processors.cpu_types import CPUTypes
from gem5.components.boards.simple_board import SimpleBoard
from gem5.components.cachehierarchies.classic.no_cache import NoCache
from gem5.components.processors.simple_processor import SimpleProcessor
from gem5.simulate.simulator import Simulator

cache_hierarchy = NoCache()
memory = SingleChannelDDR3_1600(size="32MB")
processor = SimpleProcessor(cpu_type=CPUTypes.TIMING, isa=ISA.ARM, num_cores=1)

board = SimpleBoard(
    clk_freq="3GHz",
    processor=processor,
    memory=memory,
    cache_hierarchy=cache_hierarchy,
)
      
board.set_se_binary_workload(
    BinaryResource(local_path="program-apple-silicon")
)

simulator = Simulator(board=board)
simulator.run()
```

```sh
./build/ALL/gem5.opt <config>
```

**Terminal Output**

```sh
gem5 Simulator System.  https://www.gem5.org
gem5 is copyrighted software; use the --copyright option for details.

gem5 version 23.1.0.0
gem5 compiled Jan 25 2024 13:25:13
gem5 started Jan 25 2024 14:00:32
gem5 executing on page, pid 12204
command line: ./build/ALL/gem5.opt apple-silicon-prog-run.py

AttributeError: 'NoneType' object has no attribute 'get_arch'

At:
  src/arch/power/PowerSeWorkload.py(44): _is_compatible_with
  src/sim/Workload.py(152): <lambda>
  src/sim/Workload.py(150): find_compatible
  src/sim/Workload.py(163): init_compatible
  src/python/gem5/components/boards/se_binary_workload.py(109): set_se_binary_workload
  apple-silicon-prog-run.py(49): <module>
  src/python/m5/main.py(669): main
```

**Host Operating System**

macOS Sonoma Version 14.2.1

**Host ISA**

Apple Silicon (ARM)

**Compiler used**

clang 15.0.0

andysan, 2024-01-26:
Loading this binary isn't actually supposed to work since the compiled binary is a MACH-O file and not an ELF file. The error message isn't very helpful though. The problem is most likely that `find_compatible` doesn't handle the case where `object_file.create(path)` can't find a valid loader and returns `None` instead of an a loader instance. We should probably raise a `ValueError` in this case which would be consistent with `init_compatible`.

BobbyRBruce, 2024-02-01:
> Loading this binary isn't actually supposed to work since the compiled binary is a MACH-O file and not an ELF file. The error message isn't very helpful though. The problem is most likely that [...]

Ah yes, i was far too caught up in thinking about the ISA without considering other factors.

@ivanaamit has been tasked with looking into making more helpful errors in this code. I'm curious as to if the PowerISA angle is some antiquated code which processes Apple binaries automatically as Power ISA, or if it's just the last condition in an if-else/switch-type statement.

andysan, 2024-02-05:
> Ah yes, i was far too caught up in thinking about the ISA without considering other factors.
>
> @ivanaamit has been tasked with looking into making more helpful errors in this code. I'm curious as to [...]

If memory serves me right, the reason the exception comes from PowerISA is that it is the first ISA in the list. Most implementations of `_is_compatible_with` wouldn't cope correctly with None.

closed 2024-03-04 by ivanaamit