gem5/gem5 #822: How can I specify in the configuration file or command line that my simulation can wait for remote GDB connections before starting simulation

flyeagle0, 2024-01-31:
**Type of issue:** ask a question

**when I run this command in my command line :**
$ ./build/RISCV/gem5.opt myconfigfile.py  
and the following output was obtained
build/RISCV/arch/riscv/linux/se_workload.cc:60: warn: Unknown operating system; assuming Linux.
0: system.remote_gdb: listening for remote gdb on port 7000
#########################......Omit partial output.......##############################
**I guess Gem5 has a built-in gdb server that allows me to connect to Gem5 using gdb to debug the program I simulate on Gem5, rather than using gdb to connect to Gem5 to debug it itself。 Because I know that there is a mechanism in Spike emulators, such as**
$ spike --rbb-port=9824 -m0x10100000:0x20000 rot13-64
Listening for remote bitbang connection on port 9824.
$ openocd -f spike.cfg
$ riscv64-unknown-elf-gdb rot13-64
GNU gdb (GDB) 8.0.50.20170724-git
#########################......Omit partial output.......##############################
Reading symbols from rot13-64...done.
(gdb) target remote localhost:3333
and then I can debug my program running on Spike now

**I saw that the output of GEM5 has a GDB listening port, so I guess GEM5 integrates a GDB server, but I haven't found a way to stop my program so that I can connect to GEM5 using GDB and debug my program**

flyeagle0, 2024-01-31:
I realize this may not be a bug, but I don't know where to remove this tag. I'm very sorry！

ivanaamit, 2024-01-31:
@flyeagle0, if this is just a question, you can post it on [Discussions page](https://github.com/orgs/gem5/discussions). Thank you.

flyeagle0, 2024-01-31:
@ivanaamit Thank you very much. I have already done that. And I would like to ask if I can temporarily put this question here as well.

flyeagle0, 2024-02-01:
@ivanaamit Thank you so much！

flyeagle0, 2024-02-01:
The answer is in discussion [#823](https://github.com/orgs/gem5/discussions/823)

closed 2024-02-01 by flyeagle0