gem5/gem5 #825: Fix daily tests failing due to removing gcn3_x86

Harshil2107, 2024-01-31:
We're encountering daily test failures because the support for GCN3_X86 was removed in change #781 and daily tests are trying to build gem5 with GCN3_X86  to run the gpu tests. To address this issue and restore the functionality of the daily tests, the following steps are necessary:

- [ ] Update gpu docker image to have dependencies for VEGA_X86. The PR for this change is #804 
- [ ] Push this new docker image to cloud
- [ ] Update the binaries used in the tests ( square and heterosync) to be built with VEGA_X86. PR https://github.com/gem5/gem5-resources/pull/22
- [ ] Push these new binaries to the cloud
- [ ] Update the workflow file for daily tests to uses these new images and binaries. PR #817 

These changes should make the daily tests pass.

We should also make the following changes to make sure these binaries are on gem5-resources

- [ ] Create binary resources JSON and add them to the MongoDB database, so that these resources can be used with obtain_resources
- [ ] Update the workflow file to use obtain_resource utility in gem5 to get the resources instead of wget.

The above two changes do not affect daily tests result.

powerjg, 2024-01-31:
This looks like a good overview, @Harshil2107! Thanks for putting it together. Everyone else, could you check this to make sure it looks right to you?

mattsinc, 2024-02-01:
I am not sure about the binary resources at the bottom, but the parts above that seem right.  I guess there is an open question about which binaries we want for all apps.  There are other GPU tests in the weekly runs -- but maybe we are compiling those instead of providing binaries because they use m5ops (for example)?

@abmerop since you took the lead here, it would be great if you can confirm.

abmerop, 2024-02-01:
Looks like lulesh and HACC which i've upload too.

Harshil2107, 2024-02-01:
I merged the workflow file update PR in and re ran the gpu daily tests. They should pass

Harshil2107, 2024-02-09:
Closing this Issue as the tests are passing and have been updated to use obtain_resource instead of wget.

closed 2024-02-09 by Harshil2107