gem5/gem5 #827: RISCV RVV vfredmax.vs and vfredmin.vs implementation is incorrect

debjyoti0891, 2024-02-01:
**Describe the bug**
The implementation of RISCV RVV vfredmax.vs and vfredmin.vs is incorrect. 
- The expected behavior for min(3, NaN) = 3.  Similarly, max(3,NaN) = 3. For vfredmax and vfredmin, if any of the elements in the two source vectors is not-a-NaN, the output should be the max or min of not-a-NaN values.  
- This behaviour is demonstrated by spike as well as qemu. In gem5, this is not true. 

**Affects version**
Branch: develop Commit: ce32d7c523ae5c9612dfd6242092a3630e301898

**To Reproduce**
Test code is shown below. 
- The code loads a quiet NaN (0x7fc00000) into `v9[0]` and splats a constant 1.0 into vector register `v8` and the does a max reduction.
- The result of the reduction in `ft0` is checked as a NaN using ```feq.s   a1,ft0,ft0 \n\```. For gem5, this evaluates to True, *which is an error*.
- For the second part of the code, constant 34.5 is loaded in `ft0` and a max is performed using `fmax.s` between  `ft1` and `ft0`. this results correctly in 34.5 as the value stored in `ft2`.

```
int main(){
    float ret = 123.345,misc = 234.456;
    __asm__ volatile("lui a1,0x7fc00 \n\
        vsetivli    zero,8,e32,m1,ta,ma \n\
        vmv.s.x v9,a1 \n\
        lui a1,0x3f800 \n\
        vmv.v.x v8,a1 \n\
        vfredmax.vs v9,v8,v9 \n\
        vfmv.f.s    ft0,v9 \n\
        feq.s   a1,ft0,ft0 \n\
        beqz a1,lab1 \n\
        lui  a1,0x42090 \n\
        vmv.s.x  v9,a1 \n\
        vfredmax.vs  v8,v8,v9 \n\
        vfmv.f.s ft0,v8 \n\
        j    lab2 \n\
    lab1:\n\
        lui  a1,0x7fc00 \n\
        fmv.w.x  ft0,a1 \n\
    lab2:\n\
        lui  a1,0x420A0 \n\
        fmv.w.x  ft1,a1\n\
        fmax.s  ft2,ft1,ft0\n\
        fmv.s  %0,ft0\n\
        fmv.s  %1,ft2\n\
    " : "=f"(ret),
        "=f"(misc));
    printf("ret=%f misc=%f\n",ret,misc);
}
```

**Terminal Output**
- Gem5 output
  ```shell
  ret=nan misc=34.5
  ```
- Qemu/spike output
  ```shell
  ret=1.0 misc=34.5
  ```

**Expected behavior**
The first output(`ret`) should be 1. 

**Host Operating System**
`fedora 8.6`

**Host ISA**
X86

**Compiler used**
gcc version 10.3.0 (GCC)

**Additional information**
The [riscv spec document](https://riscv.org/wp-content/uploads/2017/05/riscv-spec-v2.2.pdf) states the following behaviour:

[image]

Upon inspecting the [source code](https://github.com/gem5/gem5/blob/stable/src/arch/riscv/isa/decoder.isa) of the implementation of the instructions, it seems that the `fmax_s` has a different implementation from the `vfredmax`.  The `fmax<>` is implemented in [utility.hh](https://github.com/gem5/gem5/blob/stable/src/arch/riscv/utility.hh#L392), which in turn uses the softfloat library.
```
0x1: fmax_s({{
                        float32_t fs1 = f32(freg(Fs1_bits));
                        float32_t fs2 = f32(freg(Fs2_bits));
                        float32_t fd;
                        bool greater = f32_lt_quiet(fs2, fs1) ||
                            (f32_eq(fs2, fs1) && bits(fs2.v, 31));

                        fd = greater || isNaNF32UI(fs2.v) ? fs1: fs2;
                        if (isNaNF32UI(fs1.v) && isNaNF32UI(fs2.v))
                            fd = f32(defaultNaNF32UI);
                        Fd_bits = freg(fd).v;
                        }}, FloatCmpOp);
```

```
                0x07: VectorReduceFloatFormat::vfredmax_vs({{
                    Vd_vu[0] = reduce_loop([](const vu& src1, const vu& src2) {
                        return fmax<et>(ftype<et>(src1), ftype<et>(src2));
                    }, Vs1_vu, Vs2_vu);
                }}, OPFVV, VectorFloatReduceOp);
```
Similar observation can be made about `fmin_s` and `vfredmin` as well. 
Related issue: #442

ivanaamit, 2024-02-05:
@debjyoti0891, do you have a fix for this issue? If yes, feel free to contribute. Thank you.

Do you know if the other issue you referenced (#442) is still an issue in the new version?

debjyoti0891, 2024-02-12:
@ivanaamit  Upon further debugging with the EXEC flags, it turns out that the bug was actually in `vfmv.f.s` instruction, which was incorrectly resulting in a NaN. I am currently in the process to file a PR that fixes this. 

I have not checked the status of the #442.

ivanaamit, 2024-04-01:
Since PR #863 has been merged, I am closing this issue. @debjyoti0891, please reopen it if I am mistaken about it being solved. Thanks.

closed 2024-04-01 by ivanaamit