gem5/gem5 #860: RVV instruction vmem faults caused by unreasonable vl

AppDoraemon, 2024-02-09:
**Describe the bug**
Since the vset instruction is modified to the execution mode of the branch instruction, this will cause a problem in the generation of subsequent vector instruction microinstructions. I observed that this problem occurred when an old vset instruction set vl to a large value and a subsequent vset instruction set vl to a small value. The subsequent vle instruction of the second vset instruction will create microinstructions according to the original larger vl during the instruction fetch phase. This will cause the number of microinstructions created to seriously exceed the upper limit, and ultimately cause the rename phase to be unable to find suitable rename registers.

The problem was solved when I set the upper limit on the number of vle/vse microinstructions to 8, but I'm not sure if this is the best way to solve the problem.

**Affects version**
Branch: develop

**To Reproduce**

Assembly instructions for running code
```
   10302:	0c37f6d7          	vsetvli	a3,a5,e8,m8,ta,ma             #First vset instruction, vl is set to 256.
   10306:	02060c07          	vle8.v	v24,(a2)
   1030a:	8f95                	sub	a5,a5,a3
   1030c:	02070c27          	vse8.v	v24,(a4)
   10310:	9636                	add	a2,a2,a3
   10312:	9736                	add	a4,a4,a3
   10314:	f7fd                	bnez	a5,10302 <main+0x1fc>
       ...                                              ...
   10356:	0587f5d7          	vsetvli	a1,a5,e64,m1,ta,mu             #Second vset instruction, vl is going to be set to 4.
   1035a:	0206fc07          	vle64.v	v24,(a3)             #The second vset has not completed execution. 
    #Use vl (256) to create 32 microops according to the speculative execution method (8 is the max num of vle microops) .
```

/src/arch/riscv/isa/templates/vector_mem.isa
```
    const int32_t micro_vlmax = vlen / width_EEW(_machInst.width);
    const uint32_t num_microops = ceil((float) this->vl / (micro_vlmax));       //May be the source of the error
    int32_t remaining_vl = this->vl;
    int32_t micro_vl = std::min(remaining_vl, micro_vlmax);

    StaticInstPtr microop;

    if (micro_vl == 0) {
        microop = new VectorNopMicroInst(_machInst);
        this->microops.push_back(microop);
    }
    for (int i = 0; i < num_microops && micro_vl > 0; ++i) {
        microop = new %(class_name)sMicro(_machInst, micro_vl, i, vlen);
        microop->setDelayedCommit();
        microop->setFlag(IsStore);
        this->microops.push_back(microop);
        micro_vl = std::min(remaining_vl -= micro_vlmax, micro_vlmax);
    }
```

**Terminal Output**

`gem5.opt: src/cpu/o3/scoreboard.hh:90: bool gem5::o3::Scoreboard::getReg(gem5::PhysRegIdPtr) const: Assertion `phys_reg->flatIndex() < numPhysRegs' failed.`

ivanaamit, 2024-03-08:
@saul44203, @aarmejach Bringing this issue to your attention since you are working on #885.

erin-le, 2025-05-01:
@AppDoraemon This issue is being closed as it may have been fixed by various changes to the RVV instructions. If you are still encountering this issue on current develop, please feel free to reopen. If you do reopen this issue, it would be helpful if you could also provide a binary to help reproduce it. Thank you!

closed 2025-05-01 by erin-le