gem5/gem5 #881: arch-riscv: return values of m5 pseudo instructions in 32 bit systems

robhau, 2024-02-20:
**Return value of m5 instructions**

M5 pseudo instructions return a `64 bit` value.

Example:

https://github.com/gem5/gem5/blob/8759131df3e4af1080fb96c67d8708869c9d74c0/src/sim/pseudo_inst.cc#L153C1-L158C2

**Current State** 

At the moment, for `32 bit` systems, the result of a m5 pseudo instruction is shrunk down to `32 bit` and stored only in `a0`

https://github.com/gem5/gem5/blob/8759131df3e4af1080fb96c67d8708869c9d74c0/src/arch/riscv/isa/formats/m5ops.isa#L38C1-L47C54

**Problem**

In `32 bit` systems, `64 bit` return values are split into two `32 bit` values and stored in `a0` and `a1`. Software still asumes that `a1` is part of the return value. This leads to wrong values (`a1` can hold old values before m5 function call).
State which version of gem5 this bug was found in. If on the develop branch state the Commit revision ID you are working.

**Gem5 Modification**

In `src/arch/isa/formats/m5ops.isa`, the upper 32 bit of the result have to be stored in `a1` for `RV32`. That could be look like

```C++
def format M5Op() {{
    iop = InstObjParams(name, Name, 'PseudoOp', '''
            uint64_t result;
            if (machInst.rv_type == RV32) {
                pseudo_inst::pseudoInst<RegABI32>(xc->tcBase(), M5FUNC, result);
                a0 = mbits(result, 31, 0);
                a1 = mbits(result, 63, 32);
            } else {
                pseudo_inst::pseudoInst<RegABI64>(xc->tcBase(), M5FUNC, result);
                a0 = rvSext(result);
            }''',
            ['IsNonSpeculative', 'IsSerializeAfter'])
    header_output = BasicDeclare.subst(iop)
    decoder_output = BasicConstructor.subst(iop)
    decode_block = BasicDecode.subst(iop)
    exec_output = BasicExecute.subst(iop)
}};
```

However, when I try to compile it, I receive that a1 is not declared. 

I added `a1` in `src/arch/riscv/isa/operands`, but when i look in the execution trace, the value is not written to `a1`.

rogerchang23424, 2024-02-20:
Hi @robhau,

We may need to update the implementation in the part to fix the issue

https://github.com/gem5/gem5/blob/8759131df3e4af1080fb96c67d8708869c9d74c0/src/arch/riscv/reg_abi.hh#L56-L73

rogerchang23424, 2024-02-21:
> if I understand correct, this only specifies the argument registers. That means we have to add something like `struct Result..` for the return value as well?

> Side question: How do you achieve the inlining of code snippets? I copy & paste a permalink, but that seems not to work.

https://github.com/gem5/gem5/blob/8759131df3e4af1080fb96c67d8708869c9d74c0/src/arch/riscv/isa/formats/m5ops.isa#L38-L47

robhau, 2024-02-21:
Thank you for the inline code explanation.

I looked a bit into the code to understand what happens.

I will split my work on this issue in smaller time slots and document my thoughts here.

https://github.com/gem5/gem5/blob/8759131df3e4af1080fb96c67d8708869c9d74c0/src/arch/riscv/reg_abi.hh#L59-L73

In function `get()`, the lower 31 bits are returned of each register. `state` is the argument register index and is incremented inside `get()`. It should be possible to merge the bits of two registers for 64 bit arguments. On problem is, that the argument list is not uniform:

- `m5_write_file(void* buffer, uint64_t len, uint64_t offset, const char* filename)`
- `m5_exit(uint64_t ns_delay)`
- `m5_add_symbol(uint64_t, const char* symbol)`

Therefore, the `M5OP_..` enum should somehow be passed to the `get` function in order to return the correct arguments.

rogerchang23424, 2024-02-22:
Thanks for the point out issue of M5 interact with RV32 core. We should merge two register values for 64bits in RV32 mode.

BTW, We need to update the `write_tohost` of `riscv-tests` to avoid falled to the bare-metals tests

robhau, 2024-02-22:
I hard coded the `get` function for `m5_write_file`.

```C++
static Arg
  get(ThreadContext *tc, typename ABI::State &state)
  {
      std::cout << "Arg get() " << state << std::endl;
      panic_if(state >= ABI::ArgumentRegs.size(),
              "Ran out of syscall argument registers.");

      if (state == 0) {
          return bits(tc->getReg(ABI::ArgumentRegs[state++]), 31, 0);
      } else if (state == 1) {
          uint64_t arg =
              (bits(tc->getReg(ABI::ArgumentRegs[1]), 31, 0) << 32) |
                  bits(tc->getReg(ABI::ArgumentRegs[2]), 31, 0);
          ++state;
          return arg;
      } else if (state == 2) {
          uint64_t arg =
              (bits(tc->getReg(ABI::ArgumentRegs[3]), 31, 0) << 32) |
                  bits(tc->getReg(ABI::ArgumentRegs[4]), 31, 0);
          ++state;
          return arg;
      } else {
          ++state;
          return bits(tc->getReg(ABI::ArgumentRegs[5]), 31, 0);
      }
  }
  ```
  
 In my workload I called `m5_write_file` a length of 16 and an offset of 1,099,512,676,352.
 The C code was
 ```C
uint8_t list[16] = {0, 5, 8, 3, 11, 15, 1, 4, 7, 6, 2, 10, 14, 12, 11, 9};
// some code
uint64_t offset = (1ULL << 40) | (1ULL << 20);
uint64_t len = 16;
m5_write_file(list,  len,  offset, "data.txt");
```

I executed it with debug flags `Exec` and `PseudoInst`.

Output:
```sh
437250000: system.cpu: T0 : 0x114 @main+66    : addi a3, zero, 368         : IntAlu :  D=0x0000000000000170
437300000: system.cpu: T0 : 0x118 @main+70    : lui a2, 256                : IntAlu :  D=0x0000000000100000
437350000: system.cpu: T0 : 0x11c @main+74    : c_addi a2, 1               : IntAlu :  D=0x0000000000100001
437400000: system.cpu: T0 : 0x11e @main+76    : c_slli a2, 20              : IntAlu :  D=0x0000010000100000
437400000: system.cpu: T0 : 0x120 @main+78    : c_li a1, 16                : IntAlu :  D=0x0000000000000010
437450000: system.cpu: T0 : 0x122 @main+80    : c_addi4spn a0, sp, 16      : IntAlu :  D=0x000000000005ffe0
437450000: system.cpu: T0 : 0x124 @main+82    : jal ra, -274               : IntAlu :  D=0x0000000000000128
438550000: global: pseudo_inst::pseudoInst(79)
438550000: global: pseudo_inst::writefile(0x5ffe0, 0x10, 0x10000100000, 0x170)
src/base/output.cc:92: panic: panic condition !_fstream->is_open() occurred: Failed to open "m5out/data.txt"
```
That should proof that we can merge the register to obtain the correct results.
I will continue tomorrow

robhau, 2024-02-23:
I pushed a rudimentary solution for the argument selection in my forked repository (https://github.com/robhau/gem5/tree/rv32m5ops). Next week, I aim for a correct return value.

robhau, 2024-04-14:
Fixed after merging https://github.com/gem5/gem5/pull/900

closed 2024-04-14 by robhau