gem5/gem5 #882: CHI (Arm) near/far atomic operation causes deadlock

yichungchen84, 2024-02-20:
GEM5 23.1.0.0:
CHI near/far operation causes deadlocks when CPU is configured with L1 private, L2 private, and L3 shared on a RNF, and another shared memory on HNF. (FS mode, Arm)

System configuration:
  CPU: 4 ArmO3CPUs share a RNF (L1 private, L2 private, and L3 shared ), total 2 RNF. 2 HNFs with distributed SLC.
  RAM: DDR4
  FS OS: ubuntu 20.04
(The configuration has no deadlock with 23.0.1.0.)

Current solution:
By removing near/far atomic operation feature in 23.1.0.0, FS mode runs with no deadlock.

Further study:
When the system configured without shared L3, it has no deadlock. Therefore, an initial guess is that near atomic operation might be the issue.

andysan, 2024-02-21:
Ping @tiagormk.

vsoriap, 2024-02-23:
Hi @yichungchen84, could you attach the config.ini and config.json you use.

Note that the far AMO feature is intended to be used only with three levels of cache. However, near AMO should work as the old version 23.0.0.

yichungchen84, 2024-02-26:
Hi @vsoriap 

Sorry that due to the IT restriction, I cannot share the config,ini/json files. I will see if I can share the configuration files (CHI_config.py, etc).

My configuration has private L1, private L2, shared L3, and SLC. I try to set all cache controllers "alloc_on_atomic = false". I assume it should turn off AMO feature (am I correct?), but it still has deadlocks with "alloc_on_atomic = false".

Would you mind to let me know where I can look into source code of far AMO feature/policy, so I can configure it to support my cache hierarchy? Thanks.

vsoriap, 2024-02-26:
Okay, then I see that you have 4 levels of cache. We never tested that configuration, and you may encounter some problems. How L3 and SLC have the flags is_HN configured?

`alloc_on_atomic` is not a feature that turns off AMO. AMOs are instructions supported in the RISC-V and ARM architecture. You cannot switch them off. What you can do is remove them from your binary.

What `alloc_on_atomic` really encodes is if the HNF needs to allocate or not a cache line if an Atomic Transaction arrives at the HNF and the cache line is not present. If true, the HNF requests the cache line to the SNF, performs the operation, and stores it in the local cache associated with the HNF (for your case I assume it's the SLC). If false, the HNF requests the cache line to the SNF, performs the atomic operation, and then it writes-back to SNF the updated value.

If you want to simulate AMOs as they were executed in gem5 version 23.0.0. you need to set `policy_type` to 0 (Execute all AMO in the L1).

You can check the AMO code [here](https://github.com/gem5/gem5/pull/177/commits/6411b2255ccdab3ad782f385b94cd43b248a46a1)

yichungchen84, 2024-02-26:
hi @vsoriap 

Thanks for the info. It is really helpful. 

My configuration: L3 is_HN = false, SLC is_HN = true. 

I tried another configuration L1 private, L2 shared, and SLC. This works without the issue. I guess 4-level cache it the problem.

I will try the policy_type = 0 later on.

ivanaamit, 2024-03-07:
Hi @yichungchen84, it appears that this issue has been resolved. However, I wanted to double-check before closing it. Thanks.

yichungchen84, 2024-03-07:
hi @ivanaamit 

Thanks. The issue is resolved. I was wondering whether it will support 4-level cache int the future?

tiagormk, 2024-03-08:
@ivanaamit, @yichungchen84, there are some fixes that are required so the Atomic operation is properly forwarded from the L3 to the HNF. I'm planning to have a look at that, so maybe we could keep this issue open and will keep you guys posted.

yichungchen84, 2024-03-08:
Thanks, @tiagormk . Let me know if I can be of any help.

yichungchen84, 2024-04-19:
@ivanaamit the problem is solved. Thanks.

closed 2024-04-20 by ivanaamit