gem5/gem5 #910: arch-x86,O3,KVM switching to full simulation bug

Foxy-Boxes, 2024-03-04:
**Describe the bug**
When using the full system simulation m5_switch_cpu_addr followed by m5_exit or m5_exit_addr causes either the simulation to segfault or the simulated program to segfault(One might also see an unexpected jump, I am guessing that is the same bug).

**Affects version**
gem5 stable c890e6b113a59caeec3eea9bafea4a03a471f89d 

**gem5 Modifications**
I have modified the config file, added my disk, and wrote a simple program to reproduce the bug. Also changed the default disk device for X86 KernelDiskWorkload to "/dev/sda".
 https://github.com/Foxy-Boxes/gem5-bug, has the necessary changes.

**To Reproduce**
download this tarred image to the repo: https://drive.google.com/file/d/1qH7QNoz23UIyuU07Rf7AoEvS3Opvcl74/view?usp=sharing
```shell
scons build/X86/gem5.debug -j `nproc`
./one_time.sh
./make_disk.sh
./build/X86/gem5.debug ./configs/example/gem5_library/x86-ubuntu-run-with-kvm-test.py
````

**Terminal Output**
In the case where we get a segfault

```
warn: The `CustomResource` class is deprecated. Please use an `AbstractResource` subclass instead.
Global frequency set at 1000000000000 ticks per second
warn: No dot file generated. Please install pydot to generate the dot file and pdf.
src/mem/dram_interface.cc:690: warn: DRAM device capacity (8192 Mbytes) does not match the address range assigned (4096 Mbytes)
src/sim/kernel_workload.cc:46: info: kernel located at: resources/vmlinux
src/base/statistics.hh:279: warn: One of the stats is a legacy stat. Legacy stat is a stat that does not belong to any statistics::Group. Legacy stat is deprecated.
      0: board.pc.south_bridge.cmos.rtc: Real-time clock set to Sun Jan  1 00:00:00 2012
board.pc.com_1.device: Listening for connections on port 3456
src/base/statistics.hh:279: warn: One of the stats is a legacy stat. Legacy stat is a stat that does not belong to any statistics::Group. Legacy stat is deprecated.
src/dev/intel_8254_timer.cc:128: warn: Reading current count from inactive timer.
board.remote_gdb: Listening for connections on port 7000
src/cpu/kvm/base.cc:169: info: KVM: Coalesced MMIO disabled by config.
src/cpu/kvm/base.cc:169: info: KVM: Coalesced MMIO disabled by config.
gem5 Simulator System.  https://www.gem5.org
gem5 is copyrighted software; use the --copyright option for details.

gem5 version 23.1.0.0
gem5 compiled Mar  4 2024 20:46:18
gem5 started Mar  4 2024 20:56:29
gem5 executing on sapphire, pid 1654951
command line: ./build/X86/gem5.debug ./configs/example/gem5_library/x86-ubuntu-run-with-kvm-test.py

src/sim/simulate.cc:199: info: Entering event queue @ 0.  Starting simulation...
src/arch/x86/kvm/x86_cpu.cc:1653: warn: kvm-x86: MSR (0x3a) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1653: warn: kvm-x86: MSR (0x48) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1653: warn: kvm-x86: MSR (0xe1) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1653: warn: kvm-x86: MSR (0x309) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1653: warn: kvm-x86: MSR (0x30a) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1653: warn: kvm-x86: MSR (0x30b) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1653: warn: kvm-x86: MSR (0x38d) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1653: warn: kvm-x86: MSR (0x38e) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1653: warn: kvm-x86: MSR (0x38f) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1653: warn: kvm-x86: MSR (0x390) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1653: warn: kvm-x86: MSR (0xc1) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1653: warn: kvm-x86: MSR (0xc2) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1653: warn: kvm-x86: MSR (0xc3) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1653: warn: kvm-x86: MSR (0xc4) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1653: warn: kvm-x86: MSR (0xc5) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1653: warn: kvm-x86: MSR (0xc6) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1653: warn: kvm-x86: MSR (0xc7) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1653: warn: kvm-x86: MSR (0xc8) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1653: warn: kvm-x86: MSR (0x186) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1653: warn: kvm-x86: MSR (0x187) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1653: warn: kvm-x86: MSR (0x188) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1653: warn: kvm-x86: MSR (0x189) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1653: warn: kvm-x86: MSR (0x18a) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1653: warn: kvm-x86: MSR (0x18b) unsupported by gem5. Skipping.
[... 457 lines cut ...]
int main()
{
    m5op_addr = 0xFFFF0000;
    map_m5_mem();
    printf("Hello world\n");
    m5_switch_cpu_addr();
    while (1)
        ;
    m5_exit_addr(0);
}
 
```
Therefore I would expect an infinite loop after the "Hello world\n" print.

**Host Operating System**
Ubuntu 22.04

**Host ISA**
X86

**Compiler used**
g++ 11.4

**Additional information**
This bug is at least as old as the commit 1db206b9d371b14cb9150e37aef444d1d59db025 on stable

Foxy-Boxes, 2024-03-08:
I have not changed the switching behavior and I expect this to work, this is still a bug in the simulation.

Foxy-Boxes, 2024-03-08:
I have not changed switching behavior and this is still a bug in the simulator, I am using a workaround to avoid this situation. If I had time I would attempt fixing it but for now I will continue to use the workaround.

Lukas-Zenick, 2024-04-04:
Hello, my name is Lukas and I’m working with my partner Emily on a final project for our Virtualization class. We are Computer Science at the University of Texas at Austin and currently enrolled in CS360V Virtualization, taught by Dr. Chidambaram (VJ). As part of our coursework, we are required to contribute to a few issues within an open-source repository relevant to the class.

After reviewing the gem5 repository, we believe that the particular issue aligns well with our objectives and skill set and were wondering if we may be assigned to this issue. If that's possible, before proceeding, we wanted to ask if there are any additional details or considerations regarding this issue that we should be aware of. We are committed to completing all three issues by May 6th to the best of our abilities. 

Thank you for considering our request.
Emily and Lukas

powerjg, 2024-04-04:
Awesome! Yes, please feel free to try to tackle this issue. If you have any questions, don't hesitate to ask. @BobbyRBruce has successfully reproduced the issue, but we haven't had the resources to fix it. Hopefully Ivana's comment above gives you a starting point.

lllzy040620, 2024-06-30:
Has anyone fixed these bugs? I encountered the same issues while trying to run x86-ubuntu-run-with-kvm.py in my disk-image.

ivanaamit, 2024-07-18:
@lllzy040620, I don't think this is fixed. If you know how to resolve it, please feel free to create a PR with your changes. Thank you.

lllzy040620, 2024-07-30:
@ivanaamit I resolved this issue by downgrading gem5 from version 24.0.1 to 22.0.1. I believe the problem might have been caused by an incompatibility between the gem-resource's version and gem5.

lllzy040620, 2024-09-02:
In fact, I just followed the README to set up Gem5 and Gem5-resource. Details are recorded in this [https://github.com/orgs/gem5/discussions/1302](link).