gem5/gem5 #916: Linking Error when enable AddressSanitizer

wkxfudan, 2024-03-06:
**Describe the bug**
When enable AddressSanitizer, the linking will report an error.
```
$ python3 `which scons` build/RISCV/gem5.opt -j`nproc` --with-asan --without-tcmalloc```
```

```
scons: Reading SConscript files ...
Mkdir("/home/wangkaixuan/RantedWorkspace/pe-gem5/build/RISCV/gem5.build")
Checking for linker -Wl,--as-needed support... (cached) yes

Warning: To suppress false positive leaks, set the LSAN_OPTIONS environment variable to "suppressions=/home/wangkaixuan/RantedWorkspace/pe-gem5/util/lsan-suppressions:print_suppressions=0" when running gem5
Warning: LSAN_OPTIONS=suppressions=/home/wangkaixuan/RantedWorkspace/pe-gem5/util/lsan-suppressions:print_suppressions=0

Checking for compiler -gz support... (cached) yes
Checking for linker -gz support... (cached) yes
Info: Using Python config: python3-config
Checking for C header file Python.h... (cached) yes
Checking Python version... (cached) 3.10.12
Checking for accept(0,0,0) in C++ library None... (cached) yes
Checking for zlibVersion() in C++ library z... (cached) yes
Building in /home/wangkaixuan/RantedWorkspace/pe-gem5/build/RISCV
"build_tools/kconfig_base.py" "/home/wangkaixuan/RantedWorkspace/pe-gem5/build/RISCV/gem5.build/Kconfig" "/home/wangkaixuan/RantedWorkspace/pe-gem5/src/Kconfig"
Checking size of struct kvm_xsave ... (cached) yes
Checking for C header file fenv.h... (cached) yes
Checking for C header file png.h... (cached) yes
Checking for clock_nanosleep(0,0,NULL,NULL) in C library None... (cached) yes
Checking for C header file valgrind/valgrind.h... (cached) no
Checking for pkg-config package hdf5-serial... (cached) yes
Checking for H5Fcreate("", 0, 0, 0) in C library hdf5... (cached) yes
Checking for H5::H5File("", 0) in C++ library hdf5_cpp... (cached) yes
Checking for C header file capstone/capstone.h... (cached) no
Warning: Header file <capstone/capstone.h> not found.
         This host has no capstone library installed.
Checking for C header file linux/kvm.h... (cached) yes
Checking for timer_create(CLOCK_MONOTONIC, NULL, NULL) in C library None... (cached) yes
Checking for member exclude_host in struct perf_event_attr...(cached) yes
Checking for C header file linux/if_tun.h... (cached) yes
Checking for shm_open("/test", 0, 0) in C library None... (cached) yes
Checking for pkg-config package protobuf... (cached) yes
Checking for backtrace_symbols_fd((void *)1, 0, 0) in C library None... (cached) yes
Checking whether __i386__ is declared... (cached) no
Checking whether __x86_64__ is declared... (cached) yes
Checking for compiler -Wno-self-assign-overloaded support... (cached) yes
Checking for linker -Wno-free-nonheap-object support... (cached) yes
scons: done reading SConscript files.
scons: Building targets ...
 [VER TAGS]  -> RISCV/sim/tags.cc
 [    LINK]  -> RISCV/gem5.opt
`_ZN4gem58RiscvISAL14vecRegClassOpsE' referenced in section `.ldata.rel.local' of build/RISCV/arch/riscv/generated/inst-constrs.o: defined in discarded section `.data.rel.local._ZN4gem58RiscvISAL14vecRegClassOpsE[_ZN4gem58RiscvISA11vecRegClassE]' of build/RISCV/arch/riscv/generated/inst-constrs.o
`_ZN4gem58RiscvISAL14vecRegClassOpsE' referenced in section `.data.rel.local' of build/RISCV/arch/riscv/generated/generic_cpu_exec.o: defined in discarded section `.data.rel.local._ZN4gem58RiscvISAL14vecRegClassOpsE[_ZN4gem58RiscvISA11vecRegClassE]' of build/RISCV/arch/riscv/generated/generic_cpu_exec.o
`_ZN4gem58RiscvISAL14vecRegClassOpsE' referenced in section `.data.rel.local' of build/RISCV/arch/riscv/insts/vector.o: defined in discarded section `.data.rel.local._ZN4gem58RiscvISAL14vecRegClassOpsE[_ZN4gem58RiscvISA11vecRegClassE]' of build/RISCV/arch/riscv/insts/vector.o
collect2: error: ld returned 1 exit status
scons: *** [build/RISCV/gem5.opt] Error 1
scons: building terminated because of errors.
*** Summary of Warnings ***
Warning: To suppress false positive leaks, set the LSAN_OPTIONS environment variable to "suppressions=/home/wangkaixuan/RantedWorkspace/pe-gem5/util/lsan-suppressions:print_suppressions=0" when running gem5
Warning: LSAN_OPTIONS=suppressions=/home/wangkaixuan/RantedWorkspace/pe-gem5/util/lsan-suppressions:print_suppressions=0
Warning: Header file <capstone/capstone.h> not found.
         This host has no capstone library installed.
```

**Affects version**
V23-1

**gem5 Modifications**
No Modification

powerjg, 2024-03-06:
What compiler/version are you using? It's possible this is a compiler bug.

wkxfudan, 2024-03-13:
```
$ python3 --version                                                  
Python 3.10.12

$ scons --version                           
SCons by Steven Knight et al.:
        SCons: v4.0.1.c289977f8b34786ab6c334311e232886da7e8df1, 2020-07-17 01:50:03, by bdbaddog on ProDog2020
        SCons path: ['/usr/lib/python3/dist-packages/SCons']
Copyright (c) 2001 - 2020 The SCons Foundation

$ gcc --version                   
gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

BobbyRBruce, 2024-03-19:
I was able to recreate this error with the following script:

```sh
#!/bin/bash

set -e
set -x

git clone https://github.com/gem5/gem5.git gem5-issue-916
cd gem5-issue-916
git checkout v23.1.0.0
pwd
docker run -u ${UID}:${GID} -v $(pwd):/gem5 -w /gem5 -it --rm ghcr.io/gem5/gcc-version-11:latest scons build/RISCV/gem5.opt -j`nproc` --with-asan --without-tcmalloc
```

I'm going to run a few more experiments to see if I can hone in on exactly what/where the issue lays.

BobbyRBruce, 2024-03-21:
I have ran various tests and found that the incompatibility is with GCC and using the AddressSanitizer. Clang works just fine.

Though this is somewhat dismissive, I believe this bug is symptomatic of some complicated code we have in our codebase. For example:

```
`_ZN4gem58RiscvISAL14vecRegClassOpsE' referenced in section `.data.rel.local' of build/RISCV/arch/riscv/insts/vector.o: defined in discarded section `.data.rel.local._ZN4gem58RiscvISAL14vecRegClassOpsE[_ZN4gem58RiscvISA11vecRegClassE]' of build/RISCV/arch/riscv/insts/vector.o
```

appears (to me at least) to be getting confused by this: https://github.com/gem5/gem5/blob/e8bc4fc137a5a7e82b601432271a027b652ae69b/src/arch/riscv/regs/vector.hh#L69 

I don't even know what the `static inline` does here exactly...

I have created #951 to disable the Address Sanitizer for GCC. This doesn't fix the bug per-se but will stop users accidentally running into it.

**Note**: I also found the Undefined Behavior Sanitizer was failing in GCC with the following error (when compiling using `scons build/ALL/gem5.opt -j`nproc` --with-ubsan`):

```sh
[     CXX] src/base/loader/image_file_data.cc -> ALL/base/loader/image_file_data.o
In file included from /usr/include/string.h:535,
                 from /usr/include/c++/11/cstring:42,
                 from src/base/cprintf_formats.hh:33,
                 from src/base/cprintf.hh:38,
                 from src/base/logging.hh:49,
                 from src/base/loader/image_file_data.cc:40:
In function ‘char* strcpy(char*, const char*)’,
    inlined from ‘int gem5::loader::doGzipLoad(int)’ at src/base/loader/image_file_data.cc:70:11,
    inlined from ‘gem5::loader::ImageFileData::ImageFileData(const string&)’ atsrc/base/loader/image_file_data.cc:116:24:
/usr/include/x86_64-linux-gnu/bits/string_fortified.h:79:33: error: ‘void* __builtin_memcpy(void*, const void*, long unsigned int)’ offset [0, 19] is out of the bounds [0, 0] [-Werror=array-bounds]
   79 |   return __builtin___strcpy_chk (__dest, __src, __glibc_objsize (__dest));
      |          ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
scons: *** [build/ALL/base/loader/image_file_data.o] Error 1
scons: building terminated because of errors.
```

In similar vein, I used PR #951 to remove this for GCC. I don't know exactly what is causing the problem.

BobbyRBruce, 2024-03-23:
Correction on my note: the `--with-ubsan` bug is fixable. I've created a PR (#959) which addresses this problem. PR #951 has been updated to just remove the address sanitizer.

closed 2024-04-08 by ivanaamit